eugene-tarassov/vivado-risc-v #1: GPIO support

FuTaimeng, 2020-06-04:
Could you please add GPIO support? Or, give us a development pipeline. Thanks a lot.

eugene-tarassov, 2020-06-04:
I'm not sure what you mean by "give us a development pipeline" - it seems what this repository already does that. The main idea of the repository is to generate complete Vivado project (and Linux), then use Vivado Block Design to add things like GPIO easily. Look for AXI Uartlite (in the IO block) as an example, add AXI GPIO and connect it same way as the UART.

I'm considering writing some tutorials about modding the design, but it will take time.

FuTaimeng, 2020-06-07:
Thanks, a tutorial will help a lot. Looking forward to it!

I already have workable GPIO and HDMI on board. But I'm puzzling about how to control them in Linux. I added GPIO in block design and turned on GPIO Support when compiling the kernel. /sys/class/gpio appears but no gpiochip was found.
May you give me some advice, like tutorials or documentations I should read, or keywords to google?

eugene-tarassov, 2020-06-07:
In Linux, you need to add GPIO in the kernel and in the device tree.
See [AXI GPIO on Xilinx wiki](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841846/AXI+GPIO)

eugene-tarassov, 2020-06-08:
I have updated README with info on adding devices. Not exactly a tutorial, but it covers main points. I also changed Linux config to enable AXI GPIO driver by default.

closed 2020-06-08 by eugene-tarassov

Yuxin-Yu, 2023-03-23:
Hello, I want to confirm whether the GPIO device tree I added is correct. For example, if I use Rocket64b1 configuration and the development board is VC707, should the GPIO device tree be added to workspace/Locket64b1/system vc707.dts or workspace/Locket64b1/system. dts? In addition, if I want to add other peripherals, should I modify the device tree of the linux-stable subproject, or just modify patches/linux.config @eugene-tarassov

eugene-tarassov, 2023-03-23:
Changes in the top level design (like adding GPIO) need to be done in board/vc707 files. In particular, device tree changes should go into board/vc707/bootrom.dts.

I need to update README since device tree file location has changed a while ago.

Yuxin-Yu, 2023-03-24:
OK，thank you.