eugene-tarassov/vivado-risc-v #2: Nexys Video cannot boot Linux

Y-BoBo, 2020-07-13:
Hello, when I boot with SD card, the board cannot boot Linux, and it displays some errors like this
the first part
```
[   10.166214] 9pnet: Installing 9P2000 support
[   10.171324] Key type dns_resolver registered
[   10.218466] Freeing unused kernel memory: 232K
[   10.222030] mmc0: host does not support reading read-only switch, assuming write-enable
[   10.230936] Run /init as init process
[   10.234486] mmc0: new high speed SDHC card at address 5048
[   10.249550] mmcblk0: mmc0:5048 SD16G 14.4 GiB 
[   10.380304] blk_update_request: I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.411564] blk_update_request: I/O error, dev mmcblk0, sector 1 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.443152] blk_update_request: I/O error, dev mmcblk0, sector 2 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.474474] blk_update_request: I/O error, dev mmcblk0, sector 3 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.505794] blk_update_request: I/O error, dev mmcblk0, sector 4 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.537038] blk_update_request: I/O error, dev mmcblk0, sector 5 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   10.568482] blk_update_request: I/O error, dev mmcblk0, sector 6 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Loading, please wait...
```
the second part
```
Begin: Running /scripts/local-block ... done.
[   71.205838] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
[   71.620102] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
Begin: Running /scripts/local-block ... done.
[   73.258002] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
[   73.672090] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
Begin: Running /scripts/local-block ... done.
[   74.979542] print_req_error: 86 callbacks suppressed
[   74.984392] blk_update_request: I/O error, dev mmcblk0, sector 30244736 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.017256] blk_update_request: I/O error, dev mmcblk0, sector 30244737 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.050016] blk_update_request: I/O error, dev mmcblk0, sector 30244738 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.082648] blk_update_request: I/O error, dev mmcblk0, sector 30244739 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.115292] blk_update_request: I/O error, dev mmcblk0, sector 30244740 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.148144] blk_update_request: I/O error, dev mmcblk0, sector 30244741 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.181166] blk_update_request: I/O error, dev mmcblk0, sector 30244742 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.213994] blk_update_request: I/O error, dev mmcblk0, sector 30244743 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   75.252128] blk_update_request: I/O error, dev mmcblk0, sector 30244736 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   75.284382] blk_update_request: I/O error, dev mmcblk0, sector 30244737 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   75.425930] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
[   75.840078] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
Begin: Running /scripts/local-block ... done.
[   77.476068] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
[   77.889882] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
done.
[   78.432032] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
[   78.845992] Buffer I/O error on dev mmcblk0, logical block 3780592, async page read
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist.  Dropping to a shell!
(initramfs)
```
what should I do?

eugene-tarassov, 2020-07-13:
It looks like missing or corrupted root file system. The root FS is supposed to be the second partition on the SD card, formatted as EXT4, with UUID 68d82fa1-1bb5-435f-a5e3-862176586eec.

Can you see the partition if you mount the card on a Linux PC?
How exactly did you populate the SD card?

Y-BoBo, 2020-07-14:
I check the UUID of my SD card, it's `68d82fa1-1bb5-435f-a5e3-862176586eec` indeed. And I restart the step of making sd_card.img, Linux still cannot boot, the message of error is the same as before.

I tried as the flow steps: (didn't generate `riscv-nexys-video.bit`)
Download the `.bit` in the released file v2.3.0, then program it into the board,  insert the SD card which has been partitioned successfully before. 
Finally, Linux cannot boot as before. I changed 2 sd cards (16G), both of them can be partitioned successfully with the command `./mk-sd-card`, but Linux cannot boot
```
sudo apt install git make
git clone https://github.com/eugene-tarassov/vivado-risc-v.git
cd vivado-risc-v
make apt-install
make update-submodules

./mk-sd-card
```

eugene-tarassov, 2020-07-14:
I did exactly same steps, and it worked fine for me.
I guess it is something specific to your SD card.
What is the type of your card?
I'm using SanDisk 32GB Ultra microSDHC A1 UHS-I/U1 Class 10.

Y-BoBo, 2020-07-15:
After changing three SD cards, linux finally booted.
I have some other questions:
1. I want to add an accelerator to the soc, and I have added it to rocket in another project, what should I do in this project to add the accelerator? Add it in the block design in vivado? 
2. How can I adjust the system clock? I want to change the clock of Rocket to 100M or higher.
3. What should I do to boot linux after adding a slave device in the block of vivado? BBL?ROOTFS?

eugene-tarassov, 2020-07-15:
Please, let me know type/model of SD card that does not work, I'd like to investigate. It can be caused by a bug in the driver. In this case I need to fix it.

Rocket Chip approach is to use Scala and Chisel to implement an accelerator. This should work fine with this project - no different from any other Rocket based projects. This project also supports traditional approach: using Verilog or VHDL to implement a device connected over AXI bus. In this case you can use Xilinx IP integrator workflow - Vivado Block Design. README contains a description of that, including changes in Linux.

You cannot run Rocket at 100MHz on Nexys Video board. The board is based on slow, low power Artix-7 FPGA. It can only run Rocket up to about 55MHz. You need Virtex-7 or Virtex Ultrascale FPGA to run Rocket cores at 100MHz or more.  However, you can run your accelerator at higher frequency. For example, look at ethernet block in the design - it runs at 125MHz. Vivado handling of multiple clock domains is pretty good, having separate clock for a device is really easy to do.

Y-BoBo, 2020-07-17:
The following two cards can't work normally:
*  `Netac 16GB Ultra microSDHC A1 UHS-I/U1 Class 10` 
*  `Kingston 16GB microSDHC Canvas Select Plus 100MB/s Read A1 Class 10 UHS-I`

By the way, I have changed the clock1 of the clk_wiz_0 to 100MHz,  which connect to Rocket Chip. But the performance didn't improve obviously.
And what I am doing is adding Gemmini accelerator to rocketship, and adding HDMI peripheral, led display to this project. This progress might have a lot of problems. 😅

Y-BoBo, 2020-07-18:
When I add a new slave in I/O block which is HDMI, I tried as the step in Readme: check drivers in `linux.config` and edit `bootrom.dts`. 

* In the first step, I found there are two HDMI configs in `linux.config`, they are 
```
CONFIG_HDMI=y
```
```
# CONFIG_DRM_RCAR_DW_HDMI is not set
```
   I think the first one is the configuration of HDMI and it has been configured `yes`.
* In the second step, I understand every line in `bootrom.dts`, but I don't know how to edit it. I found some examples in `https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842279/Build+Device+Tree+Blob`. But I'm not sure whether these examples are adaptive in Nexys-video. Such as:
```
        eth: eth0@60020000 {
            compatible = "riscv,axi-ethernet-1.0";
            reg = <0x60020000 0x10000>;
            local-mac-address = [00 0a 35 00 00 00];
            interrupt-parent = <&L2>;
            interrupts = <4>;
```
I know the address of HDMI module, the reg width and the interrupts of it, Bus I don't know other informations.

eugene-tarassov, 2020-07-18:
I'm not sure what you mean by "HDMI module". Complete video subsystem consists of several IP blocks: rgb-to-dvi encoder, video timing controller, DMA controller, etc., and several drivers to control all that. I cannot really help with that - I have no experience designing video subsystem. Digilent and Xilinx provide demos and tutorials for FPGA video design - can be used as a starting point.

eugene-tarassov, 2020-07-20:
I reproduced the SD card issues after ordering Netac, Kingston and some other cards for testing. I have committed a fix - it was a bug in the Verilog code of the SD card controller. Thanks for the report!

eugene-tarassov, 2020-07-21:
I published release v2.4.0, which contains SD card fix. Closing the issue.

closed 2020-07-21 by eugene-tarassov