eugene-tarassov/vivado-risc-v #7: Peripheral device cannot work

Y-BoBo, 2020-08-11:
Hello, 
1. I have added axi-ps2 IP  into the block design, which can be found in (https://github.com/Digilent/Nexys-Video-AXI-PS2-Keyboard), but there is no Linux driver for this IP.  
2. I read the documents in the IP directory, and changed the 'Register offsets for the xps2 device' of the `xilinx_ps2.c` which can be found in `vivado-risc-v/linux-stable/drivers/input/serio/xilinx_ps2.c`.
3. After doing this, I can't use the keyboard in the FPGA, I don't know whether the step I do is right or not?

eugene-tarassov, 2020-08-11:
It looks like the driver needs more modifications, like bit positions in the Status register.
The best way to debug a driver is to use System Debugger in Vitis 2020.1.
This way you can set breakpoints in the driver code, step through the code, and see what is going on.

If you decide to debug the driver, I recommend to add `#pragma GCC optimize("O0")` at the beginning of the driver code - it is much easier to debug a code when it is not optimized by the compiler.

eugene-tarassov, 2020-08-18:
Since axi-ps2 IP is not part of this project, I'm closing the issue.

closed 2020-08-18 by eugene-tarassov