eugene-tarassov/vivado-risc-v #8: Makefile:262: recipe for target 'workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit' failed

hyy313, 2020-09-25:
when i run `make CONFIG=rocket32s1 BOARD=genesys2 bitstream ` 

```
INFO: [Common 17-344] 'launch_runs' was cancelled
INFO: [Common 17-344] 'source' was cancelled
Makefile:262: recipe for target 'workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit' failed
make: *** [workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit] Error 1
```
my env: ubuntu 18.04.5 LTS,vivado 2020.1
i run step by step,how should i fix this problem,thank you for your help!

hyy313, 2020-09-25:
when i run again it success，i don't kown why

eugene-tarassov, 2020-09-25:
"'launch_runs' was cancelled" means error in Vivado during HDL compilation.
You need to look for actual Vivado error message, it should explain what happens.
Your snippet of Vivado output is too short, the error message should be somewhere before that.
Most probably, it is a bug in Vivado 2020.1, I have seen it crashes sometimes.
I suggest to use Vivado 2019.2 - it looks much more stable than 2020.1.
Also, 2020.1.1 update might have the bug fixed, but I have not verified that.

hyy313, 2020-09-25:
when i run in win10 vmware16 ubuntu 18.04.5LTS vivado 2020.1

i will try vivado 2019 tomorrow

```
echo "open_project workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.xpr">workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
echo "update_compile_order -fileset sources_1">>workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
echo "launch_runs impl_1 -to_step write_bitstream -jobs 4">>workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
echo "wait_on_run impl_1">>workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
echo "write_cfgmem -format mcs -interface SPIx4 -size 256 -loadbit \"up 0x0 workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit\" -file workspace/rocket32s1/genesys2-riscv.mcs -force">>workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2020.1/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_sim_netlist.vhdl'. Please regenerate to continue.
open_project: Time (s): cpu = 00:00:19 ; elapsed = 00:00:23 . Memory (MB): peak = 2165.410 ; gain = 2.016 ; free physical = 3907 ; free virtual = 7800
update_compile_order: Time (s): cpu = 00:00:30 ; elapsed = 00:00:22 . Memory (MB): peak = 2293.473 ; gain = 128.062 ; free physical = 3875 ; free virtual = 7800
[Sat Sep 26 00:16:09 2020] Launched riscv_mig_7series_0_0_synth_1, riscv_RocketChip_0_synth_1, synth_1...
Run output will be captured here:
riscv_mig_7series_0_0_synth_1: /home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/riscv_mig_7series_0_0_synth_1/runme.log
riscv_RocketChip_0_synth_1: /home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/riscv_RocketChip_0_synth_1/runme.log
synth_1: /home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/synth_1/runme.log
[Sat Sep 26 00:16:10 2020] Launched impl_1...
Run output will be captured here: /home/hyy/genesys/vivado-risc-v/workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/runme.log
launch_runs: Time (s): cpu = 00:00:37 ; elapsed = 00:00:25 . Memory (MB): peak = 2293.473 ; gain = 0.000 ; free physical = 3855 ; free virtual = 7784
[Sat Sep 26 00:16:10 2020] Waiting for impl_1 to finish...
[Sat Sep 26 00:21:09 2020] impl_1 finished
WARNING: [Vivado 12-8222] Failed run(s) : 'riscv_RocketChip_0_synth_1'
[... cut ...]

    while executing
"write_cfgmem -format mcs -interface SPIx4 -size 256 -loadbit "up 0x0 workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapp..."
    (file "workspace/rocket32s1/vivado-genesys2-riscv/make-bitstream.tcl" line 5)
Makefile:262: recipe for target 'workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit' failed
make: *** [workspace/rocket32s1/vivado-genesys2-riscv/genesys2-riscv.runs/impl_1/riscv_wrapper.bit] Error 1
```

eugene-tarassov, 2021-02-25:
Problems like this are caused by insufficient RAM size.
Vivado handles memory overflow poorly. It can crash, hang or emit a cryptic message.
32GB RAM is sufficient in most cases, but some of largest designs need a machine with 128GB RAM.

closed 2021-02-25 by eugene-tarassov