eugene-tarassov/vivado-risc-v #9: Makefile stops executing after generating HDL

Kavyaborra, 2020-10-02:
When I run make CONFIG=rocket64x1 BOARD=nexys-video bitstream

Execution stops after the generate Rocket SoC HDL section.

I get the following output in my console after Makefile: 200 :

Computed transform order in: 3171.4 ms

The next make target, generate Rocket SoC Wrapper is not executed.

eugene-tarassov, 2020-10-02:
rocket64x1 is too big for Nexys Video board.
The configuration needs about 174,000 FPGA LUTs, Nexys Video has only 133,800.
So, Vivado cannot do place and route, there should be error message somewhere in the Vivado logs.
For BOOM CPU, you need a bigger FPGA: Genesys 2 (203,800 LUTs) or VC707 (303,600 LUTs).

Kavyaborra, 2020-10-03:
Hey, I got the same problem with VC707 as well.
[image]

eugene-tarassov, 2020-10-03:
I see, it hangs before launching Vivado.
This is how it looks on my machine:
```
java -Xmx12G -Xss8M -cp rocket-chip/firrtl/utils/bin/firrtl.jar:target/scala-2.12/classes:rocket-chip/target/scala-2.12/classes:rocket-chip/chisel3/target/scala-2.12/classes:rocket-chip/chisel3/core/target/scala-2.12/classes:rocket-chip/chisel3/macros/target/scala-2.12/classes firrtl.stage.FirrtlMain -i workspace/rocket64x1/system-vc707/Vivado.Rocket64x1.fir -o system-vc707.v -X verilog --infer-rw RocketSystem --repl-seq-mem \
         -c:RocketSystem:-o:workspace/rocket64x1/system.conf \
         -faf workspace/rocket64x1/system.anno.json \
         -td workspace/rocket64x1/ \
         -fct firrtl.passes.InlineInstances
Computed transform order in: 271.6 ms
Total FIRRTL Compile Time: 178144.4 ms
rocket-chip/scripts/vlsi_mem_gen workspace/rocket64x1/system.conf >workspace/rocket64x1/srams.v
mkdir -p vhdl-wrapper/bin
javac -g -nowarn \
  -sourcepath vhdl-wrapper/src -d vhdl-wrapper/bin \
  -classpath vhdl-wrapper/antlr-4.8-complete.jar \
  vhdl-wrapper/src/net/largest/riscv/vhdl/Main.java
java -classpath \
  vhdl-wrapper/src:vhdl-wrapper/bin:vhdl-wrapper/antlr-4.8-complete.jar \
  net.largest.riscv.vhdl.Main \
  workspace/rocket64x1/system-vc707.v >workspace/rocket64x1/rocket.vhdl
echo "set vivado_board_name vc707" >workspace/rocket64x1/system-vc707.tcl
echo "set vivado_board_part xilinx.com:vc707:part0:1.4" >>workspace/rocket64x1/system-vc707.tcl
echo "set xilinx_part xc7vx485tffg1761-2" >>workspace/rocket64x1/system-vc707.tcl
echo "set riscv_clock_frequency 50.0" >>workspace/rocket64x1/system-vc707.tcl
echo 'set script_folder [file dirname [file normalize [info script]]]' >>workspace/rocket64x1/system-vc707.tcl
echo 'cd $script_folder' >>workspace/rocket64x1/system-vc707.tcl
echo 'source ../../vivado.tcl' >>workspace/rocket64x1/system-vc707.tcl
+ make CONFIG=rocket64x1 BOARD=vc707 vivado-project
if [ ! -e workspace/rocket64x1/vivado-vc707-riscv ] ; then vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64x1/system-vc707.tcl ; fi
```
The next message should be _Total FIRRTL Compile Time ..._
FIRRTL  needs 3 min and 12G of memory for this step.
Does your machine have enough RAM?
It, probably, just takes very long time to finish this step.

closed 2021-02-25 by eugene-tarassov