eugene-tarassov/vivado-risc-v #10: Problems with Rocket simulation

Rucadi, 2020-12-14:
Hi,

I'm having trouble getting this to work with my FPGA board.

I'm using your project because of the nice vhdl-wrapper to interface it with vivado block-design, 
however, I'm trying to debug my custom bootrom and I'm having some trouble.

First of all, no matter the RocketChip configuration, if I try to debug it using vivado, 
the PC and other important signals get stuck at "X". 

However, If I try a BOOM configuration, the bootrom seems to execute fine, It seems like the MEM port works,
but I can't get any read/write on the IO_PORT.

Is there any bug related to this behaviour?

My simulation is quite simple, a testbench with a reset and a clock.
While the block design has connected the mem and io to a bram.

If you could give me some guidelines to get this to work I would be extremely happy.

Also, I saw that you check for "rocket64" string in the configuration in order to decide the gcc-toolchain for the bootrom.
I would suggest  to make it caps-unaware since writing "Rocket64x1" on CONFIG  will create the processor, but with a 32bit bootrom attached.

Thaks for your time!

eugene-tarassov, 2020-12-14:
By "simulation", do you mean Run Simulation commands in Vivado?

I have no idea why simulation would not work. I have not tried it myself. It is an important use case, I'll investigate.

I'll change the Makefile to allow both "rocket" and "Rocket".

Rucadi, 2020-12-14:
Yes, I'm running a simulation of my system in vivado, creating a test and looking at the waveform to check if the behaviour is correct while executing a custom bootrom that interacts with other devices.

it seems to be working with MegaBoom configuration rocket64z1 (IO included).
But for some reason the RocketChip configurations doesn't even boot. (PC stuck at X).

Since it seems that you didn't try to simulate the project with vivado, 
maybe there are issues that are not related to this project that are affecting the simulation.
I'll try to get them sorted by myself 👍  

I currently can't attach the JTAG to the rocket, and my board doesn't has an SD-card peripheral, 
(Hence, I'm writing my own bootrom), testing the connections and behaviour in simulation is quite important, 
since the board is quite big (and so are the implementation times...)

Thank you for your response.

eugene-tarassov, 2020-12-16:
It looks like the simulation fails because of a bug in the RocketChip.
Some registers in the core need explicit reset.
Missing reset causes Xs to spread into control logic of the core pipeline and the core is stalled.
FPGA is not affected, because all registers are reset when bitstream is loaded.
I'll try to create a patch.

Rucadi, 2020-12-16:
Thank you for your time, that would be awesome.

I'm able to simulate the latest chipyard (dev branch) rocket-chip (at least loading the bootrom works)
However I'm having trouble when loading a program from external memory on the sim (The data is being read correctly from the memory, but the instruction doesn't execute correctly). 

This means that maybe some of the "patches" are not necessary for a newer version of the Rocket, but still I'm not sure that the simulation works flawlessly.

eugene-tarassov, 2020-12-18:
On 10/9/2020 Andrew Waterman <andrew@sifive.com> submitted series of fixes for X problems in rocket-chip.
I'm going to upversion repo references to include those commits.
I'm running tests now. It looks good so far. I'll push the changes when testing is done.

eugene-tarassov, 2020-12-21:
I have updated submodule references. RocketChip simulation runs OK now.

closed 2020-12-21 by eugene-tarassov