eugene-tarassov/vivado-risc-v #14: What are the required steps to port to Alveo U2xx cards

gabaxter, 2021-02-05:
Hi, 

I would like to build larger systems and have access to Alveo cards.  Can you please tell us what we'd need to do to port to Alveo U2xx?  I am particularly interested in connecting to HBM on U280 to benchmark improvements.

Thank you for this excellent repo.  I hope it gets folded back in to chipyard proper as "F1" is *NOT* FPGA support.  Running a simulation is not the same as building the code for and executing on FPGAs. There are many places I've seen in the docs where that seems to get confused.  ;) 

Cheers,

Glenn

eugene-tarassov, 2021-02-06:
Alveo U2xx needs host-side simulation of disk, ethernet and uart - the board itself has no IO devices. As I understand, chipyard already has such simulation, because F1 is Alveo U2xx board installed somewhere in an Amazon data center. But I don't know all details. The goal of this repo is to provide HDL source code of a complete RISC-V SoC with all necessary IO devices instead of simulated IO in chipyard/firesim/F1.

gabaxter, 2021-02-06:
Hi Eugene,

U2xx has a number of I/O devices: (UG1289)
  • supports 2x 100GE via dual QSFP28s, 
  • supports 4x UART across USB using quad port FT4232HQ (two are really available for FPGA use), 
  • supports 4x DDR4 controllers (64GB total of 4x16GB)- and additional 16GB of HBM2 in the case of U280 
  • supports PCIe, including CCIX across Gen3x16 or Gen4x8. 
  • supports QSPI for both bitstream programming and data storage

It it possible to achieve disk in at least two ways - network attached storage, or, using RAMDISK which can be loaded from either PCIe, from QSPI, and even across serial.  It is also possible to use QSPI for disk, but that would be ultra slow.   I would personally use RAMDISK for initial filesystem and then NFS from there across network.  True this is not ideal, but it is quite functional.

There are a number of approaches to bring up Linux on Alveo cards.  Honestly, if we have just a serial port, that is enough because you can do all download, etc that way - which I've done a number of times.  It's ugly, but quite functional.  And it is the very first test I would do.  100GE is not easy and requires infrastructure, but those that have Alveo cards will generally have that.  And failing that Serial permits basic access.  It might also be possible to use the network across PCIe, though I've not tried that before.   The point is, there are ways to achieve network access, even if a bit uglier than on integrated boards.

I do not see what there is stopping Alveo from being used, and it would add enormous more capacity for testing BOOM  v3 multiple cores (which is my exact task right now).  _If you can elucidate the steps needed to add support, I'd be happy to do as much as I can to add that support here_.  I use Alveo cards frequently in the work I do and am very familiar with its infrastructure.  To be clear, I do *NOT* use any of Xilinx Alveo (SDAccel / XRT) setup - its just another FPGA board to me.

I fully support the mission you are after to have full SoC running on FPGA instead of sim on F1/Verilator.  This is the part that is missing from the Chipyard infrastructure.  It would be very good to bring your techniques into the chipyard repo.

Thank you for any help.

Cheers,

Glenn

eugene-tarassov, 2021-02-06:
> it would add enormous more capacity for testing BOOM v3 multiple cores

Yes. In fact, a while ago, I have assembled a machine with VCU1525 (the development version of Alveo U200), which I want to use for exact same purpose.

> To be clear, I do NOT use any of Xilinx Alveo (SDAccel / XRT) setup - its just another FPGA board to me.

Yes, I also see it as a relatively inexpensive way to get a big FPGA for experiments with large designs. I don't use SDAccel/XRT.

> if we have just a serial port, that is enough because you can do all download

There is a better way: Xilinx tools include RISC-V JTAG debugger. It can download, boot and debug Linux over USB-JTAG adapter. This is how I run regression tests for this repo.

> I would personally use RAMDISK for initial filesystem and then NFS from there across network.

Yes, it should work fine. The only problem:

> 100GE is not easy and requires infrastructure 

The main reason I'm still not using the board.

I could use some help with this. If this is solved, the rest should be easy. The board would act as diskless machine, which boots over JTAG or network, and uses NFS for file storage.

gabaxter, 2021-02-06:
My plan is very simple.  I would like a config that has a UART and MegaBOOM x2 + DDR.  I can generate a DCP for AXI DDR for one or all the machines - I've crafted a number of those for all U2xx sans AXI, so it is fairly simple to export the DCP for each controller.   I have not looked deeply into where you store configs or how you craft them.  But this base system will fully boot linux using RAMDISK.  We have 16GB per controller with what is on each DDR.  Could even craft it so we have all 4 in (84K LUT) and dedicate 2 of them to 32GB of RAMDISK.   That would be an insanely fast machine.

Where do we start?  :)

gabaxter, 2021-02-06:
I just thought of another 'cheap' way to get GigE,  actually quad GigE.   Using an cable, you can go from QSFP+ to SFP+ via a 4x cable (QSFP+ on one end, 4x SFP+ on other).  SFP+ is backward compatible to 1G, so in theory, we can use a DAC cable and get 4x ethernet ports into any switch which supports SFP+ and possibly SFP.  Even some consumer grade switches now offer that, so that could be a quick way to get Enet...  Thoughts?  I do not mind ugly if it gets things working.

eugene-tarassov, 2021-02-06:
> Where do we start? :)

I would start with the default RISC-V config (Rocket64b2 for Nexys Video):
```
source /opt/Xilinx/Vivado/2020.2/settings64.sh
make vivado-tcl
```
Or look into src/main/scala/rocket.scala to see how to add other configs.

Use Vivado to create a new project and block design for U200. 
Add  rocket.vhdl, srams.v and system-nexys-video.v from workspace/rocket64b2 directory to the project source files.
Add vhdl-wrapper/src/net/largest/riscv/vhdl/bscan2jtag.vhdl.
Add Rocket64b2 module to the block design.
The module will have wrong boot rom and wrong Linux device tree, but it should be OK for the first step.
Add clocks, UART, DDR, interconnect, constraints, etc. Make the project to work.
It should, at least, print "RISC-V 64, Boot ROM V3.1" on the UART console.

Next steps would be:
1. Ethernet adapter
2. new boot rom. One in the bootrom directory looks for SD card.
3. add more BOOM configs in src/main/scala/rocket.scala
4. new board files for U200 in the boards directory.
5. block design Tcl in the boards directory.
6. update Makefile

> I just thought of another 'cheap' way to get GigE...

Sounds good to me. I don't have any experience with QSFP+. If you could create an example design, it would very helpful.

gabaxter, 2021-02-06:
I defined au200, au250, au280 in Makefile.   I am working my way through crafting the tcl for bd gen.  I will pause that and do the above.  I think I get all of the above.  TY for the help.  My goal is to get a HW design through the tools.  

I do see one  issue which is that the present SW flow forces a 32-bit address instead of a 64-bit address.  As a result, DDR can not be 16GB supported by board.  Can we support this in the flow?

Happy to try QSFP later if I have the opportunity.  That is a long term update that will need some research, purchase of cables, etc.

eugene-tarassov, 2021-02-07:
> I am working my way through crafting the tcl for bd gen

I think it is easier to create and debug design in Vivado, then use write_bd_tcl command to write the Tcl file, then edit the file to make it usable in the Makefile.

> I do see one issue which is that the present SW flow forces a 32-bit address instead of a 64-bit address.

It should be possible, but I don't have an answer yet. It needs investigation.

gabaxter, 2021-02-07:
I'm actually taking the vc707 bd tcl and editing it directly.  I've done this a lot in the past, and it works well.   With that, I can then write it out and reuse it in your boards dir.   (can edit for various versions of Vivado post that).  This to me is the very fastest way to get to end goal.  I've switched back to default config per your suggestion as it goes through tools faster. ;).  I'll post what I end up with, if I get through tools.  Likely next day at this point as I have some other things to take care of, but stay tuned,  Thank you for all your help!

gabaxter, 2021-02-08:
Eugene, still working on this.  Have other priorities to focus on the rest of this week.   I will try to complete this next weekend.  I am currently stuck at it not finding the CPU cores, even though being added.  Something about manual compile order prevents files from being read in.  I stopped there but will get back to it.

eugene-tarassov, 2021-02-13:
I have added initial support for VCU1525 board. No disk, networking or boot ROM yet.
I have added config for 2-core 4-way Mega BOOM with 16GB memory - Rocket64z2m.
I can access it from the debugger, the cores look running OK.
HTH

gabaxter, 2021-02-15:
Hi Eugene,

Thank you.  I am going to test this out to see if I can build it.

Have you looked at UW Black Parrot?  What is interesting is that they have a purely PCIe attached way to access - no network no UART.  I’m still learning about it. But it might be worth you looking at too.  Maybe those techniques can solve a lot of issues with IP generation since all Xilinx devices of interest have XDMA in them.  With that a design only needs DDR and PCIe.  Interesting choices.

Cheers,

Glenn

eugene-tarassov, 2021-02-15:
Hi Glenn,

I have not seen UW Black Parrot before. 
It looks very much like RocketChip, except it is implemented in System Verilog instead of Chisel.

> What is interesting is that they have a purely PCIe attached way to access

My VCU1525 design does include PCIe, and, yes, with proper software, it can be used to simulate disk, network and other IO.
It also allows the host machine to read/write RISC-V memory and memory mapped registers.
I have not decided yet on how to use PCIe. I'll look into how UW Black Parrot does it.

Regards,
Eugene

closed 2021-04-21 by eugene-tarassov