eugene-tarassov/vivado-risc-v #16: compile error

qinaksd, 2021-03-23:
under the vivado2019.2 env, i run the follow command: make CONFIG=rocket64b2 BOARD=nexys-video bitstream
 ,then it occured errors...,

echo "open_project workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.xpr" >workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
echo "update_compile_order -fileset sources_1" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
echo "set_param general.maxThreads 4" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
echo "launch_runs impl_1 -to_step write_bitstream -jobs 4" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
echo "wait_on_run impl_1" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
echo "write_cfgmem -format mcs -interface SPIx4 -size 32 -loadbit \"up 0x0 workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/impl_1/riscv_wrapper.bit\" -file workspace/rocket64b2/nexys-video-riscv.mcs -force" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2019.2/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'riscv_io_axi_m_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_io_axi_m_0/riscv_io_axi_m_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_io_axi_m_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_io_axi_m_0/riscv_io_axi_m_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_io_axi_m_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_io_axi_m_0/riscv_io_axi_m_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_io_axi_m_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_io_axi_m_0/riscv_io_axi_m_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_io_axi_m_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_io_axi_m_0/riscv_io_axi_m_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_UART_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_UART_0/riscv_UART_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_UART_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_UART_0/riscv_UART_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_UART_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_UART_0/riscv_UART_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_UART_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_UART_0/riscv_UART_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_UART_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_UART_0/riscv_UART_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_RocketChip_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_RocketChip_0/riscv_RocketChip_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_clk_wiz_0_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_clk_wiz_0_0/riscv_clk_wiz_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_clk_wiz_0_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_clk_wiz_0_0/riscv_clk_wiz_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_clk_wiz_0_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_clk_wiz_0_0/riscv_clk_wiz_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_clk_wiz_0_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_clk_wiz_0_0/riscv_clk_wiz_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_clk_wiz_0_0' generated file not found '/home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_clk_wiz_0_0/riscv_clk_wiz_0_0_sim_netlist.vhdl'. Please regenerate to continue.
open_project: Time (s): cpu = 00:00:24 ; elapsed = 00:00:52 . Memory (MB): peak = 1601.152 ; gain = 105.641 ; free physical = 1697 ; free virtual = 4091
update_compile_order: Time (s): cpu = 00:00:20 ; elapsed = 00:00:36 . Memory (MB): peak = 1697.199 ; gain = 96.047 ; free physical = 1694 ; free virtual = 4088
INFO: [IP_Flow 19-5642] Done with IP cache export for multiple IPs
[Tue Mar 23 00:13:38 2021] Launched riscv_io_axi_m_0_synth_1, riscv_UART_0_synth_1, riscv_RocketChip_0_synth_1, riscv_clk_wiz_0_0_synth_1, synth_1...
Run output will be captured here:
riscv_io_axi_m_0_synth_1: /home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/riscv_io_axi_m_0_synth_1/runme.log
riscv_UART_0_synth_1: /home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/riscv_UART_0_synth_1/runme.log
riscv_RocketChip_0_synth_1: /home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/riscv_RocketChip_0_synth_1/runme.log
riscv_clk_wiz_0_0_synth_1: /home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/riscv_clk_wiz_0_0_synth_1/runme.log
synth_1: /home/risc/workspace/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/synth_1/runme.log
[Tue Mar 23 00:13:39 2021] Launched impl_1...
[... cut ...]
write_cfgmem: Time (s): cpu = 00:00:00 ; elapsed = 00:00:45 . Memory (MB): peak = 1712.203 ; gain = 0.000 ; free physical = 645 ; free virtual = 1172
ERROR: [Common 17-39] 'write_cfgmem' failed due to earlier errors.

    while executing
"write_cfgmem -format mcs -interface SPIx4 -size 32 -loadbit "up 0x0 workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/impl_1/riscv_..."
    (file "workspace/rocket64b2/vivado-nexys-video-riscv/make-bitstream.tcl" line 6)
make: *** [Makefile:300: workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/impl_1/riscv_wrapper.bit] Error 1

eugene-tarassov, 2021-03-23:
Vivado fails like this when it runs out of memory. It needs a machine with, at least, 16GB RAM. Even more for bigger projects, up to 128GB for largest FPGAs.

qinaksd, 2021-03-31:
hi eugene-tarassov,
   after add 16G RAM,it generate a *.mcs success. i used to 32G sdcard ,it seems no message with the serial output。。。

eugene-tarassov, 2021-03-31:
> no message with the serial output

How do you program FPGA?
Is DONE LED on?
Is serial port cable connected?
What terminal emulator do you use?
How do you select which serial port to open in the emulator?
Is the port configured to 115200, 8 bit, no parity, 1 bit stop, no flow control?

If the cable is connected correctly, when FPGA program is loaded, you should see "RISC-V 64, Boot ROM V3.2" message on the terminal.

qinaksd, 2021-04-01:
hi ,
 yeah, i program FPGA, boot from configuration memory Device, the DONE LED light on.
 i connect uart with xcom 115200 ,there is no message output.
the vivado logs is follow:
boot_hw_device  [lindex [get_hw_devices xc7a200t_0] 0]
INFO: [Labtoolstcl 44-664] Will wait up to 180 seconds for booting to complete.
INFO: [Labtools 27-32] Done pin status: HIGH
refresh_hw_device [lindex [get_hw_devices xc7a200t_0] 0]
INFO: [Labtools 27-1434] Device xc7a200t (JTAG device index = 0) is programmed with a design that has no supported debug core(s) in it.

i dont know what wrong with it?

qinaksd, 2021-04-02:
> I have automated tests running on Nexys Video board daily - I'm pretty sure it works fine.
> I'm not familiar with xcom - that would be my first suspect.
> I suggest to check serial port output like this:
>
> sudo stty -F /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A904DPTC-if00-port0 sane
> sudo stty -F /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A904DPTC-if00-port0 115200 -icrnl -ocrnl igncr -echo
> sudo cat /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A904DPTC-if00-port0
>
> Replace UART_A904DPTC with proper ID for your board.

hi 
 i got  errors as following:
ERROR: [Xicom 50-38] xicom: Device:0, user chain number:1, slave index:0. Reading intermittently wrong data from core. Try slower target speed. Make sure design meets timing requirements.
ERROR: [Xicom 50-38] xicom: Device:0, user chain number:1, slave index:0, is not a valid CseXsdb Slave core.
ERROR: [Labtools 27-3176] hw_server failed during internal command.
Resolution: Check that the hw_server is running and the hardware connectivity to the target.

is it the oscillator？

eugene-tarassov, 2021-04-02:
Yes, I think it means no clock.

qinaksd, 2021-04-02:
how to check this clock? which pin?

qinaksd, 2021-04-02:
i change the oscillator into 50Mhz, but the uart had no output

eugene-tarassov, 2021-04-02:
What board are you using?
According to the log, you build for Nexys Video, but it uses 100MHz oscillator.

closed 2021-04-21 by eugene-tarassov