eugene-tarassov/vivado-risc-v #19: Question for Boom version

Shuiliusheng, 2021-04-22:
Hi, can I use boom v2 or boom v1 code for this project and run it on fpga? Thank you so much.

eugene-tarassov, 2021-04-22:
I was not able to get BOOM v1 or v2 to work. For me, v3 was first usable version.

In a perfect world, it would be easy to switch to BOOM v2:
```
cd generators/riscv-boom
git checkout 2.2.3
cd -

```
Then run make as usual.
But it does not work, BOOM has a lot of dependencies, you need to know and select right version of each dependency, including right versions of tools, like Scala and Chisel.

wkzhng, 2021-04-28:
> I was not able to get BOOM v1 or v2 to work. For me, v3 was first usable version.
>
> In a perfect world, it would be easy to switch to BOOM v2:
>
> ```
> cd generators/riscv-boom
> git checkout 2.2.3
> cd - [...]

I think I have used right versions of tools, but there are still some errors about the Generator. 
`[error] (run-main-0) java.lang.Exception: Unable to find part "Vivado.RocketSystem.-C" from "ArrayBuffer(Vivado.RocketSystem.-C)", did you misspell it?`

eugene-tarassov, 2021-04-29:
I don't know how to fix such build errors.
It is a hard problem to find a coherent combination of versions of everything involved into BOOM v2 build.
If you find a solution, please, post it here.

closed 2021-05-28 by eugene-tarassov