eugene-tarassov/vivado-risc-v #20: Running Gemmini on Nexys4-DDR

nice-000, 2021-04-25:
hi, I'm trying to run a Gemmini on Nexys-4DDR(Nexys A7-100T)，first I run the following command ：make CONFIG=rocket64b1gem BOARD=nexys-a7-100t bitstream, then it occurred errors：
[error] (run-main-0) java.lang.Exception: Unable to find part "Vivado.Rocket64b1gem" from "ArrayBuffer(Vivado.Rocket64b1gem)", did you misspell it or specify the wrong package path?
secondly, I used the parameter rocket64b2gem，it occurred：
ERROR: [DRC UTLZ-1] Resource utilization: LUT as Logic over-utilized in Top Level Design (This design requires more LUT as Logic cells than are available in the target device. This design requires 136675 of such cell types but only 63400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.)
I am wondering if it is possible to run one rocket core with one Gemmini part on my board? Thanks!

eugene-tarassov, 2021-04-25:
A7-100T FPGA is too small for Gemmini.
For meaningful experiments you need a better board, like Digilent Genesys 2.
Even then, Gemmini FPGA emulation is incredibly slow.
I'm not sure why, but Gemmini does not fit well into FPGA.

nice-000, 2021-04-27:
All right！ I will change a better board and try again. Thank you for your reply！

closed 2021-05-28 by eugene-tarassov