eugene-tarassov/vivado-risc-v #21: Execute the command "make update-submodules" failed.....Someone has  the same problems?

lpppeipei, 2021-04-27:


lpppeipei, 2021-04-27:
ff520@ubuntu:~/Desktop/vivado-risc-v$ make update-submodules
git -c submodule.torture.update=none -c submodule.software/gemmini-rocc-tests.update=none -c submodule.software/onnxruntime-riscv.update=none submodule update --init --force --recursive
Submodule 'ethernet/verilog-ethernet' (https://github.com/alexforencich/verilog-ethernet.git) registered for path 'ethernet/verilog-ethernet'
Submodule 'generators/gemmini' (https://github.com/ucb-bar/gemmini) registered for path 'generators/gemmini'
Submodule 'generators/riscv-boom' (https://github.com/riscv-boom/riscv-boom.git) registered for path 'generators/riscv-boom'
Submodule 'generators/sifive-cache' (https://github.com/sifive/block-inclusivecache-sifive.git) registered for path 'generators/sifive-cache'
Submodule 'generators/testchipip' (https://github.com/ucb-bar/testchipip.git) registered for path 'generators/testchipip'
Submodule 'linux-stable' (git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git) registered for path 'linux-stable'
Submodule 'riscv-pk' (https://github.com/riscv/riscv-pk.git) registered for path 'riscv-pk'
Submodule 'rocket-chip' (https://github.com/ucb-bar/rocket-chip.git) registered for path 'rocket-chip'
Submodule 'u-boot' (https://github.com/u-boot/u-boot.git) registered for path 'u-boot'
Cloning into '/home/ff520/Desktop/vivado-risc-v/ethernet/verilog-ethernet'...
Cloning into '/home/ff520/Desktop/vivado-risc-v/generators/gemmini'...
Cloning into '/home/ff520/Desktop/vivado-risc-v/generators/riscv-boom'...
Cloning into '/home/ff520/Desktop/vivado-risc-v/generators/sifive-cache'...
Cloning into '/home/ff520/Desktop/vivado-risc-v/generators/testchipip'...
Cloning into '/home/ff520/Desktop/vivado-risc-v/linux-stable'...

then stopped,never go on

eugene-tarassov, 2021-04-27:
Yes, git.kernel.org servers seem slow recently.
Cloning Linux code can take some time.

lpppeipei, 2021-04-27:
> Yes, git.kernel.org servers seem slow recently.
> Cloning Linux code can take some time.

It's OK now.Thanks.

closed 2021-04-27 by lpppeipei