eugene-tarassov/vivado-risc-v #22: Running "make CONFIG=rocket64b2 BOARD=nexys-a7-100t bitstream" failed. As follows:

lpppeipei, 2021-04-27:
[error] lmcoursier.internal.shaded.coursier.error.FetchError$DownloadingArtifacts: Error fetching artifacts:
[error] https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-sriracha/scala_2.12/sbt_1.0/0.1.0/docs/sbt-sriracha-javadoc.jar: download error: Caught javax.net.ssl.SSLHandshakeException: Remote host terminated the handshake (Remote host terminated the handshake) while downloading https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-sriracha/scala_2.12/sbt_1.0/0.1.0/docs/sbt-sriracha-javadoc.jar
......
Killed
make: *** [Makefile:211: workspace/rocket64b2/system.dts] Error 137

eugene-tarassov, 2021-04-27:
RocketChip uses SBT (Scala Build Tool).
SBT downloads its plugins from external repository at scala-sbt.org.
If the download fails, just try again.

BTW, CONFIG=rocket64b2 (2 64-bit cores) is too big for nexys-a7-100t.
The board can fit only one core: CONFIG=rocket64b1
See [rocket.scala](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/src/main/scala/rocket.scala) for available configs.

lpppeipei, 2021-04-28:
> RocketChip uses SBT (Scala Build Tool).
> SBT downloads its plugins from external repository at scala-sbt.org.
> If the download fails, just try again.
>
> BTW, CONFIG=rocket64b2 (2 64-bit cores) is too [...]

Thanks.I have tried  "CONFIG=rocket64b1 BOARD=nexys-a7-100t bitstream",but it still doesn't work. It shows"
killed
make: *** [Makefile:211: workspace/rocket64b1/system.dts] Error 137

eugene-tarassov, 2021-04-28:
Is it still failing downloading sbt-sriracha-javadoc.jar?
Or different error?

I tried "make CONFIG=rocket64b1 BOARD=nexys-a7-100t bitstream" - it works fine for me,  including download of sbt-sriracha-javadoc.jar.

lpppeipei, 2021-04-29:
> Is it still failing downloading sbt-sriracha-javadoc.jar?
> Or different error?
> 
> I tried "make CONFIG=rocket64b1 BOARD=nexys-a7-100t bitstream" - it works fine for me, including download of sbt-sriracha-javadoc.jar.

I tried again,it's OK now. Thanks.

closed 2021-04-29 by lpppeipei