eugene-tarassov/vivado-risc-v #23: I used a 8G sd card,when run " ./mk-sd-card",it shows"gzip: stdin: not in gzip format tar: Child returned status 1 tar: Error is not recoverable: exiting now"

lpppeipei, 2021-04-29:


lpppeipei, 2021-04-29:
As follows:
root@ubuntu:/home/ff520/Desktop/vivado-risc-v# ./mk-sd-card
make: 'debian-riscv64/initrd' is up to date.
make: 'debian-riscv64/rootfs.tar.gz' is up to date.
1500+0 records in
1500+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 0.638453 s, 2.5 GB/s
SD image device: /dev/loop8
Checking that no-one is using this disk right now ... OK

Disk /dev/loop8: 1.48 GiB, 1572864000 bytes, 3072000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0xabd7f184.
/dev/loop8p1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 400 MiB.
/dev/loop8p2: Created a new partition 2 of type 'Linux' and of size 1.1 GiB.
/dev/loop8p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0xabd7f184

Device       Boot  Start     End Sectors  Size Id Type
/dev/loop8p1 *      2048  821247  819200  400M  e W95 FAT16 (LBA)
/dev/loop8p2      821248 3071999 2250752  1.1G 83 Linux

The partition table has been altered.
mkfs.fat 4.1 (2017-01-24)
mke2fs 1.45.5 (07-Jan-2020)
Creating filesystem with 281344 4k blocks and 70416 inodes
Filesystem UUID: 68d82fa1-1bb5-435f-a5e3-862176586eec
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

/home/ff520/Desktop/vivado-risc-v/debian-riscv64/rootfs /home/ff520/Desktop/vivado-risc-v

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

eugene-tarassov, 2021-04-29:
It looks like debian-riscv64/rootfs.tar.gz is corrupted.
The file is downloaded from github.com, bad internet connection can result in corrupted file.
Try "rm debian-riscv64/*" and run ./mk-sd-card again.

lpppeipei, 2021-04-30:
> It looks like debian-riscv64/rootfs.tar.gz is corrupted.
> The file is downloaded from github.com, bad internet connection can result in corrupted file.
> Try "rm debian-riscv64/*" and run ./mk-sd-card again.

Another problem:
Device     Boot Start       End   Sectors   Size Id Type
/dev/sdb1       32768 243597311 243564544 116.1G  7 HPFS/NTFS/exFAT
Are you sure? [y/N] y
1311637504 bytes (1.3 GB, 1.2 GiB) copied, 3 s, 437 MB/s
24000+0 records in
24000+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 3.94512 s, 399 MB/s
Error: Partition(s) 1, 2 on /dev/sdb have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

lpppeipei, 2021-04-30:
I run again:
root@ubuntu:/home/ff520/Desktop/vivado-risc-v# rm -r debian-riscv64
root@ubuntu:/home/ff520/Desktop/vivado-risc-v# ./mk-sd-card
It shows:
Device     Boot  Start     End Sectors  Size Id Type
/dev/sdb1  *      2048  821247  819200  400M  e W95 FAT16 (LBA)
/dev/sdb2       821248 3071999 2250752  1.1G 83 Linux
Are you sure? [y/N] y
24000+0 records in
24000+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 0.488363 s, 3.2 GB/s
Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

sfdisk: Use the --force flag to overrule all checks.

It still doesn't work. Thanks.

eugene-tarassov, 2021-04-30:
In both cases it says "disk is in use". In Linux terms, it is "mounted". The script will not work on a mounted disk.
So, don't mount the disk, or unmount it, or reboot to get rid of unwanted mounts.
I'll look into improving the script for better handling of a mounted disk.

lpppeipei, 2021-05-06:
> In both cases it says "disk is in use". In Linux terms, it is "mounted". The script will not work on a mounted disk.
> So, don't mount the disk, or unmount it, or reboot to get rid of unwanted mounts.
> I'll look into improving the script for better handling of a mounted disk.

Could you  tell me about how  to " don't mount the disk,or unmount it, or reboot to get rid of unwanted mounts." ? It seems I only  use my SD Card Reader  which is an USB  reader. Thanks.

lpppeipei, 2021-05-06:
> Could you tell me about how to " don't mount the disk,or unmount it, or reboot to get rid of unwanted mounts." ? It seems I only use my SD Card Reader which is an USB reader. Thanks.

Later I  did some work on it ,it shows as bellow,it is OK  now ?  What will it display after run ./mk-sd-card?

Are you sure? [y/N] y
24000+0 records in
24000+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 333.629 s, 4.7 MB/s
Checking that no-one is using this disk right now ... OK

Disk /dev/sdb: 7.43 GiB, 7969177600 bytes, 15564800 sectors
Disk model: SD Card Reader  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xb0db87c8

Old situation:

Device     Boot  Start     End Sectors  Size Id Type
/dev/sdb1  *      2048  821247  819200  400M  e W95 FAT16 (LBA)
/dev/sdb2       821248 3071999 2250752  1.1G 83 Linux

/dev/sdb2: 
New situation:
Disklabel type: dos
Disk identifier: 0xb0db87c8

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sdb1  *      2048   821247   819200  400M  e W95 FAT16 (LBA)
/dev/sdb2       821248 15564799 14743552    7G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
fsck from util-linux 2.34
e2fsck 1.45.5 (07-Jan-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 11461/70416 files (0.0% non-contiguous), 168317/281344 blocks
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/sdb2 to 1842944 (4k) blocks.
The filesystem on /dev/sdb2 is now 1842944 (4k) blocks long.

closed 2021-05-12 by lpppeipei