eugene-tarassov/vivado-risc-v #24: Porting to KC705 FPGA board

TwistsOfFate, 2021-05-07:
Thank you for creating the project. The bitstream generation flow was extremely convenient for mapping a RISC-V SoC to FPGA.

I was wondering how much effort would be needed to remap this flow (or at least the `rocket64b2gem` config) onto a Xilinx KC705 board.

I suppose I should use `genesys2` as a base config and make some modifications to adapt to KC705 as they have the same FPGA chip (xc7k325tffg900-2). I understand that the .xdc constraint files under /board/genesys2 need to be changed to that of the KC705 board. Aside from that can you please guide me on what other factors would I have to consider?

Thank you

eugene-tarassov, 2021-05-07:
It also requires substantial changes in the IO block:
1. Ethernet PHY: change interface from RGMII to GMII, re-calibrate input/output delays
2. SD card: remove reset signal. Optionally, add write protect signal
3. UART: add CTSn/RTSn signals
4. remove fan control

TwistsOfFate, 2021-05-09:
Thanks. I'll look into it.

eugene-tarassov, 2021-05-29:
KC705 port passed all tests, closing this issue.

closed 2021-05-29 by eugene-tarassov