eugene-tarassov/vivado-risc-v #26: Issue with UART

suryacharanp, 2021-05-07:
I connected UART controller to another riscv core in the same way it is connected to rocket core. I am using uart controller for read and write into an external device from user space. I used fpga-axi-uart.c as KMD. I am trying a simple loop back application on it.  I attached an led to the TxD of uart, it is not blinking when I am writing into the registers instead it is continuously staying high whether I write anything or not. stty -F /dev/ttyAU0 shows that data is written but it is showing that there is no data in RxD even though I connected TxD and RxD. I also connected RTS to CTS. I am not getting what is the problem here, please help me.

eugene-tarassov, 2021-05-07:
> I also connected RTS to CTS.

This makes TX to stop if RX buffer is full.
Try connecting CTSn to 0.

suryacharanp, 2021-05-10:
Thank you sir, It worked.

closed 2021-05-10 by suryacharanp