eugene-tarassov/vivado-risc-v #27: No UART output for VC707

eecwng, 2021-05-11:
I have programmed the 8 cores bit file to VC707 FPGA board with programmed SD card. There is no output from the UART.  However, I have done the same thing using Genesys 2 FPGA board (4 cores bit file). Everything is ok. Linux can be booted. So, how can I fix the problems on VC707 FPGA ? Thanks a lot.

eecwng, 2021-05-11:
My VC707 is a new board. Just received one week ago. I only use VC707 for this project.

eugene-tarassov, 2021-05-11:
Then you need to debug it.
I would start with launching xsdb and typing connect and ta. If FPGA is OK, you should see RISC-V cores running:

```
****** Xilinx System Debugger (XSDB) v2020.2
  **** Build date : Nov 18 2020-10:01:48
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

xsdb%
xsdb% connect
tcfchan#0
xsdb% ta
  1  xc7k325t
     2  RISC-V
        3  Hart #0 (Running)
        4  Hart #1 (Running)
xsdb%

```

eecwng, 2021-05-26:
I use xsdb to debug my vc707 FPGA. Here is the output. What is the next step ? Thanks.

[image]

eugene-tarassov, 2021-05-26:
You can read/write UART registers to see if it works as expected:
```
xsdb% target -set -filter {name =~ "RISC-V"}
xsdb% mrd 0x60010000 4
60010000:   00000000
60010004:   00000000
60010008:   00000014
6001000C:   00000010
xsdb% mwr 0x60010004 0x41
xsdb%

```
The write command `mwr 0x60010004 0x41` writes TX FIFO register, you should see letter 'A' on the terminal.
The UART register definitions can be found in the driver code: https://github.com/eugene-tarassov/vivado-risc-v/blob/master/patches/fpga-axi-uart.c

Use `help` command to see XSDB commands description:
```
xsdb% help
Available Help Categories

breakpoints   - Target Breakpoints/Watchpoints.
connections   - Target Connection Management.
device        - Device Configuration System.
download      - Target Download FPGA/BINARY.
ipi           - IPI commands to Versal PMC.
jtag          - JTAG Access.
memory        - Target Memory.
miscellaneous - Miscellaneous.
registers     - Target Registers.
reset         - Target Reset.
running       - Program Execution.
streams       - Jtag UART.
svf           - SVF Operations.
tfile         - Target File System.

Type "help" followed by above "category" for more details or
help" followed by the keyword "commands" to list all the commands

xsdb% help mrd
NAME
    mrd - Memory Read.

SYNOPSIS
    mrd [options] <address> [num]
        Read <num> data values from the active target's memory address specified
        by <address>.
...
```

eecwng, 2021-05-27:
Thanks for your reply. Here is the output of UARTLITE registers. The Tx FIFO is always "Full".

[image]

eecwng, 2021-05-27:
Finally, I can boot up linux with 8 RISC-V cores using vc707. The output from UART shown in the figure.

[image]

The issue is due to the CTS pin of the Xilinx UARTLITE. As I cannot control the CTS of UART externally, I modify the verilog code uart.v as shown in following figure. Comment out line 169 and modify as line 170. Then the UART can be worked normally.
[image]

eugene-tarassov, 2021-05-27:
I'd like to understand the root cause of  nonworking CTS.
I suspect it has something to do with the terminal emulation software that you use.
What software is it?

I usually use Tera Term on Windows, it works very well, including VC707 and other boards that support CTS:
[image]

eecwng, 2021-05-28:
My software is USR-TCP232 as shown in the figure. It is developed by a Chinese taobao company. It has dual functions. One is the COM port data and another is the network data. For the COM port function, there is no hardware flow control in the setting. So this software cannot control the CTS pin of the UART module of the RISC-V core as I had mentioned in previous post. So I removed the function of CTS in uart.v so that I can use this software. I prefer to use this software rather than Tera Term since USR-TCP232 can separate the input and output of the COM data in the screen. Also, in the README, there is no mention about the specification of the UART. So hardware flow control like CTS is ignored in the normal condition.

[image]

eugene-tarassov, 2021-05-28:
I tried USR-TCP232-Test and I see couple of problems:

1. The Linux UART console is meant to be used with a terminal, but USR-TCP232-Test is not a terminal emulation.
I mean it does not support standard terminal commands, like [ANSI escape codes](https://en.wikipedia.org/wiki/ANSI_escape_code), so most of the console functionality will not work.
2. It appears to be not compatible with the CP2103GM USB-to-UART bridge on VC707 board.
And it does not have to be, it was developed for testing specific hardware - USR-TCP232.

> For the COM port function, there is no hardware flow control in the setting.

This by itself is not a problem. When hardware flow control is disabled, CTS stays ON and TX sends characters without waiting until other side is ready.

I recommend to try a terminal emulation software. VC707 User Guide suggests TeraTerm. Another popular option is PuTTY. Both work fine with VC707, with or without hardware flow control.

closed 2021-07-07 by eugene-tarassov