eugene-tarassov/vivado-risc-v #28: Other FPGAs

Ammadali90, 2021-05-25:
Is it possible to use these implementations of ZCU102?
if yes, then what are the changes required to make it run.

Thanks in advance!

eugene-tarassov, 2021-05-25:
It is possible, but it is Zynq board. The large portion of the chip is taken by ARM CPU, which is not needed for RISC-V emulation.
As result, FPGA is rather small, 274K LUTs - not enough to work with large designs like BOOM or Gemmini. Using pure FPGA board allows much cleaner and more powerful RISC-V implementations.

> if yes, then what are the changes required to make it run

The board specific files are in the `board` directory. They will need changes/additions to add support for ZCU102. I cannot tell exact changes - it is not a simple task.

Ammadali90, 2021-05-25:
Thanks for your reply!

xiongwu12345, 2021-09-02:
> As result, FPGA is rather small, 274K LUTs - not enough to work with large designs like BOOM or Gemmini.

But I find that there are 600K LUTs on the website.

[image]

eugene-tarassov, 2021-09-02:
System Logic Cells are not same as LUTs (Look Up Tables). FPGAs consist of LUTs, while System Cells is a common measure of ASIC design complexity. Xilinx marketing uses some rather arbitrary multiplier to convert actual LUT count to some abstract System Logic Cells number. You can find the real LUT count in the Zynq UltraScale+ MPSoC Data Sheet.

xiongwu12345, 2021-09-03:
Yes, I see. Thanks.

closed 2022-06-26 by eugene-tarassov