eugene-tarassov/vivado-risc-v #30: 16 or more RISC-V cores

eecwng, 2021-05-31:
If I want to build 16 or more RISC-V cores, which FPGA board is suitable for this task ?
I have an used new VCU129-PP FPGA board, is it ok ? Many thanks.

eugene-tarassov, 2021-05-31:
Big 64-bit single issue RISC-V core (not BOOM) is about 28,000 LUTs.
16 cores + IO logic - about 500,000 LUTs.
VCU129-PP FPGA board has 1,728,000 LUTs - should be OK for up to 50-60 cores.
Most Virtex UltraScale+ boards should be OK for 16+ cores.

It looks like the board does not have SD card slot, but you can use [PMOD SD card adapter](https://store.digilentinc.com/pmod-sd-full-sized-sd-card-slot/).

eecwng, 2021-06-07:
I have purchased the PMOD SD card adapter as shown in the photo.
[image]
What is the next step to implement the 16 cores RISC-V in the VCU129-PP FPGA board ? Thanks.

eugene-tarassov, 2021-06-07:
I have added 64b16m, 64b24m and 64b32m configurations - 16, 24 and 32 cores with up to 256GB RAM.
I tested them on VCU1525 - seems working OK.
Now you need to port the design to VCU129-PP. VC707 is, probably, the best starting point.
I cannot do that - I don't have VCU129-PP board.

eecwng, 2021-06-08:
I can port the design to VCU129-PP. BTW, would you give me more guidance on the porting ? Also, which files should I modify ?
Many thanks.

eugene-tarassov, 2021-06-08:
You need to add board/vcu129-pp directory with board support files.
Use other board directories as example, e.g. board/vc707.
All files in the directory need to be changed/reworked for a new board.

eecwng, 2021-06-15:
In riscv-2020.2.tcl file, there is a line "set mig_7series_0 [ create_bd_cell -type ip -vlnv xilinx.com:ip:mig_7series:4.2 mig_7series_0 ]".
The FPGA of VC707 is Virtex 7 while the FPGA of vcu129-pp is Virtex UltraScale+. What is the name of mig ip for Virtex UltraScale+ ? Thanks.

eugene-tarassov, 2021-06-15:
I believe the IP name is xilinx.com:ip:ddr4:2.2.

I think the easiest way to produce vcu129 Tcl is to generate Vivado project using vc707 Tcl, open it in the Vivado GUI, change board to vcu129 in the project settings,  replace memory controller using the Board tab,  make other fixes as needed, then use write_bd_tcl command to see new Tcl.

eecwng, 2021-06-16:
Thanks for your support. However, I am not familiar Vivado using TCL script. In the tcl file "riscv-20202.2.tcl", it mentioned the following

"The design that will be created by this Tcl script contains the following
module references:
$rocket_module_name, synchronizer, uart

Please add the sources of those modules before sourcing this Tcl script."

So how can I sources those modules ? Thanks.

eugene-tarassov, 2021-06-16:
All Vivado functionality is available as Tcl commands. This allows all kinds of automation in Vivado.
"riscv-20202.2.tcl" is Tcl script, which creates RISC-V Block Design. It adds IPs blocks and connects them.
Everything the script does can be done manually in the GUI and vice versa.
Use of scripts is not required, but it is the best way to store Vivado designs in a source control system like git.

You can generate scripts for particular board and RocketChip configuration by running
```
make CONFIG=rocket64b2 BOARD=vc707 vivado-tcl
```
The command uses "riscv-20202.2.tcl" as kind of template for board specific Tcl.
See Makefile for details.

You can re-create Vivado project by running the generated scripts in the Vivado Tcl Console:
```
source workspace/rocket64b2/system-vc707.tcl
```

To add support for a new board, you need to craft a new "riscv-20202.2.tcl".
One way to create the script is to edit a design in the GUI, then save is as a script using write_bd_tcl command.
All "riscv-20202.2.tcl" files in this repo are created this way, then manually edited to make them a bit more generic.
You can use any Vivado or Tcl commands in the script.
Use other board Tcl scripts as examples.
See Vivado User Guide for details.

eecwng, 2021-07-26:
I have loaded the tcl file of vc707 to Vivado and re-design the schematic for the VCU129-PP FPGA board. Then I run the "SYNTHESIS" and "IMPLEMENTATION" of Vivado. However, I got a problem for the IP 1G/2.5G Ethernet PCS/PMA or SGMII(16.2). The error message is shown in the attached photo. Please help to solve this problem.
[image]

eugene-tarassov, 2021-07-26:
It looks like Ethernet PCS/PMA is not configured properly.
In case of VC707 board, the configuration comes from the board files in Vivado.
VCU129 board files, probably, also have pre-defined configuration for Ethernet PCS/PMA interface.
So, if you setup your design to use VCU129 board files, you can select which interface to use (SGMII, SFP, etc.), and it should just work.

xiongwu12345, 2021-11-13:
Hi, how to config the input clock of the ' xilinx.com:ip:ddr4:2.2.'? 

We use a 250MHz clock generated by a clock wizard. And the error says:

[image]

Thanks

xiongwu12345, 2021-11-13:
Hi, we have migrated the Rocket to VCU118 manually. We use PMOD to connect a MicroSD card and we have checked the pin constrain carefully. But when boot the Linux, it says:

[image]

Do you have any idea about this situation?

eugene-tarassov, 2021-11-13:
> Hi, how to config the input clock of the ' xilinx.com:ip:ddr4:2.2.'?

The VCU118 board has an 250 MHz oscillator (U14), which generates DDR4 input clocks: 250MHZ_CLK1_P/N and 250MHZ_CLK2_P/N.

The best and easiest way to config DDR4 is to click "Connect Board Component..." in "Board" page in Vivado Block Design.

> We use PMOD to connect a MicroSD card...

I don't know why the card is not responding. Can be dirty or bent contacts, or a mistake in XDC, or non-working SD reset signal, etc.

cheungivan, 2022-09-15:
@eecwng i got same problem as un-route issue in VCU129.. Do you have solution yet?

Ziggy644, 2022-10-18:
How do you exactly make a configuration for a desired amount of CPU cores? Maybe I'm overlooking something?

Yuxin-Yu, 2024-05-27:
Hi @xiaolingyi , can I refer to your VCU118 configuration? My uart doesn't even display

ipapal, 2025-03-27:
Hello, can you give me some information on how to assign the current SD card pinouts to the PMOD SD card adapter?

ipapal, 2025-03-29:
Hi, do you have any updates regarding the porting on VCU118 ?