eugene-tarassov/vivado-risc-v #31: Bootrom on PCIe cards

rvccx, 2021-06-03:
How can I boot linux and run programs on PCIe-based accelerator cards, like Alveo U200 and VCU1525?
And if possible, can I build a larger amount of cores on them?
I changed the amount of cores to 50 it failed to run synthesis

eugene-tarassov, 2021-06-03:
I tried 32-core system on VCU1525 and it worked OK.
FPGA utilization is 80%, CPU clock 100MHz.
50 cores are, probably, too many for this board.

I'm going to add 16, 32, 48 and 64 core configurations to rocket.scala.

> How can I boot linux and run programs on PCIe-based accelerator cards, like Alveo U200 and VCU1525?

I don't have bootrom or drivers for accelerator cards yet.
I have debugger script that downloads Linux kernel and ramdisk.
It is enough to verify that Linux can boot, but not much more than that.

rvccx, 2021-06-03:
Thank you! Is there a way to run codes without booting Linux (or simulation in vivado)? Or it has to be run in assembly?

rvccx, 2021-06-03:
Thank you! Is there a way to run codes without booting Linux? For RTL sim is it possible to run codes in vivado IDE or something like a kernel is required?

eugene-tarassov, 2021-06-03:
You can download and run any code using Xilinx debugger (xsdb):

```
****** Xilinx System Debugger (XSDB) v2020.2
  **** Build date : Nov 18 2020-10:01:48
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

xsdb% connect
tcfchan#0
xsdb% target
  1  xcvu9p
xsdb% target 1
xsdb% cd C:/riscv/risc-v
xsdb% fpga workspace/rocket64b32m/vivado-vcu1525-riscv/vcu1525-riscv.runs/impl_1/riscv_wrapper.bit
100%   63MB   1.1MB/s  00:55
xsdb% target
  1* xcvu9p
    35  Legacy Debug Hub
     2  RISC-V
        3  Hart #0 (Running)
        4  Hart #1 (Running)
        5  Hart #2 (Running)
        6  Hart #3 (Running)
        7  Hart #4 (Running)
        8  Hart #5 (Running)
        9  Hart #6 (Running)
       10  Hart #7 (Running)
       11  Hart #8 (Running)
       12  Hart #9 (Running)
       13  Hart #10 (Running)
       14  Hart #11 (Running)
       15  Hart #12 (Running)
       16  Hart #13 (Running)
       17  Hart #14 (Running)
       18  Hart #15 (Running)
       19  Hart #16 (Running)
       20  Hart #17 (Running)
       21  Hart #18 (Running)
       22  Hart #19 (Running)
       23  Hart #20 (Running)
       24  Hart #21 (Running)
       25  Hart #22 (Running)
       26  Hart #23 (Running)
       27  Hart #24 (Running)
       28  Hart #25 (Running)
       29  Hart #26 (Running)
       30  Hart #27 (Running)
       31  Hart #28 (Running)
       32  Hart #29 (Running)
       33  Hart #30 (Running)
       34  Hart #31 (Running)
xsdb% target 3
xsdb% stop
Info: Hart #0 (target 3) Stopped at 0x153fe (Suspended)
xsdb% dow workspace/boot.elf
Downloading Program -- C:/riscv/risc-v/workspace/boot.elf
        section, .text: 0x80000000 - 0x8000e18f
        section, .rodata: 0x8000f000 - 0x800105ef
        section, .dynstr: 0x800105f0 - 0x800107c8
        section, .gnu.hash: 0x800107d0 - 0x800108a3
        section, .data: 0x80011000 - 0x80011937
        section, .dynamic: 0x80011938 - 0x80011a47
        section, .got: 0x80011a48 - 0x80011b0f
        section, .got.plt: 0x80011b10 - 0x80011b1f
        section, .htif: 0x80011b20 - 0x80011b2f
        section, .dynsym: 0x80011b30 - 0x80011d9f
        section, .rela.dyn: 0x80011da0 - 0x80012b97
        section, .bss: 0x80013000 - 0x80015a37
        section, .payload: 0x80200000 - 0x802751e7
100%    0MB   0.1MB/s  00:04
Setting PC to Program Start Address 0x80000000
Successfully downloaded C:/riscv/risc-v/workspace/boot.elf
xsdb% con
Info: Hart #0 (target 3) Running
xsdb%

```

rvccx, 2021-06-03:
Thanks for the information!

closed 2021-06-03 by rvccx