eugene-tarassov/vivado-risc-v #33: Debug with Vitis

tollomo, 2021-06-08:
Hi, 

I'm new in RISCV FPGA Design, so sorry if I am a bit of a noob here.
My question is how can I debug with Vitis ? When I export Hardware the xsa file is created.
After that I launch Vitis and create a new application project and select the xsa file and it is asked to choose a processor for an application project, but there is no existing target processor.
So my question is how to debug with the RISCV with Vitis.

Thanks very much

eugene-tarassov, 2021-06-08:
Vitis does not support managed application projects for RISC-V.
You have to craft a Makefile to build your application.

Debugging does not require an application project.
To launch the debugger in Vitis, you need to create Debug Configuration:

- Open Debug Configurations - press small triangle next to the bug icon and select Debug Configurations.
- Select Single Application Debug and press New Configuration button.
- Press Debug

In addition to GUI, the debugger has command line interface - XSDB (aka XSCT, aka Xilinx Software Commandline Tool), which can be used to download application to the target memory, program FPGA, run Tcl scripts, etc.

tollomo, 2021-06-09:
Aha got it! 
Thank you!

closed 2021-06-09 by tollomo