eugene-tarassov/vivado-risc-v #39: When we using Vitis 2021.1, wouldn't we need that riscv-2021.1.tcl files?

paranlee, 2021-07-31:
Hello, I am paran lee.

I am enthusiast risc-v and hardware IP. and i have a vc707 board as a hobby.

Follow as guide, then  he needed riscv-2021.1.tcl.

`couldn't read file "../../board/vc707/riscv-2021.1.tcl": no such file or directory`

Shall I fix it once I get back and request a pull request?

@eugene-tarassov  Thank you so much for writing such a great guide! :)

```bash
kim@kim-GL702ZC:~/vivado-risc-v$ make CONFIG=rocket64z2m BOARD=vc707 bitstream
make -C rocket-chip/firrtl SBT="java -Xmx4G -Xss8M  -jar /home/kim/vivado-risc-v/rocket-chip/sbt-launch.jar" build-scala
make[1]: 디렉터리 '/home/kim/vivado-risc-v/rocket-chip/firrtl' 들어감
java -Xmx4G -Xss8M  -jar /home/kim/vivado-risc-v/rocket-chip/sbt-launch.jar "assembly"
[info] Loading global plugins from /home/kim/.sbt/1.0/plugins
[info] Loading settings for project firrtl-build from plugins.sbt ...
[info] Loading project definition from /home/kim/vivado-risc-v/rocket-chip/firrtl/project

# ... 

[success] Total time: 147 s (02:27), completed 2021. 8. 1. 오전 12:45:25
rm workspace/bootrom.img
java -Xmx12G -Xss8M  -cp rocket-chip/firrtl/utils/bin/firrtl.jar:target/scala-2.12/classes:rocket-chip/rocketchip.jar firrtl.stage.FirrtlMain -i workspace/rocket64z2m/system-vc707/Vivado.Rocket64z2m.fir -o system-vc707.v -X verilog --infer-rw RocketSystem --repl-seq-mem \
  -c:RocketSystem:-o:workspace/rocket64z2m/system.conf \
  -faf workspace/rocket64z2m/system.anno.json \
  -td workspace/rocket64z2m/ \
  -fct firrtl.passes.InlineInstances
rocket-chip/scripts/vlsi_mem_gen workspace/rocket64z2m/system.conf >workspace/rocket64z2m/srams.v
mkdir -p vhdl-wrapper/bin
javac -g -nowarn \
  -sourcepath vhdl-wrapper/src -d vhdl-wrapper/bin \
  -classpath vhdl-wrapper/antlr-4.8-complete.jar \
  vhdl-wrapper/src/net/largest/riscv/vhdl/Main.java
java -Xmx4G -Xss8M  -cp \
  vhdl-wrapper/src:vhdl-wrapper/bin:vhdl-wrapper/antlr-4.8-complete.jar \
  net.largest.riscv.vhdl.Main -m Rocket64z2m \
  workspace/rocket64z2m/system-vc707.v >workspace/rocket64z2m/rocket.vhdl
echo "set vivado_board_name vc707" >workspace/rocket64z2m/system-vc707.tcl
echo "set vivado_board_part xilinx.com:vc707:part0:1.4" >>workspace/rocket64z2m/system-vc707.tcl
echo "set xilinx_part xc7vx485tffg1761-2" >>workspace/rocket64z2m/system-vc707.tcl
echo "set rocket_module_name Rocket64z2m" >>workspace/rocket64z2m/system-vc707.tcl
echo "set riscv_clock_frequency 40.0" >>workspace/rocket64z2m/system-vc707.tcl
echo 'cd [file dirname [file normalize [info script]]]' >>workspace/rocket64z2m/system-vc707.tcl
echo 'source ../../vivado.tcl' >>workspace/rocket64z2m/system-vc707.tcl
if [ ! -e workspace/rocket64z2m/vivado-vc707-riscv ] ; then env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64z2m/system-vc707.tcl ; fi
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
couldn't read file "../../board/vc707/riscv-2021.1.tcl": no such file or directory
    while executing
"source ../../board/${vivado_board_name}/riscv-${current_vivado_version}.tcl"
    (file "../../vivado.tcl" line 79)
make: *** [Makefile:239: workspace/rocket64z2m/vivado-vc707-riscv/timestamp.txt] error 1

```

eugene-tarassov, 2021-07-31:
I'm not sure why you get the error. The file is there: [board/vc707/riscv-2021.1.tcl](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/vc707/riscv-2021.1.tcl), and it woks fine for me. Perhaps, you just need to update your repo, the file is relatively new, it was committed 26 days ago.

BTW, rocket64z2m configuration is too big for vc707 - it will not fit. Emulation of such big processor requires one of high end Virtex Ultrascale+ boards. For vc707, rocket64y1 is the biggest BOOM configuration it can run.

paranlee, 2021-07-31:
> I'm not sure why you get the error. The file is there: [board/vc707/riscv-2021.1.tcl](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/vc707/riscv-2021.1.tcl), and it woks fine for [...]

@eugene-tarassov  Thank you so much for your kindness. 

As you said, I checked out the last release version. I made a mistake while looking at the guide based on the GitHub master content. I had no problems running rocket64y1. 

Thanks again for the quick and kind reply!

```bash

// ...

===================================
Configuration Memory information
===================================
File Format        MCS
Interface          BPIX16
Size               32M
Start Address      0x00000000
End Address        0x01FFFFFF

Addr1         Addr2         Date                    File(s)
0x00000000    0x00D939CB    Aug  1 06:45:56 2021    workspace/rocket64y1/vivado-vc707-riscv/vc707-riscv.runs/impl_1/riscv_wrapper.bit
5 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_cfgmem completed successfully
kim@kim-GL702ZC:~/vivado-risc-v$ 

```

closed 2021-07-31 by paranlee