eugene-tarassov/vivado-risc-v #40: How to connect Internet on VC707 board

xiongwu12345, 2021-08-11:
I use the released vc707 bit and the ./mk-sd-cad command, the Linux can boot successfully.
And I use the internet cable to connect the board and the wired network port.
But when I try to connect to the Internet, as I use the command "sudo ping 182.61.200.6", it returns "the network is unreachable".
[image]

And the IP address is as below:

[image]

And I noticed that :
[image]

It seems that I need to change some MAC addresses and rebuild the bitstream. But what exactly should I do to connect the board to the Internet?

Thanks a lot!

eugene-tarassov, 2021-08-12:
> But what exactly should I do to connect the board to the Internet?

You need to connect the board to 1Gb router using Ethernet cable and it should just work:

[image]

It looks like your cable is not connected or not working, because your screenshot shows NO-CARRIER status.

> It seems that I need to change some MAC addresses and rebuild the bitstream.

Only if you have multiple VC707 boards.

xiongwu12345, 2021-08-12:
It solves my problem,thanks!

closed 2021-08-12 by xiongwu12345

xiaolingyi, 2022-01-18:
Hi, after some time, when I wanted to connect to the internet again, it suddenly showed NO-CARRIER status again. I did not change any file related to the Ethernet. And I changed to another brand new  VC707 board, the phenomenon was the same and I am sure the internet, as well as the Ethernet cable, is fine. 
So could you please provide any advice on how to test the ETHERNET devices on the board or how to solve this problem(for example, any config change)?
Thanks a lot!

eugene-tarassov, 2022-01-18:
The board has SGMII interface to Ethernet PHY, so the design uses "1G/2.5G Ethernet PCS/PMA or SGMII" IP. The IP is proprietary and encrypted, so debug options are limited. I suggest to check the IPs Product Guide, it has plenty of info on how the IP should work, how to configure it, and how to create a test bench. You can also use Vivado to add ILA for the IP signals and check the waveforms.

I have VC707 board set up to run daily tests and have not seen any issues, I don't know what causes the NO-CARRIER status.