eugene-tarassov/vivado-risc-v #41: Unspecified I/O standard and Unconstrained Logic Port error

xiongwu12345, 2021-08-12:
Hi, when I tried to generate the bitstream using this command "make CONFIG=rocket64b1 BOARD=vc707 bitstream",I used vivado2019.1 to run the project firstly，and find there is no corresponding vivado tcl file, so I renamed the 2020.1 version tcl file as 2019.1.tcl to run through the makefile. And  later  I encountered this error:
[image]

And In the vivado project, the error shows:
[image]
So I installed the vivado of 2021.1 and delete the vivado project file( all the files in directory workspace/rocket64b1 ). And I try to run again this command "make CONFIG=rocket64b1 BOARD=vc707 bitstream".But It still has the same error as above.
What does these error mean and how to fix them? And I am confused about if I didn't delete some file(generated by 2019.1 vivado) completely?
Thanks a lot!

eugene-tarassov, 2021-08-12:
Vivado 2019.1 is not supported, and it looks like you are still using 2019.1 after installing 2021.1.
Did you run `source /opt/Xilinx/Vivado/2021.1/settings64.sh` to update environment variables?

xiongwu12345, 2021-08-16:
Thanks, this problem is solved!

closed 2021-08-16 by xiongwu12345