eugene-tarassov/vivado-risc-v #45: Why ssh work only in the same ip domain

xiongwu12345, 2021-08-18:
Hi, I know nothing about computer networks or ethernet. But I find that after the board connect to the internet and it has an IP address, and when the local machine connecting to it using ssh, the local machine must have the same the first three IP address as the board. Such as when the IP address of the board is 192.168.0.62, then the local machine must have an IP address like 192.168.0.xx, otherwise, it simply cannot connect to the board. Furthermore, this situation is the same for using the tvm runtime with rpc(Remote Procedure Call). 
This situation causes a little inconvenience in practice. So what is the principle behind the board connecting to the internet and is there any method to remove this "IP same" constrain?

Thanks a lot~

eugene-tarassov, 2021-08-18:
> So what is the principle behind the board connecting to the internet and is there any method to remove this "IP same" constrain?

192.168.0 is address of a network  (aka subnet or LAN), 62 is address of a machine on that network. Machines on a same subnet can communicate directly. Machines on different subnets need a 3rd machine (aka gateway) to route (or relay) traffic between subnets.  However, 192.168.xx is a special class of subnets, which are not "routable" by default. Anyway, this has nothing to do with the board or RISC-V. Your board is 100% ready for networking. "IP same" constrain is property of your network, and it is, probably, intentional.

>  I tried some commands like "power off", "shutdown", "halt"

These commands only available for superuser (aka root), use sudo to run the command:

`sudo poweroff`

xiongwu12345, 2021-08-19:
Oh, I see. Thank you~

closed 2021-08-19 by xiongwu12345