eugene-tarassov/vivado-risc-v #46: Linux cannot boot after a custom RoCC added

xiongwu12345, 2021-08-28:
Hi, I add a custom RoCC to the rocket tile, and the bitstream is generated successfully. But when I want to boot the Linux, the serial port shows nothing, even the opensbi logo is gone. 
So could you give me some advice on how to debug and what kind of signals I need to pay attention to? Because I just can not find where to shart. 
And I find there are thousands of timing constraints no met. But fix them one by one seems impossible. So what is your opinion on this timing problem?
Thanks a lot.

eugene-tarassov, 2021-08-28:
> And I find there are thousands of timing constraints no met. 

Usually, it is caused by RISC-V CPU clock frequency being too high.

In Vivado, open Block Design, double-click on Clocking Wizard and change clk_out1 frequency to something really low, like 20MHz. When you get the design working OK, you can try to increase the frequency.

xiongwu12345, 2021-08-30:
I change clk_out1 frequency to 10MHz and get the bitstream. When I boot the Linux, it shows no boot image file, as shown below :
 _[image]_
Seems the rocket can not read the file on the SD card. Sigh.

eugene-tarassov, 2021-08-30:
I think 10MHz is too low. At this speed SD card controller don't get enough throughput to push data from the card to the main memory, which causes buffer overflow and read errors. In general, supported clock speeds are 160 125 100 80 62.5 50 40 31.25 25 20 MHz. Select one that is not too high to cause timing violations, and not too low to cause IO errors.

xiongwu12345, 2021-09-01:
>Select one that is not too high to cause timing violations, and not too low to cause IO errors.

It works,thank you so much.

closed 2021-09-01 by xiongwu12345

xiongwu12345, 2021-09-28:
>In general, supported clock speeds are 160 125 100 80 62.5 50 40 31.25 25 20 MHz. 

This recommended frequency works pretty well. And I am puzzled as to why only these frequencies are supported.

Thanks.

eugene-tarassov, 2021-09-28:
> And I am puzzled as to why only these frequencies are supported.

The clock manager (MMCM) uses phase locked loop (PLL) and counters to multiply and divide frequency of the clock coming from a crystal oscillator on the board. Supported frequencies are determined by the oscillator frequency, length of MMCM counters, timing constraints, and the fact that multiple output clocks share same oscillator and multiplier.