eugene-tarassov/vivado-risc-v #47: Top module not set for fileset 'source_1'

wkxfudan, 2021-08-31:
Under the vivado 2018.2 env, it occurred error when i run 
`make CONFIG=rocket64b2 BOARD=nexys-video bitstream`

Scanning sources...
Finished scanning sources
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
ERROR: [Common 17-70] Application Exception: Top module not set for fileset 'sources_1'. Please ensure that a valid value is provided for 'top'. The value for 'top' can be set/changed using the 'Top Module Name' field under 'Project Settings', or using the 'set_property top' Tcl command (e.g. set_property top <name> [current_fileset]).

I think 'source_1' means the fileset 'rocket64', and all the file is in next picture
[image]

**What's the top module name and how I fix it?**

eugene-tarassov, 2021-08-31:
[board/vc707](https://github.com/eugene-tarassov/vivado-risc-v/tree/master/board/vc707) directory contains Tcl script that creates  the top module. The are multiple versions of the script for different versions of Vivado. However, Vivado 2018.2 is not supported. Supported Vivado versions are 2020.1, 2020.2 and 2021.1.

yangshuaixian, 2021-09-02:
I have the same question, and my Vivado version is 2020.1.  I dont konw why i cant generate sources_1. 
**if [ ! -e workspace/ZRV64x1/vivado-vc707-riscv ] ; then env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/ZRV64x1/system-vc707.tcl ; fi
date >workspace/ZRV64x1/vivado-vc707-riscv/timestamp.txt
echo "open_project workspace/ZRV64x1/vivado-vc707-riscv/vc707-riscv.xpr" >workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
echo "update_compile_order -fileset sources_1" >>workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
echo "set_param general.maxThreads 1" >>workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
echo "launch_runs impl_1 -to_step write_bitstream -jobs 1" >>workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
echo "wait_on_run impl_1" >>workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/ZRV64x1/vivado-vc707-riscv/make-bitstream.tcl
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
Scanning sources...
Finished scanning sources
WARNING: [filemgmt 56-3] Default IP Output Path : Could not find the directory '/home/rongyitong/code/vivado-risc-v/workspace/ZRV64x1/vivado-vc707-riscv/vc707-riscv.srcs/sources_1'.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
ERROR: [Common 17-70] Application Exception: Top module not set for fileset 'sources_1'. Please ensure that a valid value is provided for 'top'. The value for 'top' can be set/changed using the 'Top Module Name' field under 'Project Settings', or using the 'set_property top' Tcl command (e.g. set_property top <name> [current_fileset]).
Makefile:244: recipe for target 'workspace/ZRV64x1/vivado-vc707-riscv/vc707-riscv.runs/impl_1/riscv_wrapper.bit' failed
make: *** [workspace/ZRV64x1/vivado-vc707-riscv/vc707-riscv.runs/impl_1/riscv_wrapper.bit] Error 1**

eugene-tarassov, 2021-09-02:
Your Vivado project looks broken. Probably because of one of previous attempts to run make failed with error. I suggest to delete or backup the workspace, e.g. `mv workspace ../workspace.backup`, and run make again.

yangshuaixian, 2021-09-06:
@eugene-tarassov you are right, it's my Vivado project broken. And now i have another question, How do I modify my image file because I want to load some test files?

eugene-tarassov, 2021-09-06:
I don't know what you mean by "I import the file through the network port". Apparently, something is missing.

This is how you can run coremark on the board:
```
sudo apt update
sudo apt upgrade
sudo apt install git make gcc
git clone https://github.com/eembc/coremark.git
cd coremark
make
cat run1.log
```

And the results should look like this:
```
debian@debian:~/coremark$ cat run1.log
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 15151
Total time (secs): 15.151000
Iterations/Sec   : 198.006732
Iterations       : 3000
Compiler version : GCC10.3.0
Compiler flags   : -O2 -DPERFORMANCE_RUN=1  -lrt
Memory location  : Please put data memory location here
                        (e.g. code in flash, data on heap etc)
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0xcc42
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 198.006732 / GCC10.3.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap
debian@debian:~/coremark$
```

closed 2021-12-05 by eugene-tarassov

zhaokkkkkkkkk, 2023-06-14:
I cant generate sources_1 too,and my Vivado version is 2023.1
echo "wait_on_run synth_1" >>workspace/rocket64b2/vivado-nexys-video-riscv/make-synthesis.tcl
env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64b2/vivado-nexys-video-riscv/make-synthesis.tcl
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
INFO: [filemgmt 56-3] Default IP Output Path : Could not find the directory '/home/debian/git/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.gen/sources_1'.
Scanning sources...
Finished scanning sources
open_project: Time (s): cpu = 00:00:11 ; elapsed = 00:00:13 . Memory (MB): peak = 1307.203 ; gain = 96.836 ; free physical = 16861 ; free virtual = 19218
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
WARNING: [Vivado 12-7122] Auto Incremental Compile:: No reference checkpoint was found in run synth_1. Auto-incremental flow will not be run, the standard flow will be run instead.
ERROR: [Common 17-53] User Exception: Top module not set for synthesis run
Ensure that a valid value is provided for 'top'. The current value for 'top' can be changed using the 'Top Module Name' field under 'Project Settings', or using the 'set_property top' Tcl command
make: *** [Makefile:314: workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.runs/synth_1/riscv_wrapper.dcp] Error 1

eugene-tarassov, 2023-06-14:
Vivado version 2023.1 is not supported yet. You have to use 2022.2 for now.