eugene-tarassov/vivado-risc-v #54: ERRORs when running implement with vivado 2021.2 on Genesys2

Arima-Kawori, 2021-11-12:
Hi, I'm trying to use Vivado 2021.2 and Genesys2 board to run linux, using this project.
Since there is no Tcl script for 2021.2 version in board/genesys2/, I use the script for 2021.1 instead.
and Then when I try to generate the bitstream, the following error occurred.

```
Running DRC as a precondition to command place_design
ERROR: [DRC UTLZ-1] Resource utilization: LUT as Logic over-utilized in Top Level Design (This design requires more LUT as Logic cells than are available in the target device. This design requires 261110 of such cell types but only 203800 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.)
ERROR: [DRC UTLZ-1] Resource utilization: Slice LUTs over-utilized in Top Level Design (This design requires more Slice LUTs cells than are available in the target device. This design requires 270419 of such cell types but only 203800 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.)
```

And after I `set tcl parameter "drc.disableLUTOverUtilError" to 1` as it mentioned in the ERROR, the following ERRORs occurred:

```
ERROR: [Place 30-487] The packing of instances into the device could not be obeyed. There are a total of 50950 slices in the device, of which 45188 slices are available, however, the unplaced instances require 62365 slices. Please analyze your design to determine if the number of LUTs, FFs, and/or control sets can be reduced.
```

I think I'd done nothing wrong except the script version above. I'll try to run the whole job in the 2021.1 again, but I'm wondering whether this problem is caused by the script version.
Thank you very much!

eugene-tarassov, 2021-11-12:
Those scripts require exact version of Vivado they were generated for.
No scripts for Vivado 2021.2 yet. At the moment, I recommend to use 2021.1.

> Resource utilization: LUT as Logic over-utilized...

This looks like a valid error, it means you are trying to use RocketChip configuration that is too large for your board.

Arima-Kawori, 2021-11-13:
Thanks for your answer.
I tried 2021.1 and re-generate a BOOM using rocket64x1 (rocket64y1 used before), and now all works well.
Best wishes!

closed 2021-11-13 by Arima-Kawori