eugene-tarassov/vivado-risc-v #55: Vivado 2018.3 support for the VCU1525 board

ncppd, 2021-11-18:
Hello and thanks for your work in this repository. I am using Vivado 2018.3 for the VCU1525 board, is there an easy way to port your board files in the older Vivado version?

eugene-tarassov, 2021-11-18:
Porting to Vivado 2018.3 means creating one file: board/vcu1525/riscv-2018.3.tcl
riscv-2020.2.tcl can be used as a starting point and edited to match whatever Vivado 2018.3 requires.

Note that Linux is not currently supported on the VCU1525 board.
I have VCU1525 design with 10G Ethernet and Linux using NFS for rootfs, but it is not committed yet - I'm still testing it.

ncppd, 2021-11-19:
Thank you for your answer. How do you deploy designs on the VCU1525 using Vivado 2021.1 since (as far as I know) the last supported XRT and development shell support the 2018.3 version? Do you have more information regarding deployment on the VCU1525?

Also, are you going to commit Linux support for the VCU1525? It would be nice to have Rocket Chip + Linux support for the VCU1525, I would be glad to help you in this task (for example as a beta tester). 

Thank you and sorry for all those questions!

eugene-tarassov, 2021-11-19:
> the last supported XRT and development shell support the 2018.3 version?

I'm not using XRT. I want to keep it simple. For me VCU1525 is just a large FPGA with up to 64GB of DDR4 memory and up to 8 10G Ethernets. This works with any version of Vivado. RISC-V Linux boots with rootfstype=nfs kernel option.

> Also, are you going to commit Linux support for the VCU1525?

Yes. Soon. Like in few days.

Rocket Chip + Linux works fine with VCU1525 and Vivado 2021.1, but I see some regression tests failures with other boards and other versions of Vivado. I need to fix, at least, regressions in other boards.

Also, I don't have a boot ROM for VCU1525 yet. I download Linux kernel using JTAG debugger (XSDB). Other boards boot from SD card, but VCU1525 does not have SD card slot, should use TFTP. But I'll commit regardless of that.

ncppd, 2021-11-22:
Would this mean that after you commit support for the VCU1525, that Rocket Chip + Linux could be easily be ported to the Alveo 250/280 boards? (since the VCU1525 is the development version of these boards)

Thank you for your all your help and support!

eugene-tarassov, 2021-11-22:
Yes, I think so. As far as I know, VCU1525 is nearly identical to Alveo U200.
However, I don't plan to support cloud/data center use case. It is supported by [FireSim](https://github.com/firesim/firesim).

BTW, I have committed first version of the VCU1525 support, e.g.:
```
make CONFIG=rocket64b2 BOARD=vcu1525 ROOTFS_URL=192.168.0.100:/home/nfsroot/riscv jtag-boot
```

ncppd, 2021-11-26:
I am trying to port your design in Vivado 2018.3 using board part `xilinx.com:vcu1525:part0:1.1` and I am getting the following errors:

> ERROR: [DRC NSTD-1] Unspecified I/O Standard: 277 out of 277 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: m_axi_araddr[31:0], m_axi_arlen[7:0], m_axi_awaddr[31:0], m_axi_awlen[7:0], m_axi_rdata[31:0], m_axi_wdata[31:0], s_axi_araddr[15:0], s_axi_awaddr[15:0], s_axi_bresp[1:0], s_axi_rdata[31:0], s_axi_rresp[1:0], s_axi_wdata[31:0], sdio_dat[3:0], async_resetn, clock... and (the first 15 of 42 listed).

> ERROR: [DRC UCIO-1] Unconstrained Logical Port: 277 out of 277 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: m_axi_araddr[31:0], m_axi_arlen[7:0], m_axi_awaddr[31:0], m_axi_awlen[7:0], m_axi_rdata[31:0], m_axi_wdata[31:0], s_axi_araddr[15:0], s_axi_awaddr[15:0], s_axi_bresp[1:0], s_axi_rdata[31:0], s_axi_rresp[1:0], s_axi_wdata[31:0], sdio_dat[3:0], async_resetn, clock... and (the first 15 of 42 listed).

I ignored these errors by adding `set_property BITSTREAM.General.UnconstrainedPins {Allow} [current_design]` in `board/vcu1525/top/xdc`. Is this correct?

After those errors are ignored Vivado reports that write_bitstream completed successfully.

However, after write_bitstream finishes, I am getting the following errors:
> ERROR: [Vivado 12-4739] set_false_path:No valid object(s) found for '-through [get_pins -hier RocketChip/clock_ok]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:5]
ERROR: [Vivado 12-4739] set_false_path:No valid object(s) found for '-through [get_pins -hier RocketChip/mem_ok]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:6]
ERROR: [Vivado 12-4739] set_false_path:No valid object(s) found for '-through [get_pins -hier RocketChip/io_ok]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:7]
ERROR: [Vivado 12-4739] set_false_path:No valid object(s) found for '-through [get_pins -hier RocketChip/sys_reset]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:8]
ERROR: [Vivado 12-4739] set_max_delay:No valid object(s) found for '-from [get_clocks -of_objects [get_pins -hier SD/clock]]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:31]
ERROR: [Vivado 12-4739] set_max_delay:No valid object(s) found for '-from [get_clocks -of_objects [get_pins -hier RocketChip/clock]]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:103]
ERROR: [Vivado 12-4739] set_max_delay:No valid object(s) found for '-from [get_clocks -of_objects [get_pins -hier RocketChip/clock]]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:114]
ERROR: [Vivado 12-4739] set_max_delay:No valid object(s) found for '-from [get_clocks -of_objects [get_pins -hier ddr4_0/c0_ddr4_ui_clk]]'. [/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/board/timing-constraints.tcl:115]

Do you know how can I correct those errors? Thank you!

eugene-tarassov, 2021-11-26:
> I ignored these errors by adding set_property BITSTREAM.General.UnconstrainedPins {Allow} [current_design] in board/vcu1525/top/xdc. Is this correct?

No, don't do that. You need to set [board/vcu1525/riscv_wrapper.v](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/vcu1525/riscv_wrapper.v) as the top module. Actually, it is set as top by vivado.tcl script, I'm not sure how you got wrong module at the top.

ncppd, 2021-11-30:
It turns out that the `xilinx.com:ip:qdma:4.0` IP version does not exist in Vivado 2018.3, I had to change it to `xilinx.com:ip:qdma:3.0`. Due to this error the tcl script created the project but did not finalize all the settings correctly. When I tried to re-run make the project had wrong configuration, and I got the above errors. After I fixed the IP version, it compiles correctly, but I have not tested it on the VCU1525 yet.
I do not know if version 4.0 of qdma IP is backwards compatible with version 3.0, because I get several critical warnings:

During synthesis Vivado reported that `qdma_0 property cfg_mgmt_if: does not exist` (critical warning)

> [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group [get_clocks -of_objects [get_ports -scoped_to_current_instance sys_clk]]'. ["/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/workspace/rocket64b2/vivado-vcu1525-riscv/vcu1525-riscv.srcs/sources_1/bd/riscv/ip/riscv_qdma_0_0/ip_0/synth/riscv_qdma_0_0_pcie4_ip_late.xdc":63]
 
+6 more critical warnings same as this, and 
> [Vivado 12-5201] set_clock_groups: cannot set the clock group when only one non-empty group remains. ["/media/ncppd/Data/Projects/riscv/hardware/vivado-risc-v/workspace/rocket64b2/vivado-vcu1525-riscv/vcu1525-riscv.srcs/sources_1/bd/riscv/ip/riscv_qdma_0_0/ip_0/synth/riscv_qdma_0_0_pcie4_ip_late.xdc":63]

I am going to test if linux boots on the board and get back to you. Thanks!

eugene-tarassov, 2021-11-30:
Yes, xilinx.com:ip:qdma:3.0 looks broken. I have seen same critical errors. QDMA is not needed to run RISC-V Linux, you can remove from the design.

ncppd, 2021-12-01:
I compiled the bitstream correctly and I am now stuck on the boot flow, I have some questions:
I noticed that there are some hardcoded paths in the Makefile such as `ROOTFS_URL ?= 192.168.0.100:/home/nfsroot/192.168.0.243`. Do those 2 IPs refer to the development machine, the deployment machine or the machine that runs the hardware server? Also, what path should I replace the `/home/nfsroot/` with?
Regarding the JTAG boot flow, do I have to plug an external JTAG on the IO shield of the VCU1525, or just the PCIe will do? Thanks!

eugene-tarassov, 2021-12-01:
> I noticed that there are some hardcoded paths in the Makefile such as ROOTFS_URL ?= ...

You need to rebuild the bootloader. Run make with parameters like this:
```
make BOARD=vcu1525 CONFIG=<cfg> \
 ROOTFS_URL=<your NFS server IP>:<NFS directory to mount as root FS> \
 HW_SERVER_ADDR=<your workstation IP> JTAG_BOOT=1  clean bootloader linux
```
I recommend to create a script to avoid typing the parameters every time.

You can setup NFS server on the development machine or any other machine on your local network, create the directory and make it exported by adding something like this into /etc/exports:
```
/home/nfsroot 192.168.0.0/24(rw,no_subtree_check,no_root_squash,no_all_squash)
```
Download [rootfs.tar.gz](https://github.com/eugene-tarassov/vivado-risc-v/releases/download/v3.2.0/rootfs.tar.gz) from the Releases area and un-tar it to the NFS directory. During boot, RISC-V Linux kernel will use DHCP to get network info, then mount the directory as the root file system.

> Regarding the JTAG boot flow, do I have to plug an external JTAG on the IO shield of the VCU1525, or just the PCIe will do?

You have to plug an external JTAG on the IO shield of the VCU1525. Debugging over PCIe is not supported yet.

ncppd, 2021-12-02:
I successfully set up nfs and the other components and now `make BOARD=vcu1525` completes without errors, I do not know though how to connect to the board. It seems that the board is not getting an IP so I cannot ssh in it. Does this mean that something has gone wrong? Is there another way to connect to Debian running on the board to check if everything works?

eugene-tarassov, 2021-12-02:
> Is there another way to connect to Debian running on the board to check if everything works?

Yes, you can use serial port. When the board USB-JTAG cable is connected, you should see 4 USB serial ports: /dev/ttyUSB0 .. /dev/ttyUSB3. Port #2 (/dev/ttyUSB2) is connected to the FPGA and is used as RISC-V/Linux console. Connect with any serial port communications program, for example [minicom](https://help.ubuntu.com/community/Minicom), and you will see all boot ROM, u-boot, kernel messages and login prompt.

When FPGA is programmed successfully, you should see, at least, boot ROM message:
```
RISC-V 64, Boot ROM V3.3
```

eugene-tarassov, 2021-12-02:
Serial port output of VCU1525 Rocket64b2m booting Linux over JTAG and NFS should look like this:
```
RISC-V 64, Boot ROM V3.3

OpenSBI v0.9
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : Vivado RISC-V
Platform Features         : medeleg
Platform HART Count       : 32
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1250000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Firmware Base             : 0x80000000
Firmware Size             : 468 KB
Runtime SBI Version       : 0.3

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*,8*,9*,10*,11*,12*,13*,14*,15*,16*,17*,18*,19*,20*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000080000000-0x000000008007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcsux
Boot HART Features        : scounteren,mcounteren,mcountinhibit
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 36
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2021.07-dirty (Dec 02 2021 - 13:56:57 -0800)

CPU:   rv64imafdc
Model: freechips,rocketchip-vivado
DRAM:  14 GiB
MMC:
Loading Environment from nowhere... OK
In:    uart@60010000
Out:   uart@60010000
Err:   uart@60010000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Moving Image from 0x81000000 to 0x80200000, end=81316c30
## Flattened Device Tree blob at 00010080
   Booting using the fdt blob at 0x010080
   Loading Device Tree to 0000000084ffc000, end 0000000084fffdff ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.4-dirty (eugene@gate) (riscv64-linux-gnu-gcc (Ubuntu 9.3.0-17ubunt1
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[... 273 lines cut ...]
[  OK  ] Reached target Timer Units.
[  OK  ] Listening on D-Bus System Message Bus Socket.
[  OK  ] Reached target Socket Units.
[  OK  ] Reached target Basic System.
         Starting Automounts filesystems on demand...
[  OK  ] Started Regular background program processing daemon.
[  OK  ] Started D-Bus System Message Bus.
         Starting Remove Stale Onli▒…t4 Metadata Check Snapshots...
         Starting System Logging Service...
         Starting OpenBSD Secure Shell server...
         Starting User Login Management...
         Starting Permit User Sessions...
[  OK  ] Started System Logging Service.
[  OK  ] Finished Permit User Sessions.
[  OK  ] Started Serial Getty on ttyAU0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started Automounts filesystems on demand.
[  OK  ] Started OpenBSD Secure Shell server.
[  OK  ] Started User Login Management.
[  OK  ] Reached target Multi-User System.

Debian GNU/Linux bookworm/sid debian ttyAU0

debian login:
```

ncppd, 2021-12-07:
I got Linux to boot on the VCU1525 and the Alveo U250 until the stage when it tries to mount the remote filesystem, and now I am trying to figure out on how to setup the QSFP28 connector. From what I have read online either: 
1. I must use an QSFP+ to 4x10G SFP+ cable and connect it to a switch in the lab that has SFP+ ports or 
2.  a QSFP+ to SFP port module (such as a Cisco QSFP to SFP SFP+ adapter) and then use an RJ-45 to SFP port module to connect it to a standard 1G RJ-45 port on the switch. 

Regarding scenario (1), do you know if the switch will automatically recognize the enabled channel on the board and set up either 1G or 10G speed (if supported)? Will scenario (2) work, since the connection speed will be 1G?
Can you give me any hints on what connectors you use on your setup? Thanks!

eugene-tarassov, 2021-12-07:
I use QSFP+ to 4x10G SFP+ cable and 10G SFP+ switch. The switch automatically recognizes the enabled channel and runs at 10G. I have not tried 1G QSFP to SFP adapter. It probably will require some changes in the design to support 1G speed.

ncppd, 2021-12-07:
Until I set up the network equipment, is there a way to boot Linux without an internet connection?

eugene-tarassov, 2021-12-07:
It is possible to download rootfs into RAM over JTAG and tell u-boot to boot the kernel with RAM disk. 
This involves creating RAM disk image, e.g 'mkimage -A riscv -T ramdisk -C gzip -d "ext4 root fs image" ramdisk', changing kernel arguments to 'root=/dev/ram0 earlycon rw console=ttyAU0,115200n8', XSDB dow command to download the image, and u-boot command like 'booti ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr}'. I don't have a ready-to-use example for this.

Geekdude, 2021-12-10:
I was able to successfully port the VCU1525 board files to the Alveo U250, by following the advice in this issue and from other issues. The system successfully booted into the Debian Linux environment using JTAG with NFS root. One additional step I needed to successfully boot into the NFS root was to change the /etc/fstab file in the NFS root to 
`192.168.0.0:/home/nfsroot     /       nfs    defaults    0       1`
to have the "Remount Root and Kernal File Systems" step succeed during boot.

I also tried to download rootfs into RAM over JTAG, but the download estimated that it would take 25 days to complete, so doing so is impractical.

eugene-tarassov, 2021-12-11:
> The system successfully booted into the Debian Linux environment using JTAG with NFS root.

Excellent!

> One additional step I needed to successfully boot into the NFS root was to change the /etc/fstab file...

Actually, deleting the line in the /etc/fstab file should work even better, "Remount Root and Kernel File Systems" is redundant step in this case.

eugene-tarassov, 2021-12-11:
BTW, VCU1525 (and Alveo) allows prototyping very large systems. But, according to my tests, large RocketChip configs (e.g. 64x2 64z1 64z2m 64b4m4) are currently unstable, the processor can freeze. I suspect a bug in RocketChip RTL. Smaller configs (e.g. 64b2 64b2m 64b2m2 64b4) are working fine.

jeehoonkang, 2021-12-22:
Thanks for the information! I'm also interested in running Rocket Chip on Alveo U250 (or U200/U280), and I wonder if you could upstream your development to this repository. I'm nearly new to FPGA and struggling to figure out how to "bring up" everything...

ncppd, 2022-01-12:
@jeehoonkang sorry for the late reply, you can check out the pull request I have opened that adds support for the Alveo U250 #67

ncppd, 2022-02-15:
@eugene-tarassov Hello, I am running into an error when I try to compile the design with Vivado 2020.2 for the VCU1525 board. After the bitstream generation has completed successfully it fails on the "Download Bitstream" step:

> Downloading bitstream                                                                                                                                   
no targets found with "name =~ "Hart #0*"". available targets:                                                                                          
  1* xcvu9p
     2  RISC-V JTAG DTM (Debug Transport Module timeout)
     3  Legacy Debug Hub
    while executing
"error "no targets found with \"$params(filter)\". available targets:$target_list""
    (procedure "targets" line 177)
    invoked from within
"targets -set -filter {name =~ "Hart #0*"}"
    (file "board/jtag-boot.tcl" line 7)
make: *** [Makefile:333: jtag-boot] Error 1

`xsdb targets ` command output:

> xsdb% targets                                                                                                                                                                             
  1  xcvu9p
     2  RISC-V JTAG DTM (Debug Transport Module timeout)
     3  Legacy Debug Hub

It seems that the bitstream is not correctly configured. The command I am using is `make CONFIG=rocket64b1 BOARD=vcu1525 ROOTFS_URL=XXX.XXX.XXX.XXX:/home/ncppd/Projects/nfsroot/ jtag-boot
`
Compiling the Project for the U250 boards works perfectly, in this case I cannot understand why it fails. I tried deleting the project and recompile it from the start but the error persists. I also tried reinstalling Vivado 2020.2 to no avail. Do you have any thoughts on what could be going wrong, I think I am missing something obvious but I cannot pinpoint the issue.

Thanks!

eugene-tarassov, 2022-02-15:
I see same problem with CONFIG=rocket64b2m2 for the VCU1525 board. Other configs work OK, but I have not tried rocket64b1. I think it has something to do with timing constraints, because it works if I reduce Rocket clock frequency, or make minor changes in the design. It behaves like under-constrained design. I don't have a solution yet.

kosterz96, 2022-02-27:
I'm trying to port my design of Rocket Chip into Alveo U250 as part of my thesis but I encounter some errors regarding xsdb. I am able to generate the bitstream, load xsdb, connect, but when I type "targets" I cannot see the FPGA board listed. Actually, nothing is printed. I use ssh to connect to the remote server where U250 is plugged in. The strangest thing is that other users are able to connect and even port Rocket Chip to U250 with no problem.

As a result of this situation when I try to run 
> make CONFIG=rocket64b1 BOARD=u250 ROOTFS_URL=[url_here]:[path_to_rootfs] HW_SERVER_ADDR=127.0.0.1:3121 jtag-boot

I encounter the following error:

> env HW_SERVER_URL=tcp:127.0.0.1:3121 \
>  xsdb -quiet board/jtag-freq.tcl
> rlwrap: warning: your $TERM is 'xterm-256color' but rlwrap couldn't find it in the terminfo database. Expect some problems.
> target list is empty
>     while executing
> "error "target list is empty""
>     (procedure "::xsdb::jtag::targets" line 154)
>     invoked from within
> "jtag targets -set -filter {level == 0 && is_open}"
>     (file "board/jtag-freq.tcl" line 5)
> Makefile:332: recipe for target 'jtag-boot' failed
> make: *** [jtag-boot] Error 1

I use Xilinx System Debugger (XSDB) v2020.2. Do you have any ideas what could be wrong?

eugene-tarassov, 2022-02-27:
It is hard to tell. You can try:
1. Check U250 USB cable is connected:
```
eugene@vcu1525:~$ lsusb
Bus 001 Device 002: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
...
```
2. Make sure you run HW Server and XSDB on the same machine where U250 is plugged in. If you run XSDB remotely, check HW_SERVER_ADDR value. HW Server cannot run remotely.
3. Start HW Server with `-L- -ljtag` command line options to enable printing of JTAG log.

kosterz96, 2022-03-01:
It seems like rebooting twice the machine where U250 is plugged in solved the problem. Is there a way I could skip this part and make it work without rebooting?

eugene-tarassov, 2022-03-01:
It is not required to reboot the machine - unless something went really wrong. One possibility is that you cannot access the board because it is locked by another instance of HW Server. Perhaps, started by another user. Only one HW Server can access the board at a time.

kosterz96, 2022-04-01:
Hello, I am trying to figure out how to increase the amount of RAM used by the design in order to run SPEC2017 benchmarks suite. I have changed the value of MEMORY_SIZE both into vivado-risc-v//board/u250/Makefile.inc and vivado-risc-v/Makefile. The new value is 0x4000000000. However every time I try to boot I keep getting only 2GB of RAM. I also tried changing rocket64b1 parameters by adding the following line:
> new WithExtMemSize(0x3f80000000L) ++

After adding this line U-Boot and /proc/meminfo say than 128GB of RAM are available which is super weird considering that Alveo U250 only supports up to 64GB of RAM. Even so, I think that no more than 2GB of RAM are utilized by Rocket Core because SPEC benchmarks are too memory hungry and /proc/meminfo gives me 126GB of available memory.

So, in order to conclude, my questions are:
1.  I guess MEMORY_SIZE refers to bytes as a hex value, right?
2.  What steps do I have to follow in order to increase available RAM?

Thank you everyone in advance!

ncppd, 2022-04-04:
Hello, is the 16GB limit the same for the VCU1525 board? Do you know how can we utilize the whole 64GB RAM of the U250 & VCU1525? Thanks!

eugene-tarassov, 2022-04-04:
VCU1525 has 4 memory controllers, each with 16GB of DRAM, total 64GB of RAM.
To access all 64GB, the design needs to include all 4 controllers and additional interconnect.

There are 3 different Tcl design files for VCU1525 - with 1, 2 or 4 memory controllers.

In particular, `make BOARD=vcu1525 CONFIG=Rocket64b4m4 ...` will produce 4-core RISC-V with 4 memory channels, 4 memory controllers and 64GB RAM.

ncppd, 2022-06-02:
Hello, I tam trying these days to boot linux with the VCU1525 board without success (it works fine with the U250). Could this be due to missing XRT or DSA packages? 

I downgraded a machine to Ubuntu 18.04, followed the instructions in the VCU1525 Getting Started Guide, successfully installed the deb packages and it managed to download the bitstream correctly. Unfortunately, during the "Downloading Linux Image" step the machine freezed (I suspect a kernel panic).

I have now installed both cards (U250 & VCU1525) in a new machine (Ubuntu 20.04) and use another machine to host the JTAG cables. The U250 board works fine and boots linux correctly, on the other hand for the VCU1525, I am still getting the same error as an above post `no targets found with "name =~ "Hart #0*"".` The two cards are correctly recognized by XSDB. The VCU1525 is installed in a Gen3x4 PCIe slot, could this be a problem?

I am confused and cannot understand if there is a need to setup XRT and DSA, since we only use the JTAG cable for the programming of the FPGA and not the PCIe.
Unfortunately since the VCU1525 is discontinued there is no support for the XRT or DSA for the most recent editions of the Vivado tools. Do you have any hints how to solve this problem?

Output log of `xbutil examine`

> XRT
  Version              : 2.13.466
  Branch               : 2022.1
  Hash                 : f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  Hash Date            : 2022-04-14 17:43:11
  XOCL                 : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  XCLMGMT              : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776

> Devices present
BDF             :  Shell                         Platform UUID                         Device ID       Device Ready*  
[0000:01:00.1]  :  xilinx_u250_gen3x16_base_4    F8DAC62E-49D9-B0AA-E9FC-6F260D9D0DFB  user(inst=129)  Yes            
[0000:08:00.1]  :  xilinx_vcu1525_xdma_201830_1  0x5bf05050                            user(inst=130)  No             
Devices that are not ready will have reduced functionality when using XRT tools

Thank you in advance!

eugene-tarassov, 2022-06-02:
VCU1525 port in this repo does not use XRT or DSA. No XRT installed, the board flash is empty - erased using Vivado command to avoid any surprises.

Only JTAG and QSFP are used to boot and debug Linux. XRT or DSA are not tested with this workflow, and may cause problems if you install them.

The board only needs power and PCIe clock from the host machine, nothing else. Actually, PCIe block can be removed from the design - it is not used yet, so the board would not need a host machine, only a power cable.

I have the board setup to run daily regression tests, using multiple RISC V configurations - it works pretty well.

ncppd, 2022-06-02:
Do you know how can I erase the board flash? Could this cause problems with your design?

ncppd, 2022-06-15:
I have not made any progress, and I cannot pinpoint the problem. Can you provide a bitstream file that you have verified that works correctly? If that works then I may have a problem with the bitstream generation. I am using Vivado 2020.2.

Could this error be caused by setting the wrong board_part or Xilinx_part? Could my VCU1525 board be faulty?

ncppd, 2022-06-15:
I tried it and I get the same error message with the bitstream you provided, so the problem is in my setup. I will try to get the bitstream in the board flash and let you know tomorrow. what happened If even that does not work should I check the DDR modules?

ncppd, 2022-06-16:
I flashed the board and the error persists. Should I -as a next step- check the DDR modules?

eugene-tarassov, 2022-06-16:
You can check DDR status in Vivado. It should look like this:

[image]

If you see calibration errors, you need a new DDR4 module.
If calibration not even started, the problem is something else - the flash or PCIe.

ncppd, 2022-06-17:
This is what it looks like:

[image]

On the other hand, on the U250 there are no calibration errors:

[image]

Do you think the VCU1525 board could be faulty?

eugene-tarassov, 2022-06-17:
It looks like VCU1525  FPGA has no memory controller enabled in its bitstream.
It is either wrong bitstream or something preventing the programming to start, like missing system clock or active system reset.
Try to reprogram FPGA using Program Device command in Vivado to see if it helps.

ncppd, 2022-06-17:
It does not help, programming results to the same warning. i used the command `make CONFIG=rocket64b2m BOARD=vcu1525`.

eugene-tarassov, 2022-06-17:
Then next most likely problem is active system reset.
One way to debug it is to modify the design to connect pcie_perstn and resetn to the board LEDs to see the state of reset signals.

closed 2023-05-30 by eugene-tarassov

tahir2018, 2025-01-22:
Hello, can someone who implemented design on U280 please comment, how to handle IIC interface issue, as IIC interface is not available on board level in u280 unlike u250:

For U280 following error is reported:
set IIC [ create_bd_cell -type ip -vlnv xilinx.com:ip:axi_iic:2.1 IIC ]
/IIC
  set_property -dict [ list \
   CONFIG.IIC_BOARD_INTERFACE {iic_main} \
   CONFIG.USE_BOARD_FLOW {true} \
 ] $IIC
ERROR: [IP_Flow 19-3461] Value 'iic_main' is out of the range for parameter 'IIC Board Interface(IIC_BOARD_INTERFACE)' for BD Cell 'IIC' . Valid values are - Custom
INFO: [IP_Flow 19-3438] Customization errors found on 'IIC'. Restoring to previous valid configuration.
INFO: [Common 17-17] undo 'set_property'
ERROR: [Common 17-39] 'set_property' failed due to earlier errors.

For U250, no error reported:
  set IIC [ create_bd_cell -type ip -vlnv xilinx.com:ip:axi_iic:2.1 IIC ]
/IIC
  set_property -dict [ list \
   CONFIG.IIC_BOARD_INTERFACE {iic_main} \
   CONFIG.USE_BOARD_FLOW {true} \
 ] $IIC