eugene-tarassov/vivado-risc-v #56: Using SFP module to connect Internet in VC707

xiongwu12345, 2021-11-30:
Hi, the Ethernet PHY hardware in our VC707 board seems to break down. So I want to use the SFP+ module and use an SFP port to RJ45 converter to use 1000 Mb/s Ethernet communications. Because we do not have 10 Gigabit Network. So what changes should be done to use the SFP+ module?

[image]

Thanks.

eugene-tarassov, 2021-11-30:
In IO/EthernetVC707/gig_ethernet_pcs_pma_0 IP configuration, change Board/ETHERNET from "sgmii" to "sfp sgmii".

closed 2022-06-26 by eugene-tarassov

Yuxin-Yu, 2023-03-24:
Hello, when using SFP for gigabit network communication, does the top-level signal also need to be replaced with an SFP interface? Is there any relevant routine reference？ @eugene-tarassov

eugene-tarassov, 2023-03-24:
Changing ETHERNET_BOARD_INTERFACE property of IO/EthernetVC707/gig_ethernet_pcs_pma_0 from "sgmii" to "sfp sgmii" is supposed to regenerate top-level signals automatically. You need to check and modify board/vc707/ethernet.xdc for pin assignments. I'm not sure about MDIO, I think you need to disable it in IO/Ethernet: enable_mdio = 0, and delete all related signals.

Yuxin-Yu, 2023-04-04:
Hello, I have found the IO/Ethernet: enable_mdio = 0 you mentioned, confirm if the relevant code is for these four pins?
```
set_property -dict { PACKAGE_PIN AH31 IOSTANDARD LVCMOS18 } [get_ports eth_mdio_clock]
set_property -dict { PACKAGE_PIN AK33 IOSTANDARD LVCMOS18 } [get_ports eth_mdio_data]
set_property -dict { PACKAGE_PIN AJ33 IOSTANDARD LVCMOS18 } [get_ports eth_mdio_reset]
set_property -dict { PACKAGE_PIN AL31 IOSTANDARD LVCMOS18 } [get_ports eth_mdio_int]
```

@eugene-tarassov

eugene-tarassov, 2023-04-04:
Yes, these four pins.

Yuxin-Yu, 2023-04-06:
Additionally, I noticed  sgmii_mgt_ clk_ clk_ p and sgmii_ mgt_ clk_ clk_ n signal in the VC707  uses an external 125MHz clock. Can this clock be generated by FPGA internal MMCM? I think it should be right. @eugene-tarassov

eugene-tarassov, 2023-04-06:
I believe gigabit transceivers cannot be clocked by MMCM. They have their own clock management and require an external oscillator.

Yuxin-Yu, 2023-04-06:
ok,thank you