eugene-tarassov/vivado-risc-v #57: Porting vivado-riscv-v to Alveo U250

ncppd, 2021-12-02:
I tried to port your design to a U250 board we have in the lab since the VCU1525 is similar to the U250. I changed all the board parts accordingly to `BOARD_PART  ?= xilinx.com:au250:part0:1.3` and `XILINX_PART ?= xcu250-figd2104-2L-e`. I am however running into an error and I am not sure how to get past it.

> ERROR: [IP_Flow 19-3478] Validation failed for parameter 'Enable channel(CHANNEL_ENABLE)' with value 'X1Y48' for IP 'gtwizard_ultrascale_0'. No supported channels found, valid channels are: X1Y16 X1Y17 X1Y18 X1Y19 X1Y20 X1Y21 X1Y22 X1Y23 X1Y24 X1Y25 X1Y26 X1Y27 X1Y28 X1Y29 X1Y30 X1Y31 X1Y40 X1Y41 X1Y42 X1Y43 X1Y44 X1Y45 X1Y46 X1Y47
INFO: [IP_Flow 19-3438] Customization errors found on 'gtwizard_ultrascale_0'. Restoring to previous valid configuration.
INFO: [Common 17-17] undo 'set_property'
ERROR: [Common 17-39] 'set_property' failed due to earlier errors.
while executing
"rdi::add_properties -dict {CONFIG.CHANNEL_ENABLE X1Y48 CONFIG.TX_MASTER_CHANNEL X1Y48 CONFIG.RX_MASTER_CHANNEL X1Y48 CONFIG.TX_LINE_RATE 10.3125 CONFI..."
    invoked from within
"set_property -dict [list \
    CONFIG.CHANNEL_ENABLE {X1Y48} \
    CONFIG.TX_MASTER_CHANNEL {X1Y48} \
    CONFIG.RX_MASTER_CHANNEL {X1Y48} \
    CONFI..."
    (file "../../board/u250/ethernet-u250.tcl" line 40)
while executing
"source ../../board/${vivado_board_name}/ethernet-${vivado_board_name}.tcl"
    (file "../../vivado.tcl" line 53)

Do you know which channel is best to use, or it does not matter? Thanks!

eugene-tarassov, 2021-12-02:
There is [Verilog Ethernet Alveo U250 Example Design](https://github.com/alexforencich/verilog-ethernet/tree/4aa672f8f391bb30cafe5a332e42b0c3432854e8/example/AU250/fpga_10g), and it looks like X1Y44 is one of 8 available 10G channels.

ncppd, 2021-12-06:
Thanks, that seems to be working!

closed 2021-12-06 by ncppd