eugene-tarassov/vivado-risc-v #58: Running baremetal programs on nexysvideo

violettttttt, 2021-12-09:
On  #15 Openocd is not working. So, is there any way to run an elf file on nexys video?

eugene-tarassov, 2021-12-09:
Sure.
You can copy the file to SD card. On reset, RISC-V CPU runs boot.elf file from the card.

Or, you can use XSDB (Xilinx System Debugger, aka XSCT - Xilinx System Command Tool) to run and debug an elf file:
```
xsdb% connect
tcfchan#1
xsdb% ta
  1  xcvu9p
     2  RISC-V
        3* Hart #0 (Running)
        4  Hart #1 (Running)
     5  Legacy Debug Hub
xsdb% ta 3
xsdb% rst
xsdb% stop
Info: Hart #0 (target 3) Stopped at 0x114b4 (Suspended)
xsdb% dow boot.elf
Downloading Program -- M:/projects/risc-v/vivado-risc-v/workspace/boot.elf
        section, .text: 0x80000000 - 0x8000fbbf
        section, .rodata: 0x80010000 - 0x80011a37
        section, .dynstr: 0x80011a38 - 0x80011c1a
        section, .gnu.hash: 0x80011c20 - 0x80011cf7
        section, .data: 0x80012000 - 0x80012a37
        section, .dynamic: 0x80012a38 - 0x80012b47
        section, .got: 0x80012b48 - 0x80012c17
        section, .got.plt: 0x80012c18 - 0x80012c27
        section, .htif: 0x80012c28 - 0x80012c37
        section, .dynsym: 0x80012c38 - 0x80012ebf
        section, .rela.dyn: 0x80012ec0 - 0x80013dbf
        section, .bss: 0x80014000 - 0x80034c4f
        section, .payload: 0x80200000 - 0x80275847
100%    0MB   0.3MB/s  00:01
Setting PC to Program Start Address 0x80000000
Successfully downloaded M:/projects/risc-v/vivado-risc-v/workspace/boot.elf
xsdb% con
Info: Hart #0 (target 3) Running
xsdb%
```

Or, if you prefer graphical UI, you can use Eclipse or Vitis (which is Eclipse based) with [TCF C/C++ Debugger](https://www.eclipse.org/tcf/).

Also, in Eclipse or Vitis, you can setup [CDT (C/C++ Development Tooling)](https://www.eclipse.org/cdt/) project to build the elf file and you will get complete edit/compile/debug workflow for baremetal RISC-V software development.

violettttttt, 2022-01-06:
I compiled a hello world program using riscv64-unknown-elf-gcc hello.c -o boot.elf. Copied this boot.elf to a sd card formatted as fat32. But output is not visible in the console window
[image]
.

eugene-tarassov, 2022-01-06:
> I compiled a hello world program using riscv64-unknown-elf-gcc hello.c -o boot.elf.

This does not look like baremetal program.
By definition, baremetal program includes device drivers, memory management code, statically linked libraries, etc.
At the very least, you need to add UART driver and a linker script with right memory addresses.

closed 2022-06-26 by eugene-tarassov