eugene-tarassov/vivado-risc-v #62: Support for Alveo U280

ZhaoYi1222, 2021-12-16:
I'm trying to port your design from VCU1525 to U280
I have successfully generated bitstream.
But when I try to run bare-metal application using Vitis xsct, there is no output in serial terminal.
I tried `mwr 0x60010004 0x41`, still without any output. Any advise?
Since there is no SD card support on U280,  is there a convenient way to make it support Linux?

eugene-tarassov, 2021-12-16:
> But when I try to run bare-metal application using Vitis xsct, there is no output in serial terminal.

The board has 4-port USB UART adapter. You need to connect to the right port, which is port 2, usually /dev/ttyUSB2.

> Since there is no SD card support on U280, is there a convenient way to make it support Linux?

Yes, using NFS.

You can setup NFS server on the development machine or any other machine on your local network, create a directory and make it exported by adding something like this into /etc/exports:
```
/home/nfsroot 192.168.0.0/24(rw,no_subtree_check,no_root_squash,no_all_squash)
```
Connect the board to 10G SFP+ switch using QSFP+ to 4x10G SFP+ cable. Download [rootfs.tar.gz](https://github.com/eugene-tarassov/vivado-risc-v/releases/download/v3.2.0/rootfs.tar.gz) from the Releases area and un-tar it to the NFS directory. Delete SD card mount from etc/fstab - the file should be empty.  During boot, RISC-V Linux kernel will use DHCP to get network info, then mount the NFS directory as the root file system.
Run make like this to boot Linux:

```
make clean
make BOARD=<your board name> CONFIG=<your RISC-V config> \
 ROOTFS_URL=<your NFS server IP>:<NFS directory to mount as root FS> \
 HW_SERVER_ADDR=<your HW server IP> JTAG_BOOT=1 jtag-boot
```

Also, see issue #55 for more details.

Rucadi, 2022-01-25:
I've made a fork where I put support for the Alveo U280: https://github.com/rucadi/vivado-risc-v based on the Alveo U250 Support (I wanted to do a pull-request after having it working)

However, somehow I'm getting rubbish on the UART port, instead of the correct characters. (My baudrate is 115200).

I've tested it with screen and with miniterm, no luck (and tested other common baudrates for the ifs)

The first bootrom message is shown (As rubbish) and with the scripts to download and boot linux it shows a lot of rubbish output.

Is there a configuration on the UART that maybe I'm missing? 

I'm using the following commands:
screen /dev/ttyUSB2 115200
and 
python3 -m serial.tools.miniterm  /dev/ttyUSB2   115200

[image]

eugene-tarassov, 2022-01-25:
It looks like wrong baud rate. UART clock input frequency should be 100MHz. Baud rate is configurable and set to 115200 by default. There is no other configuration. All clocks are derived from clk_user, which is SI570 programmable oscillator. It is expected to run at 156.25MHz. Perhaps, U280 has different oscillator? Or something in your system has re-configured the oscillator to run at different speed?

Rucadi, 2022-01-26:
The UART clock input frequency is 100MHz, and it comes from a clocking wizard.

The clocking wizard is fed with the clk_user port that comes from the SI570.
https://github.com/Xilinx/XilinxBoardStore/blob/2020.2/boards/Xilinx/au280/es1/1.1/part0_pins.xml

# SI570 user clock
```
<pin index="0"  name ="sysclk0_p" iostandard="LVDS" loc="BJ43" dqs_bias="TRUE"/>
<pin index="1"  name ="sysclk0_n" iostandard="LVDS" loc="BJ44" dqs_bias="TRUE"/>	

<pin index="2"  name ="sysclk1_p" iostandard="LVDS" loc="BH6"  dqs_bias="TRUE"/>
 <pin index="3"  name ="sysclk1_n" iostandard="LVDS" loc="BJ6"  dqs_bias="TRUE"/>
```

set_property -dict {PACKAGE_PIN BJ6 IOSTANDARD LVDS} [get_ports clk_user_clk_n]
set_property -dict {PACKAGE_PIN BH6 IOSTANDARD LVDS} [get_ports clk_user_clk_p]

# UART

And the UART is constrained as in 

```
<pin index="19" name ="USB_UART_TX"  iostandard="LVCMOS18" loc="B33"/>
<pin index="20" name ="USB_UART_RX"  iostandard="LVCMOS18" loc="A28"/>
```
		
set_property -dict {PACKAGE_PIN B33 IOSTANDARD LVCMOS18} [get_ports usb_uart_txd]
set_property -dict {PACKAGE_PIN A28 IOSTANDARD LVCMOS18} [get_ports usb_uart_rxd]

I doubt someone has re-configured the oscillator, but just to be sure, I'll try to derive the clock from the DDR clock and see if the problem persists, let's see if I can get this sorted out.

Rucadi, 2022-01-26:
Well, I just loaded the bitstream and now UART is working.

I spotted the main problem, that was that ALVEO U250 support uses a 300MHz clock as clk_user, while sysclk seems to be a 100MHz clock if I'm not wrong, connecting a clocking wizard to the DDR ui clock and taking the clocks from there worked.

I'm going to check that the QSFP is working and fix correctly all the issues, thanks for your help @eugene-tarassov

Rucadi, 2022-02-21:
I would like to merge the Alveo U280 Support, I have checked that the QSFP1 and it's working (Right now, with my resources I can't check QSFP0, but I assume it works as well).

About using the flash memory, unfortunately I can't spend time on it, since I have other priorities, and I've been told to not work on it.

Since I'm including the QSFP0 and the QSFP1 on the same bitstream, the **envvar**
`ETHER_MAC   ?= 00 0a 35 00 00 03`
is not working as expected, but since we have already the mac address on the device-tree prior to changing it, is it really necessary to do this **sed**?

While I also made changes on how sbi/u-boot is loaded into memory (in my case, it's embedded on a ROM), I think that it will be easier to only merge the U280 folder that should work with the current codebase.

eugene-tarassov, 2022-02-22:
> I would like to merge the Alveo U280 Support

It would be great.

> ETHER_MAC ?= 00 0a 35 00 00 03 is not working as expected

The idea was to allow to override MAC address on a command line, e.g. `make ETHER_MAC='00 0a 35 00 00 03' ...`. This does not work for multiple MACs. I have changed ETHER_MAC to be optional. If it is not set, MAC addresses are taken from the device tree.

> it will be easier to only merge the U280 folder that should work with the current codebase.

I agree.

nickbeckwith, 2022-04-08:
Hi @Rucadi, I saw your fork and appreciate the work. My team is also trying to put together a port for the U280. Do you know which commit has a minimal working product (even with only 1 QSFP port)? 

If you're busy, maybe I can fork off that commit and try to clean it up so that we only have to merge the board directory and rocket-freq file. Later, as my requirements evolve, I can try to merge in the additional support for both QSFP ports.

Rucadi, 2022-04-08:
Hi, I've not been working on this part for a while, due to other priorities.

My current fork works only with the configuration: Rocket64MeepConfig (you can change it a little).
Both QSFP are working.
OpenSBI+U-boot are embedded on the bitstream, so the bring-up of the machine is far easier (and you can see that it's working).

OpenSBI+U-boot embedded on the bitstream is performed by creating a ROM inside the rocket (and applying the  patch I made to do so). These are not in this parent repository and probably will not be, but it's helpful for your needs.

Something like:
make CONFIG=Rocket64MeepConfig BOARD=u280 vivado-gui
and creating the bitstream from the gui (for some reason it fails to do from the command line) is the way to go.

If you want, you could clean-up my code a little and prepare it for the merge, I can just accept your PR or give you access (or you can fork my repo).

If you just use the Linux Kernel generated by the repo and the bitstream as-is, you will already have the network-interfaces after successfully booting linux.

nickbeckwith, 2022-04-08:
Thanks for the response, especially on a Friday. I'll see what I can do getting your ROM version working. If I find myself with extra time, I'll submit a PR that should help merge it upstream. Have a good weekend!

nickbeckwith, 2022-04-12:
I was able to get the uboot prompt through your advice. Thank you! Could you share your boot configuration/boot command/args? I understand the NFS flow proposed for the U250, but am not sure if/where your flow differs.

Rucadi, 2022-04-13:
About the NFS flow, I think that eugene has not made drivers for U-boot, so I think that if you don't develop them (I suppose it's not too difficult replicating the linux-driver Eugene made) you can't boot from it yet.

I generate a debian image using debootstrap with this configuration, but we also use buildroot with success.

````conf
[General]
arch=riscv64
aptsources=Unstable Unreleased Sid-main
bootstrap=Unstable Unreleased Sid-main

[Unstable]
source=http://deb.debian.org/debian-ports
suite=unstable
omitdebsrc=true
keyring=debian-ports-archive-keyring
packages=build-essential  base-files  util-linux sudo base-passwd htop nano vim kmod bash base busybox dialog bison flex bc byobu

````

Once the rootfs is generated, I copy a script called gen.sh
(Double configure made more things for some reason, this part can be done on the RISC-V but is SLOW there)

```shell
export RUNLEVEL=1
export DEBIAN_FRONTEND=noninteractive

dpkg --configure -a
dpkg --configure -a
apt clean
rm -rf /var/lib/apt/lists/*

rm -rf  /usr/share/doc/* 
rm -rf /usr/share/groff/* /usr/share/info/*
rm -rf /usr/share/lintian/* /usr/share/linda/* /var/cache/man/*
rm -rf /usr/share/man/*

```

and init:
```
#!/bin/sh
mount -t proc none /proc
mount -t sysfs none /sys
mount -t devtmpfs none /dev

export PATH=$PATH:/usr/local/sbin/
export PATH=$PATH:/sbin
export PATH=$PATH:/usr/sbin/

exec /bin/bash

```

To the / of the rootfs.

Then I just call the reconfiguration script.

```make

configure_packages: rootfs 
	cp `which qemu-riscv64-static` rootfs/bin/
	update-binfmts --import blobs/qemu-riscv64 #this just configures qemu binfmt, may be not needed
	PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin chroot $(realpath rootfs) qemu-riscv64-static /bin/bash /gen.sh
```

And finally create a .cpio.gz of it (This will be our initramfs).
The file blob.cpio  you can see I'm using is just a cpio containing console and loop0 created and packed previously with mknod, since it requires root privileges to create this files, it is just easier to have them already packed.

```make
rootfs.cpio.gz: rootfs configure_packages
	cd rootfs &&  find . -print0 | cpio --null -ov --owner +0:+0 --format=newc > ../rootfs_only.cpio 
	cat blobs/blob.cpio rootfs_only.cpio > rootfs.cpio
	rm rootfs_only.cpio
	gzip -f rootfs.cpio
```

Then I make a U-boot FIT Image containing the device-tree, kernel and the rootfs.cpio.gz.

Upload the fit image to the machine using QDMA (dma-to-device), to, for example, 0x9000_0000

Prepare the bootargs on U-boot and boot:

```
env set bootm_size 0x3000000
env set bootargs rw root=/dev/ram0 earlycon console=ttyAU0,115200n8 locale.LANG=en_US.UTF-8
bootm 0x90000000

```

If you decide to go for another rootfs, just generate the fit image with it, and you may be good to go.
Sometimes it can be a little tricky to get things working, tho.

eugene-tarassov, 2022-04-14:
> About the NFS flow, I think that eugene has not made drivers for U-boot

Just to clarify, an Ethernet driver is needed in u-boot to support TFTP, which can be used to download Linux kernel, and, optionally, RAM rootfs. This is unrelated to NFS. Alternatively, the kernel can be downloaded over JTAG or QDMA.

After the kernel starts, it can mount rootfs either in RAM or over NFS, depending on bootargs.

zhileiz, 2022-05-02:
> Then I make a U-boot FIT Image containing the device-tree, kernel and the rootfs.cpio.gz.
> Upload the fit image to the machine using QDMA (dma-to-device), to, for example, 0x9000_0000
> Prepare the [...]

Thank you @Rucadi for sharing this process. I tried to replicate your method but did not succeed. Here is what I get from uboot on a U280 board:

[image]

I wonder if you could elaborate a little more on how you created the FIT image so I can check at which step did my approach differ from yours. Thank you very much.

**(To give you some more context, here is what I did:)**
To generate the FIT image, I prepared the rootfs with busybox and this command:
```
find -print0 | cpio -0oH newc | gzip -9 > initramfs.cpio.gz
```
I prepared the kernel image by compiling the latest stable kernel using ARCH=RISCV64, and found the resulting image at /arch/riscv/boot/Image. 
After grabbing the device tree (dtb file), I wrote a file named "u280-fit.its" with content:
```
/dts-v1/;

/ {
        description = "U280-Custom RISCV FIT Image";
        #address-cells = <1>;

        images {
                kernel {
                        description = "Kernel";
                        data = /incbin/("<some path...>/Image");
                        type = "kernel";
                        arch = "riscv";
                        os = "linux";
                        compression = "none";
                        load = <0x80480000>;
                        entry = <0x80480000>;
                        hash {
                                algo = "sha1";
                        };
                };
                fdt {
                        description = "DTB";
                        data = /incbin/("<some path...>/system.dtb");
                        type = "flat_dt";
                        arch = "riscv";
                        compression = "none";
                        load = <0x83000000>;
                        entry = <0x83000000>;
                        hash {
                                algo = "sha1";
                        };
                };
                initrd {
                        description = "Initrd";
                        data = /incbin/("<some path...>/initramfs.cpio.gz");
                        type = "ramdisk";
                        arch = "riscv";
                        os = "linux";
                        compression = "none";
                        hash {
                                algo = "sha1";
                        };
                };
        };

        configurations {
                default = "standard";
                standard {
                        description = "Standard Boot";
                        kernel = "kernel";
                        fdt = "fdt";
                        ramdisk = "initrd";
                        hash {
                                algo = "sha1";
                        };
                };
        };

}; 
```
The mkimage command I then used was thus:
```
mkimage -f u280-fit.its myfit.itb   
```
which successfully creates the itb file with this description:
```
FIT description: U280-Custom RISCV FIT Image
Created:         Mon May  2 17:19:38 2022
 Image 0 (kernel)
  Description:  Kernel
  Created:      Mon May  2 17:19:38 2022
  Type:         Kernel Image
  Compression:  uncompressed
  Data Size:    15570944 Bytes = 15206.00 KiB = 14.85 MiB
  Architecture: RISC-V
  OS:           Linux
  Load Address: 0x80480000
  Entry Point:  0x80480000
  Hash algo:    sha1
  Hash value:   844af19d0c17d48751f7c4c308bc4ef9f3292afc
 Image 1 (fdt)
  Description:  DTB
  Created:      Mon May  2 17:19:38 2022
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    3836 Bytes = 3.75 KiB = 0.00 MiB
  Architecture: RISC-V
  Load Address: 0x83000000
  Hash algo:    sha1
  Hash value:   5a5a795c269c21c5279dfbd635867768a205163f
 Image 2 (initrd)
  Description:  Initrd
  Created:      Mon May  2 17:19:38 2022
  Type:         RAMDisk Image
  Compression:  uncompressed
  Data Size:    800889 Bytes = 782.12 KiB = 0.76 MiB
  Architecture: RISC-V
  OS:           Linux
  Load Address: unavailable
  Entry Point:  unavailable
  Hash algo:    sha1
  Hash value:   f7f182141d17e89f62f78bab2fcd32afbcc5cad2
 Default Configuration: 'standard'
 Configuration 0 (standard)
  Description:  Standard Boot
  Kernel:       kernel
  Init Ramdisk: initrd
  FDT:          fdt
  Hash algo:    sha1
  Hash value:   unavailable
```
I then passed this itb file to the u280 memory using QDMA:
```
dma-to-device -d /dev/qdmab3000-MM-1 -a 0x88000000 -f myfit.itb -v -s 32 -c 51180
```
The 32 and 51180 were was determined by the size of the itb, which is 32 * 51180 = 16377656 bytes.

After the itb transfer, I ran your boot commands and `bootm 0x88000000` and resulted in that error that we saw. Do you spot any problems with this approach? What would you suggest I try next to debug this? Thanks.

Greatly appreciate any help and/or suggestions!

Rucadi, 2022-05-03:
Check bootm_low and bootm_size environment on u-boot, 
it looks like you might be overwriting your own image. 

I would probably try to put them higher in memory.

You probably have a lot of memory to use, if that the case, try defining better the regions used by u-boot and your image, also I  think that maybe you have your Initrd badly specified.

In my case, I set the compression as gzip and have a load attribute, while your compression seems to be "none"

```
/dts-v1/;

/ {
        description = "U-boot FIT image RISCV";
        #address-cells = <2>;

        images {

                kernel {
                        description = "Linux kernel";
                        data = /incbin/("Image.gz");
                        type = "kernel";
                        arch = "riscv";
                        os = "linux";
                        load = <0x80200000>;
                        entry = <0x80200000>;
                        compression = "gzip";
                        hash-1 {
                                algo = "sha256";
                        };
                };

                fdt {
                        description = "FDT";
                        data = /incbin/("riscv.dtb");
                        type = "flat_dt";
                        arch = "riscv";
                        load = <0x1 0x0>;
                        compression = "none";
                        hash-1 {
                                algo = "sha256";
                        };
                };

                ramdisk {
                        description = "initramfs";
                        data = /incbin/("rootfs.cpio.gz");
                        type = "ramdisk";
                        arch = "riscv";
                        os = "linux";
                        load = <0x82000000>;
                        compression = "gzip";
                };

        };

        configurations {
                default = "config-1";

                config-1 {
                        description = "RiscV";
                        kernel = "kernel";
                        fdt = "fdt";
                        ramdisk = "ramdisk";
                };
        };
};

```

zhileiz, 2022-05-04:
Thank you @Rucadi,

Using your its file, booting worked for me with two additional tweaks:
1) The kernel .config file needs to have CONFIG_BLK_DEV_RAM=y 
2) The bootargs need to include rootfstype=ramfs

If the first is not done, the kernel will boot but panic at /dev/ram0 not found. The linux.config file in the patch folder in this repo does not have that CONFIG_BLK_DEV_RAM set.

If the second is not done, the kernel will find /dev/ram0 but will still panic after trying a bunch of file system formats.

Currently, I am stuck at the init process:
[image]

If anyone could shed a light on how this could be resolved, that would be great!

zhileiz, 2022-05-09:
> I hope this serves you, if you need more support I don't mind talking through a chat or something.

Thank you @Rucadi for your generous help. I sent a message to your email a few days back. Were you able to receive it? If not, what would be your preferred way to be reached? I would love to chat more about this issue with you. My email is zhileiz@seas.upenn.edu.

zhileiz, 2022-05-11:
@Rucadi , I sent another message to you. But either way, would it be possible for you to let share the linux kernel version as well as the corresponding .config file that you ended up using? The patch in the repo seems to require a few addtional manual resolution of in menuconfig (such as support for initramfs). Thanks.

Rucadi, 2022-05-11:
Hi! I think you could try to use the repo I just published:
https://github.com/Rucadi/RiscV-ROOTFS

This is the rootfs generator that I made using multistrap and qemu, it should generate all you need (With this you get the same environment as me).

If you have any doubt don't hesitate to answer.

MrJimbo2002, 2023-08-14:
Hi Eugene, many thanks for your such detailed open-source project sharing. 

I am new to FPGA and Vivado and please accept my apologies asking not deep questions:

I want to port your U250 design to U280 but encountered the problems below:
[image]
[image]
[image]

I have checked with the documents of QSFP28 Interface, but not sure how to modify the LOC parameter of "dict" as N4, N3 ... AU22 listed in top.xdc. 

In terms of the Memory Core Error, I attempted to grep riscv_i/DDR/ddr4_0 and seems that there is no indication of SRL(Super Regional Logic block) and hence no I/O banks for each SLR were found. 

Could you help me to debug the above critical warnings and error messages?

Thank you very much in advance for your time commitment and help~