eugene-tarassov/vivado-risc-v #63: make update submodules not doing anything

salmansheikh, 2022-01-04:
It seems to be sitting here and not doing anything:

 make update-submodules
git -c submodule.torture.update=none -c submodule.software/gemmini-rocc-tests.update=none -c submodule.software/onnxruntime-riscv.update=none submodule update --init --force --recursive
Cloning into '/home/ssheikh/vivado-risc-v/linux-stable'...

eugene-tarassov, 2022-01-04:
One possibility is your firewall is blocking git port: TCP 9418.
As a workaround, you can try to change Linux repo URL in .gitmodules to
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git

salmansheikh, 2022-01-05:
Hello,

I am on another machine and i changed the URL in .gitmodules to what you
said (remove -stable from the URL) and it is still hanging on the same
part. Very strange. I don't think I have that port blocked at home if it is
indeed blocked. Or maybe my router automatically does that.

eugene-tarassov, 2022-01-05:
> i changed the URL in .gitmodules to what you said (remove -stable from the URL)

Actually, the new URL uses different protocol - HTTPS instead of GIT, in addition to different path.
The idea is HTTPS is very unlikely to be blocked.

Also, it needs to be changed before you run make update-submodules first time.
Git appears to record URLs somewhere in .git and ignore changes after that.
All other modules already use HTTPS, I'm going to commit this change for consistency.

salmansheikh, 2022-01-05:
Okay, I tried again and it skips linux-stable and then has this output:

make update-submodules
git -c submodule.torture.update=none -c
submodule.software/gemmini-rocc-tests.update=none -c
submodule.software/onnxruntime-riscv.update=none submodule update --init
--force --recursive
Submodule 'ethernet/verilog-ethernet' (
https://github.com/alexforencich/verilog-ethernet.git) registered for path
'ethernet/verilog-ethernet'
Submodule 'generators/gemmini' (https://github.com/ucb-bar/gemmini)
registered for path 'generators/gemmini'
Submodule 'generators/riscv-boom' (
https://github.com/riscv-boom/riscv-boom.git) registered for path
'generators/riscv-boom'
Submodule 'generators/sifive-cache' (
https://github.com/sifive/block-inclusivecache-sifive.git) registered for
path 'generators/sifive-cache'
Submodule 'generators/testchipip' (https://github.com/ucb-bar/testchipip.git)
registered for path 'generators/testchipip'
Submodule 'linux-stable' (
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git)
registered for path 'linux-stable'
Submodule 'opensbi' (https://github.com/riscv/opensbi.git) registered for
path 'opensbi'
Submodule 'rocket-chip' (https://github.com/ucb-bar/rocket-chip.git)
registered for path 'rocket-chip'
Submodule 'u-boot' (https://github.com/u-boot/u-boot.git) registered for
path 'u-boot'
Cloning into '/home/ssheikh/vivado-risc-v/ethernet/verilog-ethernet'...
Cloning into '/home/ssheikh/vivado-risc-v/generators/gemmini'...
Cloning into '/home/ssheikh/vivado-risc-v/generators/riscv-boom'...
Cloning into '/home/ssheikh/vivado-risc-v/generators/sifive-cache'...
Cloning into '/home/ssheikh/vivado-risc-v/generators/testchipip'...
Cloning into '/home/ssheikh/vivado-risc-v/linux-stable'...
fatal: unable to access '
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/': server
certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt
CRLfile: none
fatal: clone of '
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git' into
submodule path '/home/ssheikh/vivado-risc-v/linux-stable' failed
Failed to clone 'linux-stable'. Retry scheduled
Cloning into '/home/ssheikh/vivado-risc-v/opensbi'...
Cloning into '/home/ssheikh/vivado-risc-v/rocket-chip'...
Cloning into '/home/ssheikh/vivado-risc-v/u-boot'...
Cloning into '/home/ssheikh/vivado-risc-v/linux-stable'...
fatal: unable to access '
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/': server
certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt
CRLfile: none
fatal: clone of '
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git' into
submodule path '/home/ssheikh/vivado-risc-v/linux-stable' failed
Failed to clone 'linux-stable' a second time, aborting
Makefile:26: recipe for target 'update-submodules' failed
make: *** [update-submodules] Error 1

eugene-tarassov, 2022-01-05:
I'm not sure why https://git.kernel.org causes "certificate verification failed". It works fine for me. The site is official upstream origin of all Linux distributions, it is very unlikely that it is broken. There must be something wrong with your machine or network. I hope you use Ubuntu 20 LTS machine as recommended in the README, otherwise you may encounter all kinds of compatibility problems.

salmansheikh, 2022-01-05:
Ok, thanks. I will upgrade to Ubuntu 20.04

closed 2023-05-30 by eugene-tarassov