eugene-tarassov/vivado-risc-v #66: Warning messages

salmansheikh, 2022-01-07:
Okay, I got Ubuntu 20.04 (WSL) and Vivado installed. I got to this point where it gave a bunch of messages to regenerate some files that aren't found. Is there a way to open this design in the gui and see what's going on?

Salman

make CONFIG=rocket64b2 BOARD=vc707 bitstream
echo "open_project workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.xpr" >workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
echo "update_compile_order -fileset sources_1" >>workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
echo "set_param general.maxThreads 1" >>workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
echo "reset_run synth_1" >>workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
echo "launch_runs -jobs 1 synth_1" >>workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
echo "wait_on_run synth_1" >>workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64b2/vivado-vc707-riscv/make-synthesis.tcl
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/tools/Xilinx/Vivado/2021.2/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'riscv_axi_smc_1_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_axi_smc_1_0/riscv_axi_smc_1_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_axi_smc_1_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_axi_smc_1_0/riscv_axi_smc_1_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_axi_smc_1_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_axi_smc_1_0/riscv_axi_smc_1_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_axi_smc_1_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_axi_smc_1_0/riscv_axi_smc_1_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_axi_smc_1_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_axi_smc_1_0/riscv_axi_smc_1_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mem_reset_control_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mem_reset_control_0_0/riscv_mem_reset_control_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mem_reset_control_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mem_reset_control_0_0/riscv_mem_reset_control_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mem_reset_control_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mem_reset_control_0_0/riscv_mem_reset_control_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mem_reset_control_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mem_reset_control_0_0/riscv_mem_reset_control_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mem_reset_control_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mem_reset_control_0_0/riscv_mem_reset_control_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_mig_7series_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/riscv_mig_7series_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_Ethernet_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_Ethernet_0/riscv_Ethernet_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_Ethernet_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_Ethernet_0/riscv_Ethernet_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_Ethernet_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_Ethernet_0/riscv_Ethernet_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_Ethernet_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_Ethernet_0/riscv_Ethernet_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_Ethernet_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_Ethernet_0/riscv_Ethernet_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_ethernet_stream_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_ethernet_stream_0_0/riscv_ethernet_stream_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_ethernet_stream_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_ethernet_stream_0_0/riscv_ethernet_stream_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_ethernet_stream_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_ethernet_stream_0_0/riscv_ethernet_stream_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_ethernet_stream_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_ethernet_stream_0_0/riscv_ethernet_stream_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_ethernet_stream_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_ethernet_stream_0_0/riscv_ethernet_stream_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_gig_ethernet_pcs_pma_0_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_gig_ethernet_pcs_pma_0_0/riscv_gig_ethernet_pcs_pma_0_0.dcp'. Please regenerate to continue.
[... cut ...]
WARNING: [IP_Flow 19-3664] IP 'riscv_sys_diff_clock_buf_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_sys_diff_clock_buf_0/riscv_sys_diff_clock_buf_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'riscv_sys_diff_clock_buf_0' generated file not found '/home/ssheikh/vivado-risc-v/workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.gen/sources_1/bd/riscv/ip/riscv_sys_diff_clock_buf_0/riscv_sys_diff_clock_buf_0_sim_netlist.vhdl'. Please regenerate to continue.

eugene-tarassov, 2022-01-07:
> Is there a way to open this design in the gui and see what's going on?
```
make CONFIG=rocket64b2 BOARD=vc707 vivado-gui
```
The Vivado project looks corrupted. There must be an error message somewhere before these warnings.

salmansheikh, 2022-01-07:
Seems like Vivado is crashing inside of WSL. I opened the project inside of Vivado gui and tried regenerating thing in synth.
 Might be something with WSL that is causing it. I unfortunately don't have a Vivado ML license with my Ubuntu system at home.  Is there an easy way I can run the make script to just generate the project in Linux and then move things over to Windows Vivado and run the bitstream generation? 

#
# An unexpected error has occurred (11)
#
Stack:
/opt/tools/Xilinx/Vivado/2021.2/tps/lnx64/jre11.0.11_9/lib//server/libjvm.so(+0xbefecb) [0x7f858f191ecb]
/opt/tools/Xilinx/Vivado/2021.2/tps/lnx64/jre11.0.11_9/lib//server/libjvm.so(JVM_handle_linux_signal+0xd1) [0x7f858f198c81]
/opt/tools/Xilinx/Vivado/2021.2/tps/lnx64/jre11.0.11_9/lib//server/libjvm.so(+0xbead23) [0x7f858f18cd23]
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f85eb9b6210]
/lib/x86_64-linux-gnu/libc.so.6(malloc_usable_size+0x48) [0x7f85eba0f378]
/lib/x86_64-linux-gnu/libudev.so.1(+0x10319) [0x7f856a670319]
/lib/x86_64-linux-gnu/libudev.so.1(+0x167c4) [0x7f856a6767c4]
/lib/x86_64-linux-gnu/libudev.so.1(+0x1b255) [0x7f856a67b255]
/lib/x86_64-linux-gnu/libudev.so.1(+0x1b4bb) [0x7f856a67b4bb]
/lib/x86_64-linux-gnu/libudev.so.1(udev_enumerate_scan_devices+0x294) [0x7f856a67dd84]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(+0x129015) [0x7f85e3419015]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(xilinxd_52bd866351b78202+0x9) [0x7f85e3419499]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(+0xd6317) [0x7f85e33c6317]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(xilinxd_52bd862318b59a70+0x86) [0x7f85e33c60d6]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(+0xc364f) [0x7f85e33b364f]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(xilinxd_52bd9e9e1c8e52fb+0x1b) [0x7f85e33bd3eb]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libXil_lmgr11.so(xilinxd_52bd700d1bd3c616+0x30) [0x7f85e33bd480]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonxillic.so(XilReg::Utils::GetHostInfo[abi:cxx11](XilReg::Utils::HostInfoType, bool) const+0x1c8) [0x7f85e77976d
8]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonxillic.so(XilReg::Utils::GetHostInfoFormatted[abi:cxx11](XilReg::Utils::HostInfoType, bool) const+0x52) [0x7f8
5e779b2e2]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonxillic.so(XilReg::Utils::GetHostInfo[abi:cxx11]() const+0x183) [0x7f85e779b5a3]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonxillic.so(XilReg::Utils::GetRegInfo(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<ch
ar> > const&, bool, bool)+0xc6) [0x7f85e77a8856]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonxillic.so(XilReg::Utils::GetRegInfoWebTalk(std::__cxx11::basic_string<char, std::char_traits<char>, std::alloc
ator<char> > const&)+0x60) [0x7f85e77a8ae0]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_project.so(HAPRWebtalkHelper::getRegistrationId[abi:cxx11]() const+0x3a) [0x7f85a94e63da]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_project.so(HAPRWebtalkHelper::HAPRWebtalkHelper(HAPRProject*, HAPRDesign*, HWEWebtalkMgr*, std::__cxx11::basic_strin
g<char, std::char_traits<char>, std::allocator<char> > const&)+0x158) [0x7f85a94e6cd8]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_tcltasks.so(+0x1ae525e) [0x7f85ddc6525e]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_tcltasks.so(+0x1af4a4d) [0x7f85ddc74a4d]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_common.so(+0xa80712) [0x7f85ecc40712]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x3356f) [0x7f85e6fc356f]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x76945) [0x7f85e7006945]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x7e0f9) [0x7f85e700e0f9]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(TclEvalObjEx+0x76) [0x7f85e6fc5216]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_common.so(+0xa7efff) [0x7f85ecc3efff]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(Tcl_ServiceEvent+0x7f) [0x7f85e7037bef]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(Tcl_DoOneEvent+0x154) [0x7f85e7037f24]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commontasks.so(+0x24aa85) [0x7f85e054aa85]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commontasks.so(+0x24dc6d) [0x7f85e054dc6d]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commontasks.so(+0x24ed0e) [0x7f85e054ed0e]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_common.so(+0xa80712) [0x7f85ecc40712]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x3356f) [0x7f85e6fc356f]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x76945) [0x7f85e7006945]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(+0x7e0f9) [0x7f85e700e0f9]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(TclEvalObjEx+0x76) [0x7f85e6fc5216]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_commonmain.so(+0xce23) [0x7f85ebface23]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/libtcl8.5.so(Tcl_Main+0x1d0) [0x7f85e70302f0]
/opt/tools/Xilinx/Vivado/2021.2/lib/lnx64.o/librdi_common.so(+0xac239b) [0x7f85ecc8239b]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f85eb956609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f85eba92293]

eugene-tarassov, 2022-01-07:
Yes, you can re-create the project in Windows Vivado GUI and do everything there.
Assuming the files are visible in Windows, delete the project directory (e.g. workspace/rocket64b2/vivado-vc707-riscv), start Vivado, go to the Tcl Console and run the project creation script (change the path to match you setup):
```
source c:/vivado-risc-v/workspace/rocket64b2/system-vc707.tcl
```

You will still need Linux to generate the script:
```
make CONFIG=rocket64b2 BOARD=vc707 vivado-tcl
```

salmansheikh, 2022-02-01:
Finding the path via cmd shell to the WSL directory of Ubuntu 20.04 wasn't easy and now seems like path is too long..sigh. 

[image]

eugene-tarassov, 2022-02-01:
I have updated README to include info on running Ubuntu on Windows. HTH.

salmansheikh, 2022-02-01:
Awesome, thanks!

eugene-tarassov, 2022-02-01:
> Do i need to install another drive at D?

No, you can use any drive. Just make sure it is mounted in WSL with metadata support and use a short path.

salmansheikh, 2022-02-01:
Well C is already used. And when I created /mnt/d it didn't work.

eugene-tarassov, 2022-02-01:
You can use C.

See [https://askubuntu.com/questions/1115564/wsl-ubuntu-distro-how-to-solve-operation-not-permitted-on-cloning-repository](https://askubuntu.com/questions/1115564/wsl-ubuntu-distro-how-to-solve-operation-not-permitted-on-cloning-repository)

Actually, metadata might be already enabled in latest versions of WSL. So, if you use /mnt/c without re-mounting, it might just work. Just don't use /home.

eugene-tarassov, 2022-02-03:
Yes, riscv-vc707.bit in the Releases area [https://github.com/eugene-tarassov/vivado-risc-v/releases](https://github.com/eugene-tarassov/vivado-risc-v/releases)

salmansheikh, 2022-02-10:
Do you have the mcs file for it?

eugene-tarassov, 2022-02-10:
> Do you have the mcs file for it?

No.
write_cfgmem command in Vivado can convert bit to mcs.

> I coped the debian-riscv64.sd.img using WinDiskImager to the SD card

It is dd image (see [mk-sd-card](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/mk-sd-card) for details):

```
sudo dd if=debian-riscv64.sd.img of=<SD card device> bs=64K status=progress
```
I don't think WinDiskImager supports dd images.

closed 2022-06-26 by eugene-tarassov