eugene-tarassov/vivado-risc-v #74: BOOT failed!!!

codoor, 2022-03-09:
I want to migrate to vcu118 and run Linux. I did this based the vc707 project and remove the enternet in the block design. I also modified the DDR ip.
But when I programmed the bitstream and then boot the linux from the sd card, the process of the boot stopped. Can you give me some suggestion for this. Does the network is necessary?
The serial port show as follow:
[image]

eugene-tarassov, 2022-03-09:
The network is not necessary to boot Linux.
The error message "unable to select a mode" means U-Boot cannot access SD card.
The message comes from sd_select_mode_and_width() function in U-Boot.

codoor, 2022-03-11:
The function sd_select_mode_and_width() is seems to programm the parameter of bus. I'm not familiar with the u-boot, can you tell me what i should do to fix this.

eugene-tarassov, 2022-03-11:
It fails to switch from 1-lane slow clock to 4-lane fast clock SD bus mode. So, it looks like the bus is not working: either wrong pins assignments, or timing constraints violations, or mechanical problem in the socket, etc.

These are common issues when porting a project to a new board. Usual ways to debug: double check constraints files, use integrated logic analyzer, use oscilloscope, use XSDB to step through the driver, etc.

langccc, 2022-04-21:
> The function sd_select_mode_and_width() is seems to programm the parameter of bus. I'm not familiar with the u-boot, can you tell me what i should do to fix this.

Did you solve the problem,I got the same with you,I removed eth both in device tree and block design,and boot failed:
[image]

codoor, 2022-04-22:
I did not solve this problem. It guess the reason for this is a protocol mismatch, but i do not have enough time to modify the protocol.  So I try the chipyard's vcu118 bitstream and system boot(because that is use spi and this is use sdio), and it works. and  i also want to try to change to spi in the future.

> > The function sd_select_mode_and_width() is seems to programm the parameter of bus. I'm not familiar with the u-boot, can you tell me what i should do to fix this.
> 
> Did you solve the problem,I got the same with you,I removed eth both in device tree and block design,and boot failed: [image]

langccc, 2022-04-22:
@codoor 
Thank you for reminding me，I remember I uesd to meet a error with  axi smartconnect ，So I changed axi smartconnect to axi interconnect ，and then it can access SD card ，but still hang in another place：
[image]

closed 2022-06-26 by eugene-tarassov

Yuxin-Yu, 2024-06-05:
Hello @codoor , can I refer to your SD card configuration? I started Linux using VCU118, but couldn't detect the SD card. The error is as follows:
```
RISC-V 64, Boot ROM V3.6
Cannot access SD: Timeout
Cannot mount SD: Disk not ready
```