eugene-tarassov/vivado-risc-v #75: Running gemmini on KC705

KUEE1080, 2022-03-15:
I am trying to run gemmini on the fpga board and considering to purchase KC705 board to run on it. Do you think it is a good alternative instead of purchasing VC707 board?

eugene-tarassov, 2022-03-15:
Yes, KC705 is about as good as VC707.
I, personally, prefer Digilent Genesys 2 - same FPGA as KC705, but less expensive.
Of course, the best to run Gemmini emulation would be one of Alveo boards - they have much bigger FPGA.

KUEE1080, 2022-03-15:
Thanks a lot. Do you recommend any specific Alveo boards that you prefer if you emulate Gemmini? Because the hardware requirements specified in the readme only mentions 'Xilinx VC707 or Xilinx KC705 or Digilent Genesys 2 or Digilent Nexys Video or Digilent Nexys A7 100T board'.

eugene-tarassov, 2022-03-15:
Alveo boards are not listed in the README, because they don't have SD card slot and require to setup NFS server to boot Linux - I have not documented this workflow yet. However, this repo includes support for Alveo U250 and VCU1525 (same as Alveo U200). See, for example, discussions #55 and #62 about porting RISC-V to Alveo boards.

> Do you recommend any specific Alveo boards

I don't really recommend any specific Alveo board. My point is that you will need a bigger FPGA to run Gemmini configurations with more than a few nodes. And the easiest way to get high-end FPGA is to use an Alveo board.

KUEE1080, 2022-03-17:
I really appreciate your fruitful comments..! Thanks a lot.

closed 2022-06-26 by eugene-tarassov