eugene-tarassov/vivado-risc-v #77: Is device tree necessary for building FPGA bitstream?

langccc, 2022-04-13:
I'm tring to use Vivado Block Design to add an IP on Windows, and I see device tree  need to be updated first before building bitstream. The issue is that, I used Vivado GUI on Windows to model the design and generate bitstream ，but the vivado project doesn't seem to include device tree file. It makes me confused，can you help me please！

eugene-tarassov, 2022-04-14:
The device tree is linked with bootrom, then bootrom is imported into the RocketChip Chisel code. So, if you modify the device tree, you need to regenerate bootrom and your RocketChip HDL  - the Verilog file generated from Chisel.

The Makefile knows all these dependencies, so you can just run make like this:
```
make CONFIG=rocket64b2 BOARD=nexys-video vivado-tcl
```
And RocketChip HDL file (e.g. system-nexys-video.v) will be updated with new device tree.

You don't need to change the Vivado project, just rebuild bitstream with the new HDL.

langccc, 2022-04-14:
> 

Thank you very much！I understand now.

closed 2022-04-14 by eugene-tarassov