eugene-tarassov/vivado-risc-v #78: How to get a reserved-memory

flyxia624, 2022-04-14:
I need to reserve some memory for the accelerator I added. According to a similar project, I modified the device tree in board/vcu1525/bootrom.dst by adding code like this:
/ {
    #address-cells = <1>;
    #size-cells = <1>;

    reserved-memory {
        #address-cells = <1>;
        #size-cells = <1>;
        ranges;
		reserved: buffer@a0000000 {
			 no-map;
			 reg = <0xa0000000 0x10000000>;
		};
    };
};
I want to know if I'm right，Thank you.
BTW，I noticed that the vcu1525 block design has a peripheral device IIC. But the device tree does not contain this node，Does this IIC work without driver?

eugene-tarassov, 2022-04-14:
> I want to know if I'm right

It looks OK to me.

> I noticed that the vcu1525 block design has a peripheral device IIC.

It is not used and not supported yet.

flyxia624, 2022-04-15:
Thanks for your reply.I see

closed 2022-04-15 by flyxia624