eugene-tarassov/vivado-risc-v #79: env: vivado: No such file or directory

KUEE1080, 2022-04-18:
Hi, I am facing this issue while I am compiling FPGA bitstream. Has anyone experienced this kind of issues?? I am really waiting for help. Any reply will help me a lot. Thanks in advance.

My environment

Ubuntu 16.04 LTS
Vivado 2020.1

`
(base) via@via-fpga:/home/username/vivado-risc-v$ sudo make CONFIG=rocket64b2gem BOARD=genesys2 bitstream
if [ ! -e workspace/rocket64b2gem/vivado-genesys2-riscv ] ; then env XILINX_LOCAL_USER_DATA=no vivado -mode batch -nojournal -nolog -notrace -quiet -source workspace/rocket64b2gem/system-genesys2.tcl ; fi
env: ‘vivado’: No such file or directory
Makefile:271: recipe for target 'workspace/rocket64b2gem/vivado-genesys2-riscv/timestamp.txt' failed
make: *** [workspace/rocket64b2gem/vivado-genesys2-riscv/timestamp.txt] Error 127
`

KUEE1080, 2022-04-18:
Sorry. This was somewhat related to my server's permisson issue. Closing this issue.

closed 2022-04-18 by KUEE1080

LeooLevinn, 2023-05-01:
Hi KUEE, I am facing the exact same thing.. How did you work things out plz?

LeooLevinn, 2023-05-01:
Never  mind  :) I redo the repo in /home/<user> and it worked... Previously it was under /home