eugene-tarassov/vivado-risc-v #80: How to configure DDR and SD on KCU105

langccc, 2022-04-20:
I follwed VC707 to port KCU105，and modified DDR followed from U250，but without axi_s_ddr_ctrl interface,like this:
[image]
Is it rignt?
Secondly,can you tell me how to constraint SD with KCU105,I just modified PIN name from VC707 to KCU105,but get errors like this:
`[DRC AVAL-248] OBUFT_has_two_FFs_with_IOB: The OBUFT riscv_i/IO/SD/inst/IOBUF_cmd has I (data) pin driven by Flop riscv_i/IO/SD/inst/sd_cmd_reg_o_reg and T (tri-state) pin driven by Flop riscv_i/IO/SD/inst/sd_cmd_reg_t_reg, both of which have the IOB attribute set. This cannot be honored by placement in this device architecture, which has only one register available in the IOB.`
Is IOB true necessary?

eugene-tarassov, 2022-04-20:
> Is it rignt?

It looks right to me.

> Is “IOB true” property necessary?

No. It helps to improve timing. But, if it is not supported, just remove it.

> I didn‘t find sdreset pin on KCU105，but I found other mincro-SDcard（on nexys-video）has this pin，will this affect anything

The reset is nice to have, but not required. You need to set `CONFIG.capabilies_reg {0x0001}` in your  design Tcl file to tell SD driver that reset is not supported. Like, for example, in VC707 files.

langccc, 2022-04-20:
Thank you，There are still some questions.
Do I need to set `CONFIG.capabilies_reg {0x0001}` for KCU105?
And I configured FPGA with .bit file from jtag，not mcs file from spi flash,will it matter?

closed 2022-04-27 by langccc