eugene-tarassov/vivado-risc-v #81: Boot faied hang at loop: module loaded

langccc, 2022-04-20:
I reserved some memory like this:
`/ {
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;
		
		reserved: buffer@a0000000 {
			no-map;
			reg = <0xb0000000 0x10000000>;
			status = "okay"; 
		};
  	};
};`
but failed when boot linux:
`ERROR: reserving fdt memory region failed (addr=b0000000 size=10000000 flags=4) `
And boot failed,hang at loop: module loaded ,cam you give me some advice to fix it?
This is the boot information:
`-Boot 2022.01-dirty (Apr 15 2022 - 05:07:32 -0700)                             
                                                                                
CPU:   rv64imafdc                                                               
Model: freechips,rocketchip-vivado                                              
DRAM:  1 GiB                                                                    
MMC:   mmc0@60000000: 0                                                         
Loading Environment from nowhere... OK                                          
In:    uart@60010000                                                            
Out:   uart@60010000                                                            
Err:   uart@60010000                                                            
Net:   No ethernet found.                                                       
Hit any key to stop autoboot:  0                                                
switch to partitions #0, OK                                                     
mmc0 is current device                                                          
Scanning mmc 0:1...                                                             
Found /extlinux/extlinux.conf                                                   
Retrieving file: /extlinux/extlinux.conf                                        
1:      Debain v5.16.14                                                         
Retrieving file: /extlinux/initrd-v5.16.14.img                                  
Retrieving file: /extlinux/image-v5.16.14                                       
append: ro root=UUID=68d82fa1-1bb5-435f-a5e3-862176586eec earlycon initramfs.ru8
Moving Image from 0x81000000 to 0x80200000, end=8131a190                        
## Flattened Device Tree blob at 00010080                                       
   Booting using the fdt blob at 0x010080                                       
ERROR: reserving fdt memory region failed (addr=b0000000 size=10000000 flags=4) 
   Loading Device Tree to 0000000084ffc000, end 0000000084ffff40 ... OK         
                                                                                
Starting kernel ...                                                             
                                                                                
[    0.000000] Linux version 5.16.14-dirty (eda@ubuntu) (riscv64-linux-gnu-gcc 2
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000           
[    0.000000] Machine model: freechips,rocketchip-vivado                       
[    0.000000] earlycon: axi_uart0 at MMIO 0x0000000060010000 (options '')      
[    0.000000] printk: bootconsole [axi_uart0] enabled                          
[    0.000000] Zone ranges:                                                     
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]           
[    0.000000]   Normal   empty                                                 
[    0.000000] Movable zone start for each node                                 
[    0.000000] Early memory node ranges                                         
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000afffffff]          
[    0.000000]   node   0: [mem 0x00000000b0000000-0x00000000bfffffff]          
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff] 
[    0.000000] On node 0, zone DMA32: 512 pages in unavailable ranges           
[    0.000000] SBI specification v0.3 detected                                  
[    0.000000] SBI implementation ID=0x1 Version=0x10000                        
[    0.000000] SBI TIME extension detected                                      
[    0.000000] SBI IPI extension detected                                       
[    0.000000] SBI RFENCE extension detected                                    
[    0.000000] SBI v0.2 HSM extension detected                                  
[    0.000000] riscv: ISA extensions acdfim                                     
[    0.000000] riscv: ELF capabilities acdfim                                   
[    0.000000] percpu: Embedded 16 pages/cpu s27544 r8192 d29800 u65536         
[... 48 lines cut ...]
[    0.597681] RPC: Registered named UNIX socket transport module.              
[    0.604440] RPC: Registered udp transport module.                            
[    0.609860] RPC: Registered tcp transport module.                            
[    0.615183] RPC: Registered tcp NFSv4.1 backchannel transport module.        
[    0.622750] PCI: CLS 0 bytes, default 64                                     
[    0.633645] Unpacking initramfs...                                           
[    0.645240] workingset: timestamp_bits=62 max_order=18 bucket_order=0        
[    0.758993] squashfs: version 4.0 (2009/01/31) Phillip Lougher               
[    0.781903] NFS: Registering the id_resolver key type                        
[    0.788011] Key type id_resolver registered                                  
[    0.792993] Key type id_legacy registered                                    
[    0.799011] nfs4filelayout_init: NFSv4 File Layout Driver Registering...     
[    0.806803] nfs4flexfilelayout_init: NFSv4 Flexfile Layout Driver Registerin.
[    0.818711] 9p: Installing v9fs 9p2000 file system support                   
[    0.830102] NET: Registered PF_ALG protocol family                           
[    0.837736] Block layer SCSI generic (bsg) driver version 0.4 loaded (major )
[    0.845940] io scheduler mq-deadline registered                              
[    0.851142] io scheduler kyber registered                                    
[    1.954487] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled         
[    1.980811] 60010000.uart: ttyAU0 at MMIO 0x60010000 (irq = 1, base_baud = 0t
[    1.990312] printk: console [ttyAU0] enabled                                 
[    1.990312] printk: console [ttyAU0] enabled                                 
[    1.999627] printk: bootconsole [axi_uart0] disabled                         
[    1.999627] printk: bootconsole [axi_uart0] disabled                         
[    2.138301] loop: module loaded`

Rucadi, 2022-04-20:
Maybe pmem is useful for your requirements?
Just make your ram smaller and add the pmem on the device-tree.

```
pmem@b0000000  {
		compatible = "pmem-region";
		reg = <0xb0000000 0x10000000>;
		volatile;

};
```

flyxia624, 2022-04-21:
> 

Thank you ,I will try it,and I want to know why reserved-memory failed，Do you know something about this?

langccc, 2022-04-21:
I removed reserved memory,and I got a strange error and boot failed:
`U-Boot 2022.01-dirty (Apr 15 2022 - 05:07:32 -0700)                                                     
                                                                                                        
CPU:   rv64imafdc                                                                                       
Model: freechips,rocketchip-vivado                                                                      
DRAM:  1 GiB                                                                                            
MMC:   mmc0@60000000: 0                                                                                 
Loading Environment from nowhere... OK                                                                  
In:    uart@60010000                                                                                    
Out:   uart@60010000                                                                                    
Err:   uart@60010000                                                                                    
Net:   No ethernet found.                                                                               
Hit any key to stop autoboot:  0                                                                        
switch to partitions #0, OK                                                                             
mmc0 is current device                                                                                  
Scanning mmc 0:1...                                                                                     
Found /extlinux/extlinux.conf                                                                           
Retrieving file: /extlinux/extlinux.conf                                                                
1:      Debain v5.16.14                                                                                 
Retrieving file: /extlinux/initrd-v5.16.14.img                                                          
Error reading cluster                                                                                   
** Unable to read file /extlinux/initrd-v5.16.14.img **                                                 
Skipping Debain v5.16.14 for failure retrieving initrd                                                  
SCRIPT FAILED: continuing...                                                                            
** Unable to read file / **                                                                             
Failed to load '/'                                                                                      
libfdt fdt_check_header(): FDT_ERR_BADMAGIC                                                             
Scanning disk mmc0@60000000.blk...                                                                      
Found 3 disks                                                                                           
No EFI system partition                                                                                 
BootOrder not defined                                                                                   
EFI boot manager: Cannot load any image                                                                 
No ethernet found.                                                                                      
No ethernet found.   `
How could reserved-memory node influence accessing SD card?

langccc, 2022-04-22:
I reseved memory in the middle of RAM，no error again

closed 2022-04-22 by langccc