eugene-tarassov/vivado-risc-v #82: Boot failed by ethernet

langccc, 2022-04-21:
There are three situations:
First,I configrued eth borad interface like this:
[image]
but get critial warning:
`[Vivado 12-1411] Cannot set LOC property of ports, riscv_i/IO/EthernetKCU105/gig_ethernet_pcs_pma_0/inst/pcs_pma_block_i/transceiver_inst/riscv_gig_ethernet_pcs_pma_0_0_gt_i/inst/gen_gtwizard_gthe3_top.riscv_gig_ethernet_pcs_pma_0_0_gt_gtwizard_gthe3_inst/gen_gtwizard_gthe3.gen_channel_container[2].gen_enabled_channel.gthe3_channel_wrapper_inst/channel_inst/gthe3_channel_gen.gen_gthe3_channel_inst[0].GTHE3_CHANNEL_PRIM_INST Illegal to place instance riscv_i/IO/EthernetKCU105/gig_ethernet_pcs_pma_0/inst/pcs_pma_block_i/transceiver_inst/riscv_gig_ethernet_pcs_pma_0_0_gt_i/inst/gen_gtwizard_gthe3_top.riscv_gig_ethernet_pcs_pma_0_0_gt_gtwizard_gthe3_inst/gen_gtwizard_gthe3.gen_channel_container[2].gen_enabled_channel.gthe3_channel_wrapper_inst/channel_inst/gthe3_channel_gen.gen_gthe3_channel_inst[0].GTHE3_CHANNEL_PRIM_INST on site P24. The location site type (HRIO) and bel type (HRIO_OUTINV) do not match the cell type (GTHE3_CHANNEL). Instance riscv_i/IO/EthernetKCU105/gig_ethernet_pcs_pma_0/inst/pcs_pma_block_i/transceiver_inst/riscv_gig_ethernet_pcs_pma_0_0_gt_i/inst/gen_gtwizard_gthe3_top.riscv_gig_ethernet_pcs_pma_0_0_gt_gtwizard_gthe3_inst/gen_gtwizard_gthe3.gen_channel_container[2].gen_enabled_channel.gthe3_channel_wrapper_inst/channel_inst/gthe3_channel_gen.gen_gthe3_channel_inst[0].GTHE3_CHANNEL_PRIM_INST belongs to a shape with reference instance riscv_i/IO/EthernetKCU105/gig_ethernet_pcs_pma_0/inst/pcs_pma_block_i/transceiver_inst/riscv_gig_ethernet_pcs_pma_0_0_gt_i/inst/gen_gtwizard_gthe3_top.riscv_gig_ethernet_pcs_pma_0_0_gt_gtwizard_gthe3_inst/gen_gtwizard_gthe3.gen_channel_container[2].gen_enabled_channel.gthe3_channel_wrapper_inst/channel_inst/gthe3_channel_gen.gen_gthe3_channel_inst[0].GTHE3_CHANNEL_PRIM_INST. Shape elements have relative placement respect to each other. The invalid location might results from a constraint on any of the instance in the shape. ["/home/lyuan/Desktop/kcu105-riscv/kcu105-riscv.srcs/constrs_1/imports/vivado-risc-v/board/kcu105/ethernet.xdc":4]
`
Then boot stoped at loop:module loaded
[image]
Second,I configure configrued eth borad interface like this:
[image]
Then there is no uart output

Third,based on the second，I disconnect eth/mmcm_locked)out and RocketChip/io_ok ,then connect RocketChip/io_ok and clk_wiz_0/locked
[image]
Then uart can output,but it said unable to select a mode and stopped
[image]
Can you give me some advice ,thanks

langccc, 2022-04-22:
I found AXI smartconnect makes this，I changed it to interconnecte，it works OK

closed 2022-04-22 by langccc

Yuxin-Yu, 2023-03-15:
Hello, I have the same problem. Can you tell me how to modify it, or show me your modified block design @langccc

Yuxin-Yu, 2023-03-16:
Hello, I deleted AXISmart IP and reconnected it according to your description, but the effect is the same. I wonder if it is a problem with the vivado version (I use vivado 2020.2). Can it be solved by upgrading the vivado version? @langccc .In addition, have bloggers encountered this problem @eugene-tarassov .And my error log is here:
`
CPU:   rv64imafdc
Model: freechips,rocketchip-vivado
DRAM:  1 GiB
MMC:   mmc0@60000000: 0
Loading Environment from nowhere... OK
In:    uart@60010000
Out:   uart@60010000
Err:   uart@60010000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
No ethernet found.
No ethernet found.
=> 
`

langccc, 2023-03-16:
> Hello, I deleted AXISmart IP and reconnected it according to your description, but the effect is the same. I wonder if it is a problem with the vivado version (I use vivado 2020.2). Can it be solved [...]
No，I repeated the same problem in two versions. And I see you problem is different to me. I think it may be UART problem, you can try to type some charactors on the console, if no charactor appear, may be UART TX from PC doesnot work.

Yuxin-Yu, 2023-03-16:
I can make sure my UART is OK. By the way, I just tried to replace the SD card. The program won't be stuck here, so I guess it's my SD card configuration problem