eugene-tarassov/vivado-risc-v #84: ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist

langccc, 2022-04-22:
Boot stuck  in accessing rootfs，it seems something wrong with rootfs
`U-Boot 2022.01-dirty (Apr 15 2022 - 05:07:32 -0700)                             
                                                                                
CPU:   rv64imafdc                                                               
Model: freechips,rocketchip-vivado                                              
DRAM:  1 GiB                                                                    
MMC:   mmc0@60000000: 0                                                         
Loading Environment from nowhere... OK                                          
In:    uart@60010000                                                            
Out:   uart@60010000                                                            
Err:   uart@60010000                                                            
Net:   No ethernet found.                                                       
Hit any key to stop autoboot:  0                                                
switch to partitions #0, OK                                                     
mmc0 is current device                                                          
Scanning mmc 0:1...                                                             
Found /extlinux/extlinux.conf                                                   
Retrieving file: /extlinux/extlinux.conf                                        
1:      Debain v5.16.14                                                         
Retrieving file: /extlinux/initrd-v5.16.14.img                                  
Retrieving file: /extlinux/image-v5.16.14                                       
append: ro root=UUID=68d82fa1-1bb5-435f-a5e3-862176586eec earlycon initramfs.ru8
Moving Image from 0x81000000 to 0x80200000, end=8131a190                        
## Flattened Device Tree blob at 00010080                                       
   Booting using the fdt blob at 0x010080                                       
   Loading Device Tree to 0000000084ffc000, end 0000000084ffff44 ... OK         
                                                                                
Starting kernel ...                                                             
                                                                                
[    0.000000] Linux version 5.16.14-dirty (eda@ubuntu) (riscv64-linux-gnu-gcc 2
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000           
[    0.000000] Machine model: freechips,rocketchip-vivado                       
[    0.000000] earlycon: axi_uart0 at MMIO 0x0000000060010000 (options '')      
[    0.000000] printk: bootconsole [axi_uart0] enabled                          
[    0.000000] Zone ranges:                                                     
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]           
[    0.000000]   Normal   empty                                                 
[    0.000000] Movable zone start for each node                                 
[    0.000000] Early memory node ranges                                         
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000009fffffff]          
[    0.000000]   node   0: [mem 0x00000000a0000000-0x00000000afffffff]          
[    0.000000]   node   0: [mem 0x00000000b0000000-0x00000000bfffffff]          
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff] 
[    0.000000] On node 0, zone DMA32: 512 pages in unavailable ranges           
[    0.000000] SBI specification v0.3 detected                                  
[    0.000000] SBI implementation ID=0x1 Version=0x10000                        
[    0.000000] SBI TIME extension detected                                      
[    0.000000] SBI IPI extension detected                                       
[    0.000000] SBI RFENCE extension detected                                    
[    0.000000] SBI v0.2 HSM extension detected                                  
[    0.000000] riscv: ISA extensions acdfim                                     
[    0.000000] riscv: ELF capabilities acdfim                                   
[    0.000000] percpu: Embedded 16 pages/cpu s27544 r8192 d29800 u65536         
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055    
[    0.000000] Kernel command line: ro root=UUID=68d82fa1-1bb5-435f-a5e3-8621768
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, l)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off          
[    0.000000] Memory: 729772K/1046528K available (6183K kernel code, 4837K rwd)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1       
[    0.000000] rcu: Hierarchical RCU implementation.                            
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=32 to nr_cpu_ids=2.   
[    0.000000]  Tracing variant of Tasks RCU enabled.                           
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 ji.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2     
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0                   
[    0.000000] riscv-intc: 64 local interrupts mapped                           
[    0.000000] plic: interrupt-controller@c000000: mapped 8 interrupts with 2 h.
[    0.000000] random: get_random_bytes called from start_kernel+0x44e/0x604 wi0
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[... 91 lines cut ...]
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
Begin: Running /scripts/local-block ... done.                                   
done.                                                                           
Gave up waiting for root file system device.  Common problems:                  
 - Boot args (cat /proc/cmdline)                                                
   - Check rootdelay= (did the system wait long enough?)                        
 - Missing modules (cat /proc/modules; ls /dev)                                 
ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist.  Dropping to !
(initramfs) 
`

langccc, 2022-04-22:
BTW,I used virtual machine Ubuntu 18.04 and SanDisk 32GB Ultra microSDHC A1 HC1 Class 10.Will it matter?

closed 2022-04-26 by langccc

L0stMyN4me, 2022-12-02:
请问你是怎么解决的呀  我也遇到了这样的问题 向您请教一下！

flyxia624, 2022-12-07:
> 

SD卡控制器的参数是不是有问题，大卡和小卡的参数不一样，SD卡是否带复位的参数也不一样，具体你可以看board文件夹下不同FPGA板的SD卡参数配置

langccc, 2022-12-07:
> 请问你是怎么解决的呀 我也遇到了这样的问题 向您请教一下！

SD卡控制器的参数是不是有问题，大卡和小卡的参数不一样，SD卡是否带复位的参数也不一样，具体你可以看board文件夹下不同FPGA板的SD卡参数配置