eugene-tarassov/vivado-risc-v #87: Using Vivado 2022.1

svpolonsky, 2022-05-23:
Hi Eugene, 
I am trying to use Vivado 2022.1, and get the error: 
**Couldn't read file "../../board/nexys-a7-100t/riscv-2022.1.tcl": no such file or directory**
It seems that you didn't yet include support for this version.
Questions:
(1) Any simple workaround, like **cp riscv-2021.2.tcl riscv-2022.1.tcl**?
(2) Do you plan to add the support for 2022.1
(3) or should I just download and use 2021.2?
Thanks in advance,
Stas

eugene-tarassov, 2022-05-23:
Vivado 2022.1 is not supported yet.

> (1) Any simple workaround, like cp riscv-2021.2.tcl riscv-2022.1.tcl?

You can start by coping the file, but it needs to be edited to match new Vivado release - the file contains version numbers of Vivado and various IPs. In general, the file is not guaranteed to be portable between versions.,

> (2) Do you plan to add the support for 2022.1

Yes. It will take 1-2 weeks to run all tests.

> (3) or should I just download and use 2021.2?

I recommend 2021.1 - it appears to be the most stable of supported releases.

svpolonsky, 2022-05-23:
Thanks, Eugene! I will work with 2021.1

closed 2022-06-26 by eugene-tarassov