eugene-tarassov/vivado-risc-v #88: make flash: connection refused

svpolonsky, 2022-05-25:
Hi Eugene,

I run "**make CONFIG=rocket64b1 BOARD=nexys-a7-100t flash**" and get the following error: **Connection refused**

When I check the make output, I see:

**env HW_SERVER_URL=tcp:localhost:3121 \
 xsdb -quiet board/jtag-freq.tcl
rlwrap: warning: your $TERM is 'xterm-256color' but rlwrap couldn't find it in the terminfo database. Expect some problems.
Connection refused                                                                                                             
    while executing
"error $message"
    (procedure "connect" line 143)
    invoked from within
"connect -url $::env(HW_SERVER_URL)"
    (file "board/jtag-freq.tcl" line 1)
make: *** [Makefile:310: flash] Error 1**

the setting for HW_SERVER_URL implies you expect Ethernet connection, while I connect my  nexys-a7-100t through USB. Could this be a problem?

Sincerely,
Stas

eugene-tarassov, 2022-05-25:
You need to start hw_server, which is Xilinx JTAG debugger.
It listens for TCP/IP connections and can talk to JTAG through USB.

svpolonsky, 2022-05-27:
I got it working! Thanks, Eugene!
On a separate note, is there a simple way to add Vitis to this project? The reason of my question is I'd like to write some code to run on your RV h/w and Vitis seems like a great option for this...

eugene-tarassov, 2022-05-27:
> I got it working!

Super!

> On a separate note, is there a simple way to add Vitis to this project?

Yes, you can use Vitis or Eclipse+CDT+TCF to edit, build, run and debug. Both can talk to RV h/w using hw_server. I prefer to use Eclipse, because Vitis is same thing plus a lot of additional stuff which I never need.

However, there is no single button "create RV application" - you will need some knowledge of [CDT](https://www.eclipse.org/cdt/) and [TCF](https://www.eclipse.org/tcf/) to setup a RV software project.