eugene-tarassov/vivado-risc-v #89: Testbench for ethernet

Rossamund-care, 2022-05-27:


Rossamund-care, 2022-06-24:
Thank you. I am having problem in setting ethernet registers.

eugene-tarassov, 2022-06-24:
What kind of problem?

Rossamund-care, 2022-06-25:
Hi,
I am trying to write testbench for ethernet.v.I am not able to configure the registers correctly, thus the protocol is not working completely.I am using an  AXI master BFM and is attached to the slave port of ethernet.Through the master port I am trying to read from RAM and transmit it.I gave random values to registers.I am only able to tranmsit only one packet and the reception is not complete.

eugene-tarassov, 2022-06-25:
It sounds like DMA registers are not setup correctly.

There are 2 ring buffers, one for RX and one for TX, each can hold up to 15 packet descriptors.

To transmit a packet, you fill first free TX descriptor - pointed by tx_inp, then move tx_inp = (tx_inp + 1) % 16.
You can add up to 15 packets without waiting until a packet is transmitted.
The adapter moves tx_out after transmitting a packet.
You can poll tx_out and add more packets when the TX ring buffer has free space.

To receive a packet, you fill first free RX descriptor - pointed by rx_inp, then move rx_inp = (rx_inp + 1) % 16.
You can add up to 15 (empty) packets into the ring buffer.
The adapter moves rx_out after receiving a packet.
You can poll rx_out and add more packets when the RX ring buffer has free space.

See the [Ethernet driver](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/patches/fpga-axi-eth.c) for details.