eugene-tarassov/vivado-risc-v #90: u-boot cannot reads the linux image from SD card

Shuiliusheng, 2022-06-01:
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
** Unable to read file / **
Failed to load '/'
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
Scanning disk mmc0@60000000.blk...

I change the delay of L2 Cache receiving the data from DDR, and then I cannot boot the linux. Could you give me some advice about this. Will the udelay(10000) in sdc.c affect this?

eugene-tarassov, 2022-06-01:
Yes, memory bus bandwidth needs to be higher then SD card transfer speed, otherwise you get buffer overflow or underflow.
You can try to reduce SD clock frequency, look for max-frequency in the board device tree.

Shuiliusheng, 2022-06-01:
Thank you so much!

Shuiliusheng, 2022-06-01:
Hello, if I want to reduce the SD clock frequency, do I need to change the following code in sdc directory and bootrom.dts in board directory?
    (* X_INTERFACE_INFO = "xilinx.com:signal:clock:1.0 sdio_clk CLK" *)
    (* X_INTERFACE_PARAMETER = "FREQ_HZ 50000000" *)
    output reg sdio_clk,

    (* X_INTERFACE_INFO = "xilinx.com:signal:clock:1.0 clock CLK" *)
    (* X_INTERFACE_PARAMETER = "ASSOCIATED_BUSIF M_AXI:S_AXI_LITE, FREQ_HZ 100000000" *)
    input wire clock,

   mmc: mmc0@60000000 {
            compatible = "riscv,axi-sd-card-1.0";
            clock = <100000000>;
            fifo-depth = <256>;
            max-frequency = <50000000>;
        };

eugene-tarassov, 2022-06-01:
Only max-frequency in bootrom.dts

Yuxin-Yu, 2023-03-17:
hello,I think I have the same problem.Because my DDR3 cann't boot with 800MHz,so I reduce the period to 2500ps(400MHz),then I cann't boot linux with error :

> Boot HART ID              : 0
> Boot HART Domain          : root
> Boot HART Priv Version    : v1.11
> Boot HART Base ISA        : rv64imafdcx
> Boot HART ISA Extensions  : none
> Boot HART PMP Count       : 8
> Boot HART PMP Granularity : 4
> Boot HART PMP Address Bits: 32
> Boot HART MHPM Count      : 0
> Boot HART MIDELEG         : 0x0000000000000222
> Boot HART MEDELEG         : 0x000000000000b109
> 
> 
> U-Boot 2022.01-dirty (Mar 17 2023 - 08:25:13 +0800)
> 
> CPU:   rv64imafdc
> Model: freechips,rocketchip-vivado
> DRAM:  1 GiB
> MMC:   mmc0@60000000: 0
> Loading Environment from nowhere... OK
> In:    uart@60010000
> Out:   uart@60010000
> Err:   uart@60010000
> Net:   No ethernet found.
> Hit any key to stop autoboot:  0 
> switch to partitions #0, OK
> mmc0 is current device
> No ethernet found.
> No ethernet found
> => 
I guess this is due to the bandwidth incompatibility between DDR and SD card mentioned above. If my DDR clock is 400MH, how can I configure the corresponding SD card @eugene-tarassov

Yuxin-Yu, 2023-03-17:
hello,I think I have the same problem.Because my DDR3 cann't boot with 800MHz,so I reduce the period to 2500ps(400MHz),then I cann't boot linux with error :

> Boot HART ID              : 0
> Boot HART Domain          : root
> Boot HART Priv Version    : v1.11
> Boot HART Base ISA        : rv64imafdcx
> Boot HART ISA Extensions  : none
> Boot HART PMP Count       : [...]
I guess this is due to the bandwidth incompatibility between DDR and SD card mentioned above. If my DDR clock is 400MH, how can I configure the corresponding SD card @eugene-tarassov

eugene-tarassov, 2023-03-17:
> I guess this is due to the bandwidth incompatibility between DDR and SD card mentioned above.

This issue is already fixed. SD card should work at any DDR or CPU frequency.

There no any error messages in your log. It looks like u-boot does not even try to boot from SD because it is empty or not properly formatted.

A valid log should look like this:
```
U-Boot 2022.01-dirty (Jan 22 2023 - 19:34:47 -0800)

CPU:   rv64imafdc
Model: freechips,rocketchip-vivado
DRAM:  1 GiB
MMC:   mmc0@60000000: 0
Loading Environment from nowhere... OK
In:    uart@60010000
Out:   uart@60010000
Err:   uart@60010000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
1:      Debain v5.19.17
```

Somehow `Scanning mmc 0:1...` is missing in your log.

Yuxin-Yu, 2023-03-17:
Maybe I miss some important message,some time I get another log,which like:

> U-Boot 2022.01-dirty (Mar 17 2023 - 08:25:13 +0800)
> 
> CPU:   rv64imafdc
> Model: freechips,rocketchip-vivado
> DRAM:  1 GiB
> MMC:   mmc0@60000000: 0
> Loading Environment from nowhere... OK
> In:    uart@60010000
> Out:   uart@60010000
> Err:   uart@60010000
> Net:   No ethernet found.
> Hit any key to stop autoboot:  0 
> switch to partitions #0, OK
> mmc0 is current device
> ** No partition table - mmc 0 **
> Couldn't find partition mmc 0:1
> No ethernet found.
> No ethernet found.
> => 

I am using the latest version of the code, and I still believe it is a problem with the SD card and DDR, because this is the biggest change I have made. Based on this new error report, can you identify the probable cause of the error

Yuxin-Yu, 2023-03-17:
I boot many times,some time there are "Scanning mmc 0:1..." message,but stuck because other error.Like:

> Hit any key to stop autoboot:  0 
> switch to partitions #0, OK
> mmc0 is current device
> Scanning mmc 0:1...
> Found /extlinux/extlinux.conf
> Retrieving file: /extlinux/extlinux.conf
> Error reading cluster
> ** Unable to read file /extlinux/extlinux.conf **
> Error reading config file
> SCRIPT FAILED: continuing...
> ** Unable to read file / **
> Failed to load '/'
> libfdt fdt_check_header(): FDT_ERR_BADMAGIC
> ␛7␛[r␛[999;999H␛[6n␛8Scanning disk mmc0@60000000.blk...
> Found 3 disks
> No EFI system partition
> BootOrder not defined
> EFI boot manager: Cannot load any image
> No ethernet found.
> No ethernet found.

Is this because my SD card is incorrectly configured?

Yuxin-Yu, 2023-03-17:
In addition, I have tried to replace the SD card, but another error occurred, as follows:

> [  138.282328] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> [  138.530596] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
> [  138.553682] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
> [  138.570234] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> Begin: Running /scripts/local-block ... done.
> [  140.106590] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
> [  140.129834] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
> [  140.146402] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> [  140.417758] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> done.
> [  140.817590] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> [  141.089666] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> Gave up waiting for root file system device.  Common problems:
>  - Boot args (cat /proc/cmdline)
>    - Check rootdelay= (did the system wait long enough?)
>  - Missing modules (cat /proc/modules; ls /dev)
> ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist.  Dropping to a shell!
> (initramfs) 

@eugene-tarassov

Yuxin-Yu, 2023-03-17:
> In addition, I have tried to replace the SD card, but another error occurred, as follows:
>
>
> @eugene-tarassov

In this message,there are a point which I think is important:

> [    9.521172] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0

> [    9.538658] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> [    9.556198] I/O error, dev mmcblk0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
> [    9.578902] Buffer I/O error on dev mmcblk0, logical block 0, async page read
> [    9.587662]  mmcblk0: unable to read partition table
> Loading, please wait...
> Starting version 250.3-2

It shows "mmcblk0:unable to read partition table",how can I solves this,Thanks
@eugene-tarassov

eugene-tarassov, 2023-03-18:
In bootrom.dts, try to change `bus-width = <4>` to `bus-width = <1>` and do a clean build. It looks very much like your SD card connection is broken and u-boot fails when it enables 4 lane data bus.

Yuxin-Yu, 2023-03-20:
yes,you are right.After I changed bus-width = <4> to bus-width = <1>，I can boot linux.In addtion ,because I am using a pmod sd card expansion module, referring to the arty-a7-100t routine, I have changed the SD maximum clock frequency and can also start Linux.As follows:

> mmc: mmc0@60000000 {
>             compatible = "riscv,axi-sd-card-1.0";
>             clock = <100000000>;
>             reg = <0x60000000 0x10000>;
>             bus-width = <4>;
>             fifo-depth = <256>;
>             interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
>             interrupts = <2>;
>             max-frequency = <12500000>;
>             cap-sd-highspeed;
>             cap-mmc-highspeed;
>             no-sdio;
>         };

However, both will enter the emergency mode during the Linux startup process, as follows:

> [ TIME ] Timed out waiting for device /dev/disk/by-label/BOOT.
> [DEPEND] Dependency failed for /boot.
> [DEPEND] Dependency failed for Local File Systems.
> [  OK  ] Stopped Dispatch Password …ts to Console Directory Watch.
> [  OK  ] Stopped Forward Password R…uests to Wall Directory Watch.
> [  OK  ] Reached target Timer Units.
> [  OK  ] Reached target System Time Synchronized.
> [  OK  ] Reached target Login Prompts.
> [  OK  ] Started Emergency Shell.
> [  OK  ] Reached target Emergency Mode.
>          Starting Set Up Additional Binary Formats...
>          Starting Create Volatile Files and Directories...
>          Mounting Arbitrary Executable File Formats File System...
> You are in emergency mode. After logging in, type "journalctl -xb" to view
> system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
> to boot into default mode.
> Give root password for maintenance
> (or press Control-D to continue):
> root@debian:~# ping www.baidu.com
> ping: www.baidu.com: Temporary failure in name resolution

Of course, using the prompt systemctl default can enter the default mode, but the network port is still not working. I wonder if my network port driver is not configured properly. My PHY chip is 88E1111

> root@debian:~# systemctl default
> [  OK  ] Stopped Emergency Shell.
> [  OK  ] Mounted /boot.
> [  OK  ] Finished Load Kernel Module configfs.
> [  OK  ] Finished Load Kernel Module efi_pstore.
> [  OK  ] Finished Load Kernel Module fuse.
> [  OK  ] Started Dispatch Password …ts to Console Directory Watch.
> [  OK  ] Started Forward Password R…uests to Wall Directory Watch.
> [  OK  ] Reached target Local File Systems.
> [  OK  ] Reached target System Initialization.
> [  OK  ] Started Daily dpkg database backup timer.
> [  OK  ] Started Periodic ext4 Onli…ata Check for All Filesystems.
> [  OK  ] Started Discard unused blocks once a week.
> [  OK  ] Started Daily rotation of log files.
> [  OK  ] Started Daily Cleanup of Temporary Directories.
> [  OK  ] Reached target Basic System.
> [  OK  ] Started Regular background program processing daemon.
>          Starting Remove Stale Onli…t4 Metadata Check Snapshots...
> [  OK  ] Found device /dev/ttyAU0.
> [FAILED] Failed to start Helper to …chronize boot up for ifupdown.
> See 'systemctl status ifupdown-pre.service' for details.
> [DEPEND] Dependency failed for Raise network interfaces.
> [  OK  ] Reached target Network.
> [  OK  ] Reached target Network is Online.
>          Starting Automounts filesystems on demand...
>          Starting chrony, an NTP client/server...
>          Starting Notify NFS peers of a restart...
>          Starting OpenBSD Secure Shell server...
>          Starting Permit User Sessions...
> [  OK  ] Started Notify NFS peers of a restart.
> [  OK  ] Finished Permit User Sessions.
> [  OK  ] Started Automounts filesystems on demand.
> [  OK  ] Started Serial Getty on ttyAU0.
> 
> Debian GNU/Linux bookworm/sid debian ttyAU0
> 
> debian login: root
> Password:
> Linux debian 5.19.17-dirty #1 SMP Mon Mar 20 11:11:37 CST 2023 riscv64
> 
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> root@debian:~# ping www.baidu.com
> ping: www.baidu.com: Temporary failure in name resolution

How can I solve the network interface problem here.Thank you

garthlei, 2023-06-04:
> This issue is already fixed. SD card should work at any DDR or CPU frequency.
>
> There no any error messages in your log. It looks like u-boot does not even try to boot from SD because it is empty or [...]

Hi, would you mind telling me in which commit the issue is fixed?

eugene-tarassov, 2023-06-04:
The the bandwidth incompatibility problem is fixed in [this commit](https://github.com/eugene-tarassov/vivado-risc-v/commit/d1b97b9135b01f8dceb8a9139b09a87c85cab602)