eugene-tarassov/vivado-risc-v #93: Arty S7-50 support

RossComputerGuy, 2022-07-01:
I've got an Arty S7-50 and I don't see it in `boards`. I've got an sdcard pmod on `JA`, ESP32 on `JC`, and the OLED RGB screen on `JD`. I'm not sure how to add support but if there's a guide or a similar board that is already supported then I might be able to get it work. If I can create support using Vivado's IP builder then that would be much easier.

eugene-tarassov, 2022-07-01:
Yes, you can use Vivado's IP builder to modify a project created for a similar board that is already supported.
For example, you can use Nexys A7 project as starting point.

To create and open Nexys A7 project: checkout the repository, install required packages and update submodules (see README).
Then run:
```
source /opt/Xilinx/Vivado/2022.1/settings64.sh
make CONFIG=rocket64b1 BOARD=nexys-a7-100t vivado-gui
```
Save the project as, for example, workspace/rocket64b1/vivado-arty-s7-50-riscv, change the board name, open and edit IP block design, etc.

RossComputerGuy, 2022-07-01:
Alright, I'm starting to get things working. Hopefully I can work on this on nixOS and Vivado running in a Flatpak. Only problem I've found so far is no nix shell file.

RossComputerGuy, 2022-07-01:
Unfortunately, I cannot get it to work as it says `dtc` cannot be found and the Flatpak for Vivado doesn't have Java. It seems the way nixOS is set up makes Vivado not work outside of the Flatpak.

eugene-tarassov, 2022-07-01:
I'm not familiar with Flatpak, but I think it will not work. Rocket Chip RISC-V is a huge project, it needs powerful PC and tons of software to build, dtc and Java are just tip of the iceberg. To save time, I recommend to follow instructions, starting with fresh installation of Ubuntu.

RossComputerGuy, 2022-07-01:
I have no hardware that I can install Ubuntu on and I have no free drives for dual booting.

closed 2023-03-31 by eugene-tarassov