eugene-tarassov/vivado-risc-v #94: RISC-V compare with MicroBlaze and ARM

briansune, 2022-07-03:
Hello,

First really appreciate the magnificent RISC-V project.
I had successfully run debian on the single RISC-V big core on Artix 7 but I am facing some concern here.
I would like to learn a bit more is this RISC-V able to run Ubuntu?
What is the major performance differences compared with MB and ZYNQ ARM? (Single core based).

Thank you again.

eugene-tarassov, 2022-07-03:
> I would like to learn a bit more is this RISC-V able to run Ubuntu?

Yes. Ubuntu does support RISC-V.
However, there is no ready to use Ubuntu build for this project. It would need some work to prepare one.

> What is the major performance differences compared with MB and ZYNQ ARM? (Single core based).

RISC-V, MB and ZYNQ ARM performance per MHz is about same.
Of course, hardened processors have much higher clock frequency than a soft core in FPGA.
So, ZYNQ ARM runs about 15 times faster than a typical soft core.
Main purpose of running a CPU in FPGA is prototyping.
They are not meant to compete with hardened CPUs.

I don't have data for ZYNQ ARM (Cortex A9). Here is what I have measured for other CPUs:

```
RISC-V Rocket 100 MHz   : CoreMark 1.0 :   202.169958 / GCC10.2.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   2.02 CM/MHz
RISC-V 2-way BOOM 50 MHz: CoreMark 1.0 :   183.228486 / GCC10.2.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   3.66 CM/MHz
RISC-V 3-way BOOM 40 MHz: CoreMark 1.0 :   190.609314 / GCC10.2.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   4.77 CM/MHz
Cortex A53 1200 MHz     : CoreMark 1.0 :  3994.141925 / GCC8.3.0  -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   3.32 CM/MHz
Xeon(R) E3-1281 4100 MHz: CoreMark 1.0 : 29197.080292 / GCC9.3.0  -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   7.12 CM/MHz
Xeon(R) E-2278G 5000 MHz: CoreMark 1.0 : 36719.706242 / GCC9.3.0  -O2 -DPERFORMANCE_RUN=1  -lrt / Heap   7.34 CM/MHz
```

briansune, 2022-07-03:
First OMG those data are amazing! Great~

Meantime I am on the same page of "They are not meant to compete with hardened CPUs."
However, idea is that if there is a possible chance we can introduce python Jupyter notebook to the OS then reconfigure the remain LEs. It will enable a lot of room for on air update, remote update, easy modification of design etc.

briansune, 2022-07-04:
Hello,

I manage to image the Ubuntu to the SD Card and launch the OS.
But I am facing a serious problem that had no clue to fix it:
Had tried all kinds of general setup /etc/network/interface etc yet not working.

ip a
`
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:0a:35:00:00:02 brd ff:ff:ff:ff:ff:ff
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
`

`RISC-V 64, Boot ROM V3.4

OpenSBI v1.0
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : Vivado RISC-V
Platform Features         : medeleg
Platform HART Count       : 32
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1000000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Firmware Base             : 0x80000000
Firmware Size             : 468 KB
Runtime SBI Version       : 0.3

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*,8*,9*,10*,11*,12*,13*,14*,15*,16*,17*,18*,19*,20*,21*,22*,23*,24*,25*,26*,27*,28*,29*,30*,31*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000080000000-0x000000008007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcsux
Boot HART Features        : scounteren,mcounteren,mcountinhibit
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 30
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2022.01-dirty (Jul 03 2022 - 03:16:09 +0800)

CPU:   rv64imafdc
[... 320 lines cut ...]
[  OK  ] Started Network Manager.
[  OK  ] Reached target Network.
         Starting OpenBSD Secure Shell server...
         Starting Permit User Sessions...
[  OK  ] Finished Permit User Sessions.
         Starting GNOME Display Manager...
[  OK  ] Started Serial Getty on ttyAU0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Finished resolvconf-pull-resolved.service.

Ubuntu 20.04.2 LTS localhost.localdomain ttyAU0

localhost login: ubuntu
Password:
Welcome to Ubuntu 20.04.2 LTS (GNU/Linux 5.18.6-dirty riscv64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.
`

briansune, 2022-07-04:
Hello,

Success!
Network also working and apt is working as well.

`
RISC-V 64, Boot ROM V3.4

OpenSBI v1.0
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : Vivado RISC-V
Platform Features         : medeleg
Platform HART Count       : 32
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1000000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Firmware Base             : 0x80000000
Firmware Size             : 468 KB
Runtime SBI Version       : 0.3

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*,8*,9*,10*,11*,12*,13*,14*,15*,16*,17*,18*,19*,20*,21*,22*,23*,24*,25*,26*,27*,28*,29*,30*,31*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000080000000-0x000000008007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcsux
Boot HART Features        : scounteren,mcounteren,mcountinhibit
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 30
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2022.01-dirty (Jul 03 2022 - 03:16:09 +0800)

CPU:   rv64imafdc
Model: freechips,rocketchip-vivado
DRAM:  1 GiB
MMC:   mmc0@60000000: 0
Loading Environment from nowhere... OK
In:    uart@60010000
Out:   uart@60010000
Err:   uart@60010000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
[... 375 lines cut ...]
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 816  bytes 65932 (65.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 816  bytes 65932 (65.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ubuntu@localhost:~$ ping 192.168.1.37
PING 192.168.1.37 (192.168.1.37) 56(84) bytes of data.
64 bytes from 192.168.1.37: icmp_seq=1 ttl=128 time=2.26 ms
64 bytes from 192.168.1.37: icmp_seq=2 ttl=128 time=0.746 ms
^C
--- 192.168.1.37 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1003ms
rtt min/avg/max/mdev = 0.746/1.500/2.255/0.754 ms
ubuntu@localhost:~$ ping google.com
PING google.com (142.250.204.46) 56(84) bytes of data.
64 bytes from hkg07s38-in-f14.1e100.net (142.250.204.46): icmp_seq=1 ttl=57 time=3.39 ms
64 bytes from hkg07s38-in-f14.1e100.net (142.250.204.46): icmp_seq=2 ttl=57 time=3.40 ms
^C
--- google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 3.387/3.392/3.398/0.005 ms
ubuntu@localhost:~$ clear
`

eugene-tarassov, 2022-07-04:
Nice!