eugene-tarassov/vivado-risc-v #97: JTAG TDO Signal Fails Timing Requirements

mwrnd, 2022-08-02:
A JTAG register to BSCAN2 path fails to meet timing requirements and is the only timing issue. `xsdb` cannot load any files larger than the bare-metal boot.elf but I am able to send and retrieve 1GB of data at 0x80000000 over XDMA.

```
...
aborting, 1 pending requests...
Failed to download vivado-risc-v/workspace/boot.elf
Memory write error at 0x80222C00. FPGA reprogrammed, wait for debugger resync
xsdb% Info: Hart #0 (target 3) Running (Debug Transport Module: data corruption (ID))
...
Failed to download vivado-risc-v/workspace/boot.elf
Memory write error at 0x80000100. Debug Transport Module timeout
xsdb% Info: Hart #0 (target 3) Running (Debug Transport Module: data corruption (ID))
```

[image]

Vivado Device View shows the path as being very short.

[image]

I am targeting an XCKU15P-based board and used the `vcu1525` board as a starting point. The RISC-V core is `rocket64b4l2w` running at 62.5MHz.

Any suggestions will be appreciated.

eugene-tarassov, 2022-08-02:
XCKU15P, XQKU15P and XCKU19P max JTAG frequency is 50MHz, while all other devices 66MHz.
The timing requirement is set to 15ns (66MHz) in [board/timing-constraints.tcl](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/timing-constraints.tcl#L109), this needs to be changed to 20ns to support XCKU15P.

Also, you can try to set lower JTAG frequency in xsdb to see if it helps:
```
jtag targets 1
jtag freq 10000000
dow vivado-risc-v/workspace/boot.elf
```

However, I'm not sure why xsdb access fails. The default JTAG frequency is only 15MHz, which should work fine even if the timing requirement fails.

mwrnd, 2022-08-02:
Thank you for the help. I replaced:
```
    create_clock -name jtag_clock -period 15.000 $tck_pin
```
with:
```
    set part_family [get_property FAMILY [get_parts $xilinx_part]]
    if { { $part_family eq "kintexuplus" } || { $part_family eq "qkintexuplus" } } {
      create_clock -name jtag_clock -period 20.000 $tck_pin
    } else {
      create_clock -name jtag_clock -period 15.000 $tck_pin
    }
```
and now the design implements with no timing issues.

---

`jtag freq` reports 6MHz so it should not be an issue. However, something seems to be regularly interrupting `xsdb` or the Debugger Hub.

After running `xsdb`'s `dow` command to load `boot.elf`, I read the memory back over XDMA and it is correctly but partially written.

[image]

Here is a hex dump of `boot.elf` which matches the retrieved memory:
```
xxd /home/user/vivado-risc-v/workspace/boot.elf | less

00000120: 3304 0500 b384 0500 3309 0600 ef00 c054  3.......3......T
00000130: 3308 0500 3305 0400 b385 0400 3306 0900  3...3.......3...
00000140: fd58 6304 1801 631d 050b 1728 0100 1308  .Xc...c....(....
00000150: e8fd 8548 2f28 1801 6314 080a 9722 0100  ...H/(..c...."..
00000160: 9382 c2fd 1703 0000 1303 c3fb 23b0 6200  ............#.b.
00000170: 9722 0100 9382 02fd 83b2 0200 b303 5340  ."............S@
00000180: 172e 0100 130e 0efa 2330 7e00 9732 0100  ........#0~..2..
00000190: 9382 c2eb 1743 0100 1303 43e4 6383 6206  .....C....C.c.b.
```

Each run of `dow` correctly writes a random amount of `boot.elf`:
```
 49%    0MB   0.0MB/s  00:07 ETA
Failed to download /home/user/vivado-risc-v/workspace/boot.elf
Memory write error at 0x8000B600. FPGA reprogrammed, wait for debugger resync
xsdb% Info: Hart #0 (target 3) Stopped at 0x80008ee0 (Suspended)
...
 83%    0MB   0.0MB/s  00:02 ETA
Failed to download /home/user/vivado-risc-v/workspace/boot.elf
Memory write error at 0x80226400. Debug Transport Module: data corruption (ID)
xsdb% Info: Hart #0 (target 3) Running (FPGA reprogrammed, wait for debugger resync)
```

`vbindiff` has a *next difference* function and I use it to confirm all leading data is identical.

[image]

I tried increasing the JTAG Frequency and it allows me to upload more of `boot.elf` but I have not yet had the luck to upload it completely. Any suggestions will be appreciated.

eugene-tarassov, 2022-08-03:
What JTAG cable are you using? "Debug Transport Module: data corruption" means bad data received on JTAG TDO. It looks like defective or incompatible JTAG cable. Also it can be signal integrity problem - EMI or ground noise.

mwrnd, 2022-08-04:
I am using a Waveshare Platform Cable USB II [Clone](https://www.waveshare.com/platform-cable-usb.htm). It has thus far worked well for me. No issues when JTAG programming 128MB of FPGA Configuration Memory or otherwise using Vivado Hardware Manager. Also, it writes up to 94% of `boot.elf` using `xsdb`. My hunch is that this is some system level interaction. `dow` aborts during a block write.

[image]

I have tried running the JTAG Adapter and FPGA board using different computer power supplies as well as off the same power supply. I also tried different cables. I tried a Rocket64b1 single core design with most peripherals removed. I tried different `jtag freq` settings. I tried `xsdb` from Vivado 2021.2 and 2018.2.

Here is a selection of Memory Write Errors running `dow boot.elf`:
```
Memory write error at 0x8020FC00. FPGA reprogrammed, wait for debugger resync
Memory write error at 0x80240000. Cannot read t0. Debug Transport Module: data corruption (ID)
Memory write error at 0x80000100. Debug Transport Module error 0x08
Memory write error at 0x80225200. Debug Transport Module: data corruption (ID)
Memory write error at 0x80002300. Debug Transport Module: data corruption 0x00371006 -> 0x00000000
Memory write error at 0x8021A400. Debug Transport Module: data corruption 0x00371006 -> 0x00000000
Memory write error at 0x80225000. Debug Transport Module: data corruption 0x740A70AA -> 0x00000000
Memory write error at 0x80006A00. Debug Transport Module: data corruption 0x69A26942 -> 0x01A26942
```

I traced through the `xsdb` scripts and found that Vivado's `hw_server` handles all the low-level communication and has debug options.
```
/tools/Xilinx/Vivado/2021.2/bin/unwrapped/lnx64.o/hw_server --help
...
  -l<level>        set log level, the level is comma separated list of:
      alloc        memory allocation and deallocation (0x1)
      eventcore    main event queue (0x2)
      waitpid      waitpid() events (0x4)
...
```

I ran `sudo gedit /tools/Xilinx/Vivado/2021.2/bin/xsdb` and added all debug options to the call to `hw_server` except full `jtag` tracing.
```
-l alloc,eventcore,waitpid,events,protocol,context,children,discovery,asyncreq,proxy,tcflog,elf,stack,plugin,shutdown,disasm,jtag2,slave,dpc -L hw_server_log_xsdb
```

[image]

I ran `xsdb` again and this time kept pressing PrintScreen during the call to `dow boot.elf` to keep track of event timing. `dow` aborted between 56:06 and 56:07. I do not know how to interleave `xsdb` error messages and the `hw_server` log data.

Error from `xsdb`, note write error at `0x8022D500`:
```
Failed to download /home/user/boot.elf
Memory write error at 0x8022D500. FPGA reprogrammed, wait for debugger resync
xsdb% Info: Hart #0 (target 3) Stopped at 0x10dd4 (Suspended)
```

Snippets from `hw_server_log_xsdb`:
```
...
TCF 09:56:06.421: Peer TCP:127.0.0.1:38872: Command: C 108 Memory set ...
TCF 09:56:06.421: loc_alloc(65536) = 0x30e4bb0
TCF 09:56:06.421: context: write memory ctx JTAG-jsn-DLC10-13724000000d01-14000003-0.DTM10, address 8022C000, size 16384
TCF 09:56:06.421: async_req_complete: req 0x3085730, type 5, error 0
TCF 09:56:06.421: post_event: event 0x1d0d200, handler 0x40f720, arg 0x3085730, runtime 49:49.245
TCF 09:56:06.423: post_event: event 0x1d0d138, handler 0x411690, arg 0
TCF 09:56:06.558: loc_free 0x309dce0
TCF 09:56:06.558: loc_alloc(44) = 0x30967a0
TCF 09:56:06.558: loc_free 0x309dd10
TCF 09:56:06.558: loc_alloc(78) = 0x309c340
TCF 09:56:06.558: post_event: event 0x1d0d0c0, handler 0x40fc30, arg 0x3085630
TCF 09:56:06.558: post_event: event 0x1d0d0e8, handler 0x40fd50, arg 0x3085630, runtime 49:49.383
TCF 09:56:06.558: loc_alloc_zero(64) = 0x30aa3d0
TCF 09:56:06.558: loc_alloc(78) = 0x3080840
TCF 09:56:06.558: post_event: event 0x1d0d228, handler 0x40fd50, arg 0x3085630, runtime 49:49.383
TCF 09:56:06.558: loc_free 0x30e4bb0
TCF 09:56:06.558: post_event: event 0x1d0d070, handler 0x4b5500, arg 0
TCF 09:56:06.558: run_event_loop: event 0x1d0d250, handler 0x40f4f0, arg 0x30858f0
TCF 09:56:06.558: async_req_post: req 0x30858f0, type 4
TCF 09:56:06.558: run_event_loop: event 0x1d0d278, handler 0x463770, arg 0
TCF 09:56:06.559: async_req_complete: req 0x30858f0, type 4, error 0
TCF 09:56:06.559: post_event: event 0x1d0d098, handler 0x40f4f0, arg 0x30858f0, runtime 49:49.383
[... 96 lines cut ...]
TCF 09:56:12.374:   cfa                     0
TCF 09:56:12.374: Frame 1
TCF 09:56:12.374: loc_free 0x30abdc0
TCF 09:56:12.374: loc_alloc(59) = 0x30aa420
...
```

Searching the log for `error` returns mostly `async_req_complete ... error 0`. No real errors other than the stack crawl.

```
cat hw_server_log_xsdb  |  grep -i error
TCF 09:55:52.758: async_req_complete: req 0x1d1bbc0, type 6, error 0
TCF 09:55:52.758: async_req_complete: req 0x1d1bbc0, type 6, error 0
...
TCF 09:56:12.217: async_req_complete: req 0x30858f0, type 4, error 0
TCF 09:56:12.374: Stack crawl: Memory read error at 0x10DD4. Illegal access to I/O address space. Invalid address
TCF 09:56:12.374: Stack crawl: Memory read error at 0x10DD0. Illegal access to I/O address space. Invalid address
TCF 09:56:12.376: async_req_complete: req 0x3085730, type 5, error 0
TCF 09:56:12.376: async_req_complete: req 0x30858f0, type 4, error 0
...
```

I also tried running `dow` with full `jtag` tracing enabled but nothing stood out to me.

Any suggestions for further exploration?

eugene-tarassov, 2022-08-04:
> I am using a Waveshare Platform Cable USB II Clone.

I don't have this cable - don't know how good it is.  To isolate the problem, I recommend to try one of officially supported cables like [JTAG-HS3 Programming Cable](https://digilent.com/shop/jtag-hs3-programming-cable/).

> No issues when JTAG programming 128MB of FPGA Configuration Memory or otherwise using Vivado Hardware Manager. 

RISC-V JTAG differ from other JTAG accesses, because it uses strong 32-bit CRC to detect errors in JTAG data bits. JTAG programming and Vivado Hardware Manager don't have such error detection. Only RISC-V JTAG is capable to detect all single bit errors.

> My hunch is that this is some system level interaction.

It is possible. I don't know. I have fully automated build and test system running continuously with multiple target boards. It does not show any "data corruption" errors.

> I traced through the xsdb scripts and found that Vivado's hw_server handles all the low-level communication

Actually, hw_server handles all debugger functionality, not just low-level communication. xsdb is a thin Tcl wrapper for hw_server .

> added all debug options to the call to hw_server except full jtag tracing.

In this case, the most useful option is `-ljtag` (not jtag2), you don't really need all other options.

mwrnd, 2022-08-04:
I fixed it and it turned out to be trivial.

The Innova-2 Flex I am targeting controls JTAG access to prevent accidental Configuration Memory overwrite. I thought it was simply controlling the FLASH IC Write-Protect pins but it seems to control the JTAG signals. None of this is publicly documented. It is confusing as much of JTAG functionality works even with JTAG Access supposedly disabled.

The fix is to Enable JTAG Access:

[image]

The `xsdb` `dow` command then works without issue:

[image]

Thanks for all your help. Truly appreciated!

mwrnd, 2022-08-05:
In case someone needs to debug their JTAG Adapter, interleaved `xsdb` and `hw_server` logging may be useful.

`sudo gedit /tools/Xilinx/Vivado/2021.2/scripts/xsdb/xsdb/xsdb.tcl` and add the core JTAG-related logging options to the call the `exec $server ...` which is on line 2986 in Vivado 2021.2's `xsdb`:
```
-l protocol,context,jtag,jtag2 -L hw_server_log_xsdb
```

[image]

To interleave JTAG log data with `xsdb`, its `stdout` and `stderr` can be append-redirected `>>` to the log file. `xsdb` itself does not generate useful redirected output but the raw shell command that `xsdb` generates does. This raw shell command can be found by running `bash -x xsdb`.
```Shell
source /tools/Xilinx/Vivado/2021.2/settings64.sh
/tools/Xilinx/Vitis/2021.2/bin/unwrapped/lnx64.o/rlwrap -rc -f /tools/Xilinx/Vitis/2021.2/scripts/xsdb/xsdb/cmdlist /tools/Xilinx/Vitis/2021.2/bin/loader -exec rdi_xsdb  >>hw_server_log_xsdb  2>>hw_server_log_xsdb
```

Commands must then be sent blindly to `xsdb`.
```
connect
targets 3
dow boot.elf
exit
```

[image]

Edit `hw_server_log_xsdb` and look for the `Memory write` error which is now interleaved with the `hw_server` debug log.

[image]

Unfortunately, there is no obvious JTAG log element for the Debug Transport Module error.

mwrnd, 2022-08-06:
It is possible to redirect `xsdb` output to both the screen and the log file but usage is messy. Lines are repeated or disappear.

```Shell
(/tools/Xilinx/Vitis/2021.2/bin/unwrapped/lnx64.o/rlwrap -rc -f /tools/Xilinx/Vitis/2021.2/scripts/xsdb/xsdb/cmdlist /tools/Xilinx/Vitis/2021.2/bin/loader -exec rdi_xsdb 2>  >(tee -a /dev/stderr)) | tee -a hw_server_log_xsdb

![xsdb interactive with logging](https://raw.githubusercontent.com/mwrnd/innova2_flex_xcku15p_notes/main/debug_notes/img/xsdb_into_hw_server_log_xsdb_interactive.png)

closed 2022-08-15 by mwrnd